espressif/esp-idf #8766: Flush failed inside TinyUSB-CDC-ACM

filgra96, 2022-04-12:
## Environment

- Development Kit: ESP32-S2 MINI-1
- Module or chip used: ESP32-S2 MINI-1
- IDF version (run ``git describe --tags`` to find it): v4.4
- Build System:  idf.py
- Operating System: Windows 10
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: Yes, VSCode
- Power Supply: USB

## Problem Description
I use the USB port of the ESP32-S2 to communicate informations with another device, in a bidirectional way.
I use the official ibrary TinyUSB for my ESP32-S2.
If I need to write 4 bytes on the USB, I use the functions:
```cpp
(void) tinyusb_cdcacm_write_queue(TINYUSB_CDC_ACM_0, buffer_array, 4);
(void) tinyusb_cdcacm_write_flush(TINYUSB_CDC_ACM_0, 0);
```
All works well. However sometimes I read a warning on the terminal, here the text:
`
flush fauled (res: 0)
`

### Expected Behavior

The text I mentioned is inside the function `tinyusb_cdcacm_write_flush()`, it means that the funcion `tud_cdc_n_write_flush()` wasn't able to flush the data I want to write.
Beyond the grammatical error (fauled instead of failed), I noticed that the descritpion of `tud_cdc_n_write_flush()` is:
> Force sending data if possible, return number of forced bytes

So I think it's not an error if I force 0 bytes, because it means it's all ok! Also, after that the return value of `tinyusb_cdcacm_write_flush()` is **ESP_FAIL**.

I don't think is an error, doing so and using ESP_ERROR_CHECK() I will obtain an unwanted reset.

### Code to reproduce this issue

```cpp
esp_err_t tinyusb_cdcacm_write_flush(tinyusb_cdcacm_itf_t itf, uint32_t timeout_ticks)
{
    if (!get_acm(itf)) { // non-initialized
        return ESP_FAIL;
    }

    if (!timeout_ticks) { // if no timeout - nonblocking mode
        int res = tud_cdc_n_write_flush(itf);
        if (!res) {
            ESP_LOGW(TAG, "flush fauled (res: %d)", res);
            return ESP_FAIL;
        } else {
            if (tud_cdc_n_write_occupied(itf)) {
                ESP_LOGW(TAG, "remained data to flush!");
                return ESP_FAIL;
            }
        }
        return ESP_ERR_TIMEOUT;
    } else { // trying during the timeout
        uint32_t ticks_start = xTaskGetTickCount();
        uint32_t ticks_now = ticks_start;
        while (1) { // loop until success or until the time runs out
            ticks_now = xTaskGetTickCount();
            if (!tud_cdc_n_write_occupied(itf)) { // if nothing to write - nothing to flush
                break;
            }
            if (tud_cdc_n_write_flush(itf)) { // Success
                break;
            }
            if ( (ticks_now - ticks_start) > timeout_ticks ) { // Time is up
                ESP_LOGW(TAG, "Flush failed");
                return ESP_ERR_TIMEOUT;
            }
            vTaskDelay(1);
        }
        return ESP_OK;
    }
}

```

tore-espressif, 2022-04-13:
Hello @filgra96 

Sometimes the IN endpoint can be busy, so the flush function `tinyusb_cdcacm_write_flush` can fail. But, you can leverage from the `timeout_ticks` parameter. 

If you pass `timeout_ticks` value, the flush function will retry the data transaction, until success or timeout. Could you try:
```c
tinyusb_cdcacm_write_flush(TINYUSB_CDC_ACM_0, pdMS_TO_TICKS(10));
```
?

> So I think it's not an error if I force 0 bytes, because it means it's all ok!

In my opinions it is correct that the flush function returns `ESP_FAIL` in case you try to flush an _empty buffer_. User called flush function, but nothing was flashed = fail.

tore-espressif, 2022-04-13:
That will depend solely on your usecase.

In TinyUSB there are two phases of preparing the USB Bulk transfer:
1. Fill data buffer that you want to send (e.g. single or multiple calls of `tinyusb_cdcacm_write_queue(TINYUSB_CDC_ACM_0, buffer_array, sizeof(buffer_array));`)
2. Mark the buffer as ready to send - USB Host must initiate read, so called IN transfer. (e.g. `tinyusb_cdcacm_write_flush(TINYUSB_CDC_ACM_0, pdMS_TO_TICKS(10));`)

**_Side note:_**  You don't _have to_ call the flush function. TinyUSB will mark the buffer as ready when the buffer is full. This is useful if your device is sending a stream of data.

`tinyusb_cdcacm_write_flush` function is just a convenient wrapper around `tud_cdc_n_write_flush()`, you can call it directly, if you want/need.

Unfortunately, TinyUSB does not provide peek function, that would allow you to know whether you can 'force send' the data. That's why we provide the convenience function that will retry in case of failure.

filgra96, 2022-04-14:
Dear @tore-espressif, thank you very much for your explanation, now it is more clear the behavior hidden in these functions.
I think I can close this issue.

closed 2022-04-14 by filgra96