espressif/esp-idf #8768: mbedtls_ssl_setup returned -0x7f00 while sending multiple mails

MaazSk, 2022-04-12:
## Environment

- Development Kit:      [ESP32-DevKit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1]
- Module or chip used:  [ESP32-WROOM-32|]
- IDF version (run ``git describe --tags`` to find it): v4.3
   
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
  
- Operating System:     [Linux]

- Using an IDE?: [VS Code]
- Power Supply:         [USB]

## Problem Description

I am trying to send mail, again and again, using the smtp_client example but after sending mail for 3 or 4 times the code crashes with the following error ```mbedtls_ssl_setup returned -0x7f00```. Any help related to it would be grateful.

Thanks and Regards,
Maaz Shaikh

laukik-hase, 2022-04-14:
Hello, @MaazSk!

Thank you for sending the issue report. I was investigating this issue and was able to reproduce it with the `smtp_client` example on the ESP-IDF `v4.3` tag.

Could you try out this [patch](https://github.com/espressif/esp-idf/files/8487318/smtp_client_fix.zip) and let me know if it works for you?

MaazSk, 2022-04-14:
I was able to resolve the issue by adding the following lines sorry didn't update you

    ```
mbedtls_x509_crt_free(&cacert);
    mbedtls_ssl_config_free(&conf);
    mbedtls_ctr_drbg_free(&ctr_drbg);
    mbedtls_ssl_free(&ssl);
```

Thanks and regards,
Maaz Shaikh

closed 2022-04-14 by MaazSk