espressif/esp-idf #8769: export.sh locks up PC if executed from .profile and it finds an issue

980f, 2022-04-12:
If export.sh is executed from the user's `.profile` (ubuntu 20.04, presumably any Debian) AND there is an error (in my case an old version of `openocd` in path) then it pops up a dialog box, but that box does not respond to user input. The system is locked up.

I had to boot on a different drive to remove that from my `.profile`.

A warning in the installation instructions to not do this would be nice, I do not think it is urgent to try to make the script aware of that execution context.

esp-idf version: downloaded via git checkout on 19Mar2022, master branch.

igrr, 2022-04-12:
Hi @980f, there is a warning about this in the getting started guide: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-macos-setup.html#step-4-set-up-the-environment-variables

> Technically, you can add export.sh to your shell’s profile directly; however, it is not recommended. Doing so activates IDF virtual environment in every terminal session (including those where IDF is not needed), defeating the purpose of the virtual environment and likely affecting other software.

We were not familiar with this failure mode:
> AND there is an error (in my case an old version of openocd in path) then it pops up a dialog box, but that box does not respond to user input

Can you please describe what dialog box that is? `export.sh` is a command line program, it shouldn't be able to open dialog boxes. Maybe it's something that is done by the terminal program if the profile script fails to execute? In that case, wouldn't it be different to sourcing any other script from .profile, and running into an issue that the script exits with an error?

980f, 2022-04-12:
It says that it is inadvisable, but in my case it was somewhat fatal, and
very annoying to try to recover from.
I admit that I didn't notice that warning as I was using a Clion user's
installation instructions. I should have read Espressif's as well.

The error comes out as console text when run in other contexts, such as
from .bashrc or the via the terminal-- I was quite surprised when running
from .profile generated a dialog box. export.sh was the last line of my
.profile.

The error that triggered this is reported as: "Not using an unsupported
version of tool openocd-esp32 found in PATH: 0.10.0.".

The box that came up was over a black screen, had color highlighting, and
an OK button in the bottom right corner.
--
Thank you for politely correcting me for not Reading The Fine Manual.

igrr, 2022-04-12:
Interesting, the message `Not using an unsupported version of tool openocd-esp32...` is actually not a fatal error, just a warning which gets printed to `stderr`. The export script should still return without an error.

I wonder if it has something to do with the terminal application (or the window manager) trying to show the `stderr` output of the .profile script in a dialog box. I don't why it doesn't respond to user input, though.

There seems to be a long thread on a similar sounding issue [over at Ubuntu issue tracker](https://bugs.launchpad.net/ubuntu/+source/gdm/+bug/678421), but I'm not sure what the resolution is.

980f, 2022-10-11:
X11 startup is a perpetual mystery!

If I can get an older computer to boot I might investigate this more.

mfialaf, 2024-04-23:
The error dialog box afer logging out and back in still appears. But clicking on `ok` works and user is logged in as usual.

closed 2024-04-23 by dobairoland