espressif/esp-idf #8771: [CRASH][v4.4] fileno(fd) crashes if File Descriptor is closed

chipweinberger, 2022-04-12:
ESP32-S3
ESP-IDF release/v4.4 (https://github.com/espressif/esp-idf/commit/c29343eb94d2f2ca17b3a5b38c82452e556147f2) from a few days ago.

I called code that looked like this:
```
fd = fopen
fwrite(fd)
fclose(fd)
fflush(fd)
fno = fileno(fd) // crashes here
fsync(fno)
```

```
assert failed: xQueueSemaphoreTake queue.c:1549 (pxQueue->uxItemSize == 0)

panic_abort at /components/esp_system/panic.c:402

0x40380a05: esp_system_abort at /components/esp_system/esp_system.c:128

0x4038b942: __assert_func at /components/newlib/assert.c:85

0x40381cc1: xQueueSemaphoreTake at /components/freertos/queue.c:1549 (discriminator 1)

0x40381ea2: xQueueTakeMutexRecursive at /components/freertos/queue.c:731

0x40377e9d: lock_acquire_generic at /components/newlib/locks.c:153

0x4037803a: _lock_acquire_recursive at /components/newlib/locks.c:167

0x40378257: __retarget_lock_acquire_recursive at /components/newlib/locks.c:323 (discriminator 3)

0x420e8d6a: fileno at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/fileno.c:70 (discriminator 2)

0x420110d6: pd_fileno(__sFILE*, pd_error_t*) at /pd_file_utils.cpp:440 (discriminator 2)

00x420118b0: pd_file_json_write(char const*, cJSON*, pd_error_t*) at /pd_file_json.cpp:72

:0x3fcf5e00 0x4201022a:0x3fcf5e40 0x4200c548:0x3fcf5f20 0x421020x4201022a: pd_coredump_write_summary_and_extra_to_sd_card(pd_error_t*) at /pd_coredump.cpp:327

0x4200c548: app_main at /pd_main.cpp:358

64b:0x3fcf5f80 0x40385b5e:0x3fcf5fb0 
0x40385b5e: vPortTaskWrapper at /components/freertos/port/xtensa/port.c:131
```

Alvin1Zhang, 2022-04-13:
Thanks for reporting.

igrr, 2022-04-13:
Hi @chipweinberger,

According to fclose documentation (e.g. https://man7.org/linux/man-pages/man3/fclose.3.html):

> Upon successful completion, 0 is returned.
> Otherwise, EOF is returned and errno is set to indicate the error.  In either case, any further access (including another call to fclose()) to the stream results in undefined behavior.

Essentially, `FILE* fp` is a pointer to a structure and `fclose` frees the memory allocated for this structure (among other things). The next access to the structure as part of `fileno` call triggers an access to already-freed memory. It could have also crashed in `fflush` — it's basically undefined behavior after that point.

chipweinberger, 2022-04-13:
Thanks for the explanation. You are right, the documentation is clear this is "undefined".

My gut reaction always is: crash in esp-idf = file bug report. 

IMO ESP-IDF shouldn't purposely crash. Errors + logging should always be preferred except in truly non-recoverable/ super hard to recover scenarios. People don't want embedded systems crashing.

On top of that, the place that this code crashes is pretty opaque. errors + logging makes development much friendlier.

But ofc it's your code and your choice. =) Just my 2 cents!

igrr, 2022-04-14:
@chipweinberger I understand your comment.

In IDF, where we have some design flexibility we prefer to use error codes to indicate errors.

In this case we don't intentionally choose 'crashing' as the desired esp-idf behavior. It just follows from C language allowing undetected use-after-free. The behavior would be the same if you were running this code on Linux, for instance. 

Avoiding crashes on use-after-free errors and clearly telling the developer where the issue would be really nice and would make a lot of C developers more productive. (It would be even nicer if the compiler could refuse to compile such code, but alas... Not with C.)

We will consider supporting kASAN, as it is the only run-time memory checking solution that is viable to use on microcontrollers.

closed 2022-04-14 by igrr

chipweinberger, 2022-04-23:
You could know pretty easily if you close a FP* point if you have it to a magic value, say "OPEN", then replacing those bytes with the bytes "CLSD" when we close it.

fclose, fflush(fd), etc, would check for the magic bytes "OPEN", and return an error if not found.