espressif/esp-idf #8772: LVGL example build error on Eclipse (ESP-IDE) with ESP-IDF v4.4

Bardi45, 2022-04-13:
## Environment
- Development Kit:      [ESP32-Wrover-Kit]
- Kit version: v4.1
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it):  v4.4
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     Windows
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: Yes, Eclipse (ESP-IDE)
- Power Supply:     USB

## Problem Description
Hi,
I am new to ESP32 and trying to compile LVGL example but I'm getting the attached error. My goal is to port this example to work on  SPI-LCD. It would be appreciated if you help to fix the issue.
FYI, I am able to compile, program and test other examples on the ESP-WROVER-KIT.

[error.txt](https://github.com/espressif/esp-idf/files/8477675/error.txt)

Thanks,
Bardi

suda-morris, 2022-04-13:
>   "idf_component.yml" file was found for components:

  	D:/Espressif/workspace/lvgl/main

  However, the component manager is not enabled.
Call Stack (most recent call first):
  D:/Espressif/esp-idf-v4.4/tools/cmake/project.cmake:378 (idf_build_process)
  CMakeLists.txt:4 (project)

Seems like the eclipse doesn't enable the package manager by default.

Bardi45, 2022-04-13:
So, what's the solution to fix it?

Bardi45, 2022-04-13:
Ok, found how to do it:

- Open an "ESP-IDF Terminal" from eclipse barmenu,
- Change to target project folder,
- Enter: "idf.py reconfigure",

See this: [IDF Component Manager](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-component-manager.html)

closed 2022-04-13 by Bardi45