espressif/esp-idf #8773: "usb/usb_host.h: No such file or directory" for ESP32S2 USB MSC example

Maldus512, 2022-04-13:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32S2]
- IDF version (run ``git describe --tags`` to find it):
    v4.4-367-gc29343eb94
- Build System:         [idf.py]
- Compiler version 
    xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [external 3.3V]

## Problem Description

I am trying to work with the msc USB driver on an ESP32S2 powered custom board. I have seen that the example has made its way into the release/v4.4 branch and I wanted to try it. Unfortunately the example on the release/v4.4 does not compile, failing with the following error:

```
/home/maldus/Source/esp-idf-v4.4/examples/peripherals/usb/host/msc/components/msc/src/msc_host.c:18:10: fatal error: usb/usb_host.h: No such file or directory
 #include "usb/usb_host.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
[11/21] Performing build step for 'bootloader'
[1/1] cd /home/maldus/Source/esp-idf-v4.4/examples/peripherals/usb/host/msc/build/bootloader/esp-idf/esptool_py && /home/maldus/.espressif/python_env/idf5.0_py3.9_env/bin/python /home/maldus/Source/esp-idf-master/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /home/maldus/Source/esp-idf-v4.4/examples/peripherals/usb/host/msc/build/bootloader/bootloader.bin
Bootloader binary size 0x6350 bytes. 0xcb0 bytes (11%) free.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

The `usb/usb_host.h` header file is actually present in the clone repository. Interestingly the same example works if the branch `master` is used.

### Expected Behavior

A successful compilation

### Actual Behavior

The compilation error above

### Steps to reproduce

1. clone and install ESP-IDF release/v4.4
2. try to compile the example under `examples/peripherals/usb/host/msc`

### Code to reproduce this issue

## Debug Logs

## Other items if possible

tore-espressif, 2022-04-19:
Hello @Maldus512 , 
From the log you shared it is not clear whether you are compiling for correct target? Have you run `idf.py set-target esp32s2` ?

Maldus512, 2022-04-19:
> Hello @Maldus512 , From the log you shared it is not clear whether you are compiling for correct target. Have you run `idf.py set-target esp32s2` ?

This is quite embarassing. Indeed I had not set esp32-s2 as the target, it now works as expected. Thank you for the assistance!

closed 2022-04-19 by Maldus512

tore-espressif, 2022-04-19:
It is still valuable feedback :) we have quite a few issues like this, the error message should be more indicative.

juddvinet, 2022-10-22:
[image]