espressif/esp-idf #8775: WIFI_MODE_AP: httpd server extremely slow particular android devices

mmoas21, 2022-04-14:
I observe that using particular android devices, in AP mode, httpd server is extremely slow.

I am testing the AP mode and the httpd server with Samsung Galaxy Tab S7 FE (SM-T733). I observe that on this particular device, opening an esp IP address is very slow and fails most times due to timeout. Different browsers (samsung browser, chrome, firefox) act the same. If instead of AP mode, I am connected in STA mode, through the router, there is no more problem on the Galaxy Tab. If I am using other device (iPhone 8, Samsung Note 9, OnePlus 3T, macbook), page loading is fast.

I already applied most of advices found, but none of them helped. The most important lines of the configuration:
```
	esp_netif_t* ap = esp_netif_create_default_wifi_ap();
	esp_netif_set_hostname(ap, hostname);

	wifi_init_config_t config = WIFI_INIT_CONFIG_DEFAULT();
	ESP_ERROR_CHECK(esp_wifi_init(&config));
	ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_RAM));
	ESP_ERROR_CHECK(esp_wifi_set_ps(WIFI_PS_NONE));
	...
	wifi_config_t wifi_config = {
		.ap = {
			.channel = 3,
			.max_connection = 5,
			.beacon_interval = 100,
			.authmode = WIFI_AUTH_WPA2_PSK
		}
	};

	ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_AP));
	ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_AP, &wifi_config));
	ESP_ERROR_CHECK(esp_wifi_set_bandwidth(WIFI_IF_AP, WIFI_BW_HT20));
	ESP_ERROR_CHECK(esp_wifi_start());
```

I tried turning on LWPI debug but the only unusual thing is multiple `tcp_fasttmr: delayed ACK`:
```
tcp_slowtmr: processing active pcb
tcp_slowtmr: processing active pcb
tcp_slowtmr: polling application
tcp_slowtmr: processing active pcb
tcp_slowtmr: polling application
TCP connection established 59724 -> 80.
tcp_fasttmr: delayed ACK
tcp_slowtmr: processing active pcb
tcp_slowtmr: processing active pcb
tcp_slowtmr: processing active pcb
tcp_slowtmr: processing active pcb
tcp_slowtmr: processing active pcb
tcp_slowtmr: polling application
tcp_slowtmr: processing active pcb
tcp_slowtmr: polling application
tcp_slowtmr: processing active pcb
tcp_slowtmr: processing active pcb
tcp_slowtmr: processing active pcb
tcp_slowtmr: processing active pcb
tcp_slowtmr: polling application
tcp_slowtmr: processing active pcb
tcp_slowtmr: polling application
tcp_fasttmr: delayed ACK
tcp_slowtmr: processing active pcb
tcp_slowtmr: polling application
tcp_slowtmr: processing active pcb
```

## Environment

```
ESP-WROOM-32: ESP32-D0WD (revision 1)
```
```
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)

ESP-IDF v4.4-dirty
```

nishanth-radja, 2022-04-18:
@mmoas21 can we have a wireless capture along with the ssid and password used in this wifi connection. Pls take the capture on the httpd is slow and also when it is fast.

mmoas21, 2022-04-29:
Using Wireshark I found that there is a lot of `QoS Null function (No data)`, `TCP Retransmission` and `TCP Dup ACK`.
I found this @tschak909 's post on esp32.com: https://www.esp32.com/viewtopic.php?t=18483
Initial solution (`CONFIG_LWIP_TCP_QUEUE_OOSEQ=n`) didn't help me there, but when I tested their whole WIFI and LWIP configuration all problems disappeared. 
I am attaching below for future:
```
CONFIG_ESP32_WIFI_SW_COEXIST_ENABLE=y
CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM=16
CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM=128
CONFIG_ESP32_WIFI_STATIC_TX_BUFFER=y
CONFIG_ESP32_WIFI_TX_BUFFER_TYPE=0
CONFIG_ESP32_WIFI_STATIC_TX_BUFFER_NUM=16
CONFIG_ESP32_WIFI_CACHE_TX_BUFFER_NUM=32
# CONFIG_ESP32_WIFI_CSI_ENABLED is not set
CONFIG_ESP32_WIFI_AMPDU_TX_ENABLED=y
CONFIG_ESP32_WIFI_TX_BA_WIN=32
CONFIG_ESP32_WIFI_AMPDU_RX_ENABLED=y
CONFIG_ESP32_WIFI_RX_BA_WIN=32
CONFIG_ESP32_WIFI_NVS_ENABLED=y
CONFIG_ESP32_WIFI_TASK_PINNED_TO_CORE_0=y
# CONFIG_ESP32_WIFI_TASK_PINNED_TO_CORE_1 is not set
CONFIG_ESP32_WIFI_SOFTAP_BEACON_MAX_LEN=752
CONFIG_ESP32_WIFI_MGMT_SBUF_NUM=32
# CONFIG_ESP32_WIFI_DEBUG_LOG_ENABLE is not set
# CONFIG_ESP32_WIFI_IRAM_OPT is not set
# CONFIG_ESP32_WIFI_RX_IRAM_OPT is not set
CONFIG_ESP32_WIFI_ENABLE_WPA3_SAE=y

CONFIG_LWIP_MAX_ACTIVE_TCP=16
CONFIG_LWIP_MAX_LISTENING_TCP=16
CONFIG_LWIP_TCP_HIGH_SPEED_RETRANSMISSION=y
CONFIG_LWIP_TCP_MAXRTX=12
CONFIG_LWIP_TCP_SYNMAXRTX=6
CONFIG_LWIP_TCP_MSS=1440
CONFIG_LWIP_TCP_TMR_INTERVAL=250
CONFIG_LWIP_TCP_MSL=60000
CONFIG_LWIP_TCP_SND_BUF_DEFAULT=65534
CONFIG_LWIP_TCP_WND_DEFAULT=65534
CONFIG_LWIP_TCP_RECVMBOX_SIZE=64
# CONFIG_LWIP_TCP_QUEUE_OOSEQ is not set
# CONFIG_LWIP_TCP_SACK_OUT is not set
# CONFIG_LWIP_TCP_KEEP_CONNECTION_WHEN_IP_CHANGES is not set
CONFIG_LWIP_TCP_OVERSIZE_MSS=y
# CONFIG_LWIP_TCP_OVERSIZE_QUARTER_MSS is not set
# CONFIG_LWIP_TCP_OVERSIZE_DISABLE is not set
CONFIG_LWIP_TCP_RTO_TIME=1500
```
I am closing as issue solved

closed 2022-04-29 by mmoas21