espressif/esp-idf #8776: pre_encrypted_ota

songotag, 2022-04-14:
Hello,

I use esp idf 4.4

I tried to compile the example " pre_encrypted_ota "  and i've this error 👍 

`../main/pre_encrypted_ota.c:63:30: error: unknown type name 'decrypt_cb_arg_t'; did you mean 'esp_decrypt_cfg_t'?
 static esp_err_t _decrypt_cb(decrypt_cb_arg_t *args, void *user_ctx)
                              ^~~~~~~~~~~~~~~~
                              esp_decrypt_cfg_t
../main/pre_encrypted_ota.c: In function 'pre_encrypted_ota_task':
../main/pre_encrypted_ota.c:140:10: error: 'esp_https_ota_config_t' {aka 'struct <anonymous>'} has no member named 'decrypt_cb'
         .decrypt_cb = _decrypt_cb,
          ^~~~~~~~~~
../main/pre_encrypted_ota.c:140:23: error: '_decrypt_cb' undeclared (first use in this function)
         .decrypt_cb = _decrypt_cb,
                       ^~~~~~~~~~~
../main/pre_encrypted_ota.c:140:23: note: each undeclared identifier is reported only once for each function it appears in
../main/pre_encrypted_ota.c:141:10: error: 'esp_https_ota_config_t' {aka 'struct <anonymous>'} has no member named 'decrypt_user_ctx'
         .decrypt_user_ctx = (void *)decrypt_handle,
          ^~~~~~~~~~~~~~~~
At top level:
../main/pre_encrypted_ota.c:42:18: warning: 'validate_image_header' defined but not used [-Wunused-function]
 static esp_err_t validate_image_header(esp_app_desc_t *new_app_info)`

Is there any solution to solve this ?

thanks

Alvin1Zhang, 2022-04-14:
Thanks for reporting.

hmalpani, 2022-04-14:
Hello @songotag
To support this feature, the `esp_https_ota` component has also been updated. You may try this example with the ESP-IDF master branch.
This might also help: https://github.com/espressif/esp-idf/issues/6172#issuecomment-1057022417

songotag, 2022-04-14:
Hello @hmalpani,

I tried this example here : https://github.com/espressif/esp-idf/tree/master/examples/system/ota/pre_encrypted_ota with SDK 4.4 on  esp32 and no luck at tihis time.

mahavirj, 2022-04-18:
@songotag 

As mentioned by @hmalpani, `pre_encrypted_ota` is supported on current ESP-IDF master branch only and it shall be part of v5.0 release.

If you would like to use it with current v4.4 release then please cherry-pick commits from following:

https://github.com/espressif/esp-idf/commit/69f51a989c8a9eaf8f559bb5f4f080146b464c44
https://github.com/espressif/esp-idf/commit/61c68b7f4d1fb6b97fe26844737ee60ea9bb564b

Since, v4.4 also support ESP-IDF component manager framework, I believe example should build fine. 

Please let us know if you still run into any issues.

mahavirj, 2022-04-22:
Hopefully reply above helped. Closing the issue.

closed 2022-04-22 by mahavirj

songotag, 2022-04-24:
> @songotag
> 
> As mentioned by @hmalpani, `pre_encrypted_ota` is supported on current ESP-IDF master branch only and it shall be part of v5.0 release.
> 
> If you would like to use it with current v4.4 release then please cherry-pick commits from following:
> 
> [69f51a9](https://github.com/espressif/esp-idf/commit/69f51a989c8a9eaf8f559bb5f4f080146b464c44) [61c68b7](https://github.com/espressif/esp-idf/commit/61c68b7f4d1fb6b97fe26844737ee60ea9bb564b)
> 
> Since, v4.4 also support ESP-IDF component manager framework, I believe example should build fine.
> 
> Please let us know if you still run into any issues.

@mahavirj 
thanks for your help ;-)

bobbygz, 2022-08-31:
@mahavirj 

I know this issue is closed but just to provide some feedback.  I cherry picked the two commits on idk v4.4.2 and built the pre_encrypted_ota example I successfully.  Have not yet tested but will report back when I do. 

Thanks!!

shachar-limor, 2022-09-19:
> @mahavirj
> 
> I know this issue is closed but just to provide some feedback. I cherry picked the two commits on idk v4.4.2 and built the pre_encrypted_ota example I successfully. Have not yet tested but will report back when I do.
> 
> Thanks!!

Hi @bobbygz ,
I tried to replicate what you did (clone a v4.4.2 release and then cherry pick the 2 commits), however I wasnt able to do it.
I am running into the following:
=================
ShacharLimor@DESKTOP_SHACHAR MINGW64 ~/esp/esp-idf-v4.4.2_pre_encrypted_ota (pre_encrypted_ota)
$ git status
On branch pre_encrypted_ota
nothing to commit, working tree clean

ShacharLimor@DESKTOP_SHACHAR MINGW64 ~/esp/esp-idf-v4.4.2_pre_encrypted_ota (pre_encrypted_ota)
$ git cherry-pick -x 61c68b7
error: commit 61c68b7f4d1fb6b97fe26844737ee60ea9bb564b is a merge but no -m option was given.
fatal: cherry-pick failed

ShacharLimor@DESKTOP_SHACHAR MINGW64 ~/esp/esp-idf-v4.4.2_pre_encrypted_ota (pre_encrypted_ota)
$ git cherry-pick -x 69f51a9
error: commit 69f51a989c8a9eaf8f559bb5f4f080146b464c44 is a merge but no -m option was given.
fatal: cherry-pick failed
=================

@bobbygz - can you share your experience and let me whether you run into the same issue? How did you go forward with the cherry picking?

Eventually I did made some progress by just taking the master but I am sure there is some learning here for me..
Thanks,
Shachar