espressif/esp-idf #8777: gpio_install_isr_service( ESP_INTR_FLAG_LEVEL3) fails with 0x105 (resource not found)

ultrapalace, 2022-04-14:
## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  ESP32s2 wrover on custom pcb
- IDF version 
    v4.2.3
- Build System:         vscode
- Compiler version command not found
- Operating System:     macos
- Using an IDE?: vscode
- Power Supply:         5v external

## Problem Description

i have a 62.5mhz external xtal connected to GPIO_12. I need to trigger very precise interrupts from this pin to run a UART peripheral that I have coded in c. It reads and writes 4 UARTS, and works well.

Using a scope I have determined that, although my clock input is very steady, some interrupts start quite late. My assumption is that there is a higher priority interrupt running at this moment, so my code is made to wait.

I am now trying to raise the priority of the interrupt, but calls to 

`gpio_install_isr_service( ESP_INTR_FLAG_LEVEL3);`
`gpio_install_isr_service( ESP_INTR_FLAG_LEVEL2);`

result in `E (4877) gpio: esp_intr_alloc failed (0x105)`

`gpio_install_isr_service( ESP_INTR_FLAG_IRAM);` works, but it does not prevent my interrupt from being postponed sometimes.

Why does ` ESP_INTR_FLAG_LEVEL3` fail, and how can I give my interrupt high priority so that it always runs when I receive a clock?

### Expected Behavior
To install a level 3 IRQ

### Actual Behavior
error 0x105

### Steps to reproduce
`gpio_install_isr_service( ESP_INTR_FLAG_LEVEL3);`

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    gpio_install_isr_service( ESP_INTR_FLAG_LEVEL3);
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
E (4877) gpio: esp_intr_alloc failed (0x105)
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400a2617  PS      : 0x00060333  A0      : 0x800a3cf8  A1      : 0x3ffde790  
0x400a2617: esp_intr_get_cpu at /Users/susieburke/esp/esp-idf/components/esp32s2/intr_alloc.c:766

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000608  A5      : 0x3ffc955c  
A6      : 0x3ffc954c  A7      : 0x3ffde730  A8      : 0x800a38c2  A9      : 0x3ffde760  
A10     : 0x3ffc9560  A11     : 0x00001000  A12     : 0x3f00c6fc  A13     : 0x0000130d  
A14     : 0x3f00c4ec  A15     : 0x3ffde740  SAR     : 0x00000014  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x3f00c6fc  LEND    : 0x0000130d  LCOUNT  : 0x40025d9d  
0x40025d9d: _xt_user_exc at /Users/susieburke/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:627

Backtrace:0x400a2614:0x3ffde790 0x400a3cf5:0x3ffde7b0 0x40083ab5:0x3ffde7e0 0x400837f3:0x3ffde830 0x400823e2:0x3ffde850
0x400a2614: esp_intr_get_cpu at /Users/susieburke/esp/esp-idf/components/esp32s2/intr_alloc.c:765

0x400a3cf5: gpio_isr_handler_add at /Users/susieburke/esp/esp-idf/components/driver/gpio.c:462

0x40083ab5: midi_timer_init at /Users/susieburke/mididwarf/build/../main/uart.c:306

0x400837f3: app_main at /Users/susieburke/mididwarf/build/../main/main.c:74

0x400823e2: main_task at /Users/susieburke/esp/esp-idf/components/esp32s2/cpu_start.c:453
```

## Other items if possible

negativekelvin, 2022-04-14:
There is only one available level 3 interrupt per cpu so could be a conflict

songruo, 2022-04-15:
Hi @ultrapalace,

I have tried the code with esp32s2 on idf v4.2.3. It works fine, it is able to allocate a level 3 interrupt for gpio. Do you have other drivers running in the application that might take all the available level 3 interrupts?  You can check the debug log [here](https://github.com/espressif/esp-idf/blob/v4.2.3/components/esp32s2/intr_alloc.c#L668) with Table 56 (CPU Interrupts) of the TRM to see whether there is any peripheral interrupt source taking the level 3 interrupts.

ultrapalace, 2022-04-16:
@negativekelvin @songruo Thank you so much :)
Indeed I have an external DAC taking int 23 which is LEVEL 3. I should have read the Manual more carefully before posting a bug. The documentation in the programming guide is a little vague on this subject.

My impression is that I may need to write my function in assembly now, and acquire a high-level interrupt for this timing-sensitive task. Is that accurate? Or are there other options available to me, with needing an interrupt above the i2s driver using LEVEL 3. The PG states LEVEL 4 and above cannot be in C.

songruo, 2022-04-18:
@ultrapalace It seems like your application involves quite a lot drivers (DAC, I2S, GPIO, etc. Not sure if you are also using WIFI or BLE). But, yes, with that many drivers working together, you are likely to experience delays in triggering the ISR. 

There are mainly two reasons of the delay (assuming GPIO interrupt priority is already the highest among all in-use CPU interrupts):

1. Critical sections that are protected with the spinlock. To eliminate this types of delay, you need to use a LEVEL4 or above interrupt.

2. When the system processes the interrupts, the global interrupt will be disabled for some period. To shorten this delay to a minimal, you need a non-maskable interrupt (NMI LEVEL).

So your impression is correct. To get a more precise interrupt, you need at least a LEVEL4 interrupt, which means you need to write the interrupt handler in assembly for ESP32S2.

ultrapalace, 2022-04-30:
@songruo yes it involves many many drivers you are right, USB, WIFI and SPI on top of those you named. 
I will try to write my 4xUART driver ISR in ASM, thank you so much for the clear response <3

closed 2022-04-30 by ultrapalace