espressif/esp-idf #8778: [ESP32-S3] [v4.4] bootloader_flash_read & bootloader_flash_write don't work?

chipweinberger, 2022-04-14:
ESP32-S3

ESP-IDF release/v4.4 (https://github.com/espressif/esp-idf/commit/c29343eb94d2f2ca17b3a5b38c82452e556147f2) from a few days ago.

I have some very simple code in my bootloader, trying to write json into a partition at a known offset. 

 I don't know if reading or writing is broken. **But round tripping does not work as expected.** 

Code:
```
    ESP_LOGI(TAG, "write %u%s", jsonLen, json);

    // write json length
    err = bootloader_flash_write(pos.offset, &jsonLen, sizeof(uint32_t), false);
    if (err != ESP_OK){
        ESP_LOGE(TAG, "bootloader_flash_write() length failed");
        return;
    }

    // write json
    err = bootloader_flash_write(pos.offset+sizeof(uint32_t), json, sizeof(json), false);
    if (err != ESP_OK){
        ESP_LOGE(TAG, "bootloader_flash_write() json failed");
        return;
    }

    // read json length
    err = bootloader_flash_read(pos.offset, &jsonLen, sizeof(uint32_t), false);
    if (err != ESP_OK){
        ESP_LOGE(TAG, "bootloader_flash_read() json failed");
        return;
    }

    // read json
    err = bootloader_flash_read(pos.offset + sizeof(uint32_t), json, sizeof(json), false);
    if (err != ESP_OK){
        ESP_LOGE(TAG, "bootloader_flash_read() json failed");
        return;
    }

    ESP_LOGI(TAG, "read %u%s", jsonLen, json);
```

Logs: 
```
I (18710) boot cmd: write 31{"developerMode":0,"cmds":[1]}
I (18715) boot cmd: read 0" $D
```

igrr, 2022-04-14:
Question, are you erasing flash sector before writing to it? Write operation for NOR flash can only set bits to zero. If you need to replace the contents of a sector with new data, please erase the sector first (which sets all bits to 1), then perform write operation.

The erase operation is a separate API from write operation because erase can only be done at sector (4kB) granularity, while writing into already erased space can be done without size or alignment restrictions.

chipweinberger, 2022-04-14:
Thanks for the fast response Ivan! 

That is most certainly the issue! I'll verify & close if it works. 

Even more stupid, I've hit this exact issue before! Time to get sleep!

chipweinberger, 2022-04-14:
As a small suggestion, maybe the header  **bootloader_flash_priv.h** can be amended to remind us forgetful developers of the need to erase?

```
/**
 * @brief  Write data to Flash.
 *
 * @note All of dest_addr, src and size have to be 4-byte aligned. If write_encrypted is set, dest_addr and size must be 32-byte aligned.
 *
 * Note: In bootloader, when write_encrypted == true, the src buffer is encrypted in place.
 *
 * @param  dest_addr Destination address to write in Flash.
 * @param  src Pointer to the data to write to flash
 * @param  size Length of data in bytes.
 * @param  write_encrypted If true, data will be written encrypted on flash.
 *
 * @return ESP_OK on success, ESP_ERR_FLASH_OP_FAIL on SPI failure,
 * ESP_ERR_FLASH_OP_TIMEOUT on SPI timeout.
 */
```

chipweinberger, 2022-04-15:
yes this fixed it.

closed 2022-04-15 by chipweinberger