espressif/esp-idf #8780: Crashed task handle: 0x3ffbbbf4, name: 'esp_timer' case InterruptWDTTimoutCPU0

sxsx2yzyz, 2022-04-15:
## Environment

- Module or chip used:  [ESP32-WROVER-IE]
- IDF version (run ``git describe --tags`` to find it):
    v5.0-dev-422-ga18cb70905-dirty
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    8.4.0
- Operating System:     [macOS]
- Using an IDE?: [vscode]
- Power Supply:         [external 3.3V]

## Problem Description
The program ran for about 30 hours and an InterruptWDTTimoutCPU0 error occurred

coredump:
`espcoredump.py v0.4-dev
===============================================================
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3ffbbbf4, name: 'esp_timer', GDB name: 'process 1073462260'

================== CURRENT THREAD REGISTERS ===================
exccause       0x45 (InterruptWDTTimoutCPU0)
excvaddr       0x0
epc1           0x0
epc2           0x0
epc3           0x0
epc4           0x0
epc5           0x0
epc6           0x0
epc7           0x0
eps2           0x0
eps3           0x0
eps4           0x0
eps5           0x0
eps6           0x0
eps7           0x0
pc             0x4009903f          0x4009903f <xPortEnterCriticalTimeout+67>
lbeg           0x400843d5          1074283477
lend           0x400843dd          1074283485
lcount         0x27                39
sar            0x1a                26
ps             0x60625             394789
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x8008408d          -2146942835
a1             0x3ffbbaf0          1073462000
a2             0x3ffbe388          1073472392
a3             0xb33fffff          -1287651329
a4             0xabab              43947
a5             0x60623             394787
a6             0x60620             394784
a7             0xcdcd              52685
a8             0xabab              43947
a9             0xffffffff          -1
a10            0x3                 3
[... 203 lines cut ...]
.coredump.tasks.data 0x3ffbaa94 0x158 RW 
.coredump.tasks.data 0x3ffba8e0 0x1ac RW 
.coredump.tasks.data 0x3ffe6d8c 0x158 RW 
.coredump.tasks.data 0x3ffe6b50 0x234 RW 
.coredump.tasks.data 0x3ffe894c 0x158 RW 
.coredump.tasks.data 0x3ffe8750 0x1f4 RW 
.coredump.tasks.data 0x3ffd9a78 0x158 RW 
.coredump.tasks.data 0x3ffd98b0 0x1c0 RW 
.coredump.tasks.data 0x3ffdf090 0x158 RW 
.coredump.tasks.data 0x3ffdeec0 0x1c8 RW 
.coredump.tasks.data 0x3ffb61bc 0x158 RW 
.coredump.tasks.data 0x3ffdc680 0x1c0 RW 
.coredump.tasks.data 0x3ffb7d54 0x158 RW 
.coredump.tasks.data 0x3ffba4e0 0x1a8 RW 
.coredump.tasks.data 0x3fff7520 0x158 RW 
.coredump.tasks.data 0x3fff7200 0x318 RW 
.coredump.tasks.data 0x3ffbd87c 0x158 RW 
.coredump.tasks.data 0x3ffd7870 0x1dc RW 
.coredump.tasks.data 0x3ffbd720 0x158 RW 
.coredump.tasks.data 0x3ffd38a0 0x1a8 RW 
.coredump.tasks.data 0x3ffb5a9c 0x158 RW 
.coredump.tasks.data 0x3ffd8880 0x1d0 RW 

===================== ESP32 CORE DUMP END =====================
===============================================================`

o-marshmallow, 2022-04-19:
Hi @sxsx2yzyz,

I see that Thread 1 is looping indefinetely, triggering the watchdog which means it was unable to take the `s_timeout_lock` spinlock. Most of the tasks where waiting, apart from that hard to tell what is going on or what is the  issue.

Does this always happen after 30 hours? What is your application exaclty doing? I know it's hard but can you try providing a small example that reproduces the issue? Or have your tried reproducing it?

sxsx2yzyz, 2022-04-23:
> Hi @sxsx2yzyz,
>
> I see that Thread 1 is looping indefinetely, triggering the watchdog which means it was unable to take the `s_timeout_lock` spinlock. Most of the tasks where waiting, apart from that [...]

I had the same problem again and this time after the program has been running for about 33.5 hours.I have 2 pieces of esp32wrover-E and 1 piece of esp32wrover-IE, the problem is always esp32wrover-IE.
This is the part of my code for esp_timer:
```
static xQueueHandle frameQueue = NULL;
static void frameGenerateTask()
{
    xQueueSend(frameQueue, NULL, NULL);
}
void xx()
{
...
    frameQueue = xQueueCreate(400, 0); 
    const esp_timer_create_args_t periodic_timer_args = {
        .callback = &frameGenerateTask,
        /* name is optional, but may help identify the timer when debugging */
        .name = "frame"};

    esp_timer_handle_t periodic_timer;
    ESP_ERROR_CHECK(esp_timer_create(&periodic_timer_args, &periodic_timer));
    ESP_ERROR_CHECK(esp_timer_start_periodic(periodic_timer, 25 * 1000));
...
}
```
I tried to analyse coredump, but the program was recompiled and app.elf has been changed. Here is output of run `espcoredump.py info_corefile -t raw -c . /coredump . /build/app.elf`.
```
espcoredump.py v0.4-dev
===============================================================
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3ffbbbf4, name: 'esp_timer', GDB name: 'process 1073462260'

================== CURRENT THREAD REGISTERS ===================
exccause       0x45 (InterruptWDTTimoutCPU0)
excvaddr       0x0
epc1           0x0
epc2           0x0
epc3           0x0
epc4           0x0
epc5           0x0
epc6           0x0
epc7           0x0
eps2           0x0
eps3           0x0
eps4           0x0
eps5           0x0
eps6           0x0
eps7           0x0
pc             0x4009903f          0x4009903f <spi_flash_hal_erase_block+75>
lbeg           0x400843d5          1074283477
lend           0x400843dd          1074283485
lcount         0x27                39
sar            0x1f                31
ps             0x60625             394789
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
[... 92 lines cut ...]
#0  0x4000bff0 in ?? ()
#1  0x400991d8 in spi_flash_ll_set_hold (hold_n=0, dev=0x800991d8) at /Users/yangzhen/esp/esp-idf/components/hal/esp32/include/hal/spi_flash_ll.h:403
#2  spi_flash_hal_device_config (host=0x3ffc4778 <s_correction_us>) at /Users/yangzhen/esp/esp-idf/components/hal/spi_flash_hal_common.inc:52
#3  0x400988e5 in cpu_ll_set_breakpoint (pc=1, id=1073639052) at /Users/yangzhen/esp/esp-idf/components/hal/esp32/include/hal/cpu_ll.h:77
#4  cpu_hal_set_breakpoint (/builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/gdb/gdb/inline-frame.c:155: internal-error: void inline_frame_this_id(frame_info*, void**, frame_id*): Assertion `frame_id_p (*this_id)' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Quit this debugging session? (y or n) [answered Y; input not from terminal]
/builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/gdb/gdb/inline-frame.c:155: internal-error: void inline_frame_this_id(frame_info*, void**, frame_id*): Assertion `frame_id_p (*this_id)' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Create a core file of GDB? (y or n) [answered Y; input not from terminal]
Traceback (most recent call last):
  File "/Users/yangzhen/esp/esp-idf/components/espcoredump/espcoredump.py", line 350, in <module>
    temp_core_files = info_corefile()
  File "/Users/yangzhen/esp/esp-idf/components/espcoredump/espcoredump.py", line 206, in info_corefile
    gdb.switch_thread(thr_id)
  File "/Users/yangzhen/esp/esp-idf/components/espcoredump/corefile/gdb.py", line 120, in switch_thread
    self._gdbmi_run_cmd_get_one_response('-thread-select %s' % thr_id, 'done', 'result')
  File "/Users/yangzhen/esp/esp-idf/components/espcoredump/corefile/gdb.py", line 79, in _gdbmi_run_cmd_get_one_response
    return self._gdbmi_run_cmd_get_responses(cmd, resp_message, resp_type, multiple=False)[0]
  File "/Users/yangzhen/esp/esp-idf/components/espcoredump/corefile/gdb.py", line 72, in _gdbmi_run_cmd_get_responses
    raise ESPCoreDumpError("Couldn't find response with message '{}', type '{}' in responses '{}'".format(
corefile.ESPCoreDumpError: Couldn't find response with message 'done', type 'result' in responses '[]'
```

sxsx2yzyz, 2022-04-24:
The same thing happened again, this time with esp32-wrover-E, 67 hours after the chip was booted, and here are the results of the coredump analysis.
```
espcoredump.py v0.4-dev
===============================================================
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3ffbbbf4, name: 'esp_timer', GDB name: 'process 1073462260'

================== CURRENT THREAD REGISTERS ===================
exccause       0x45 (InterruptWDTTimoutCPU0)
excvaddr       0x0
epc1           0x0
epc2           0x0
epc3           0x0
epc4           0x0
epc5           0x0
epc6           0x0
epc7           0x0
eps2           0x0
eps3           0x0
eps4           0x0
eps5           0x0
eps6           0x0
eps7           0x0
pc             0x40099037          0x40099037 <spi_flash_hal_erase_block+67>
lbeg           0x400843d5          1074283477
lend           0x400843dd          1074283485
lcount         0x26                38
sar            0x1f                31
ps             0x60625             394789
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x8008408d          -2146942835
a1             0x3ffbbaf0          1073462000
a2             0x3ffbe388          1073472392
a3             0xb33fffff          -1287651329
a4             0xabab              43947
a5             0x60623             394787
a6             0x60620             394784
a7             0xcdcd              52685
a8             0xabab              43947
a9             0xffffffff          -1
a10            0x3                 3
a11            0x60623             394787
a12            0x60620             394784
a13            0x3ffc47b4          1073498036
a14            0x7be388            8119176
a15            0x3fffff            4194303

==================== CURRENT THREAD STACK =====================
#0  0x40099037 in spi_flash_ll_erase_block (dev=0xabab) at /Users/yangzhen/esp/esp-idf/components/hal/esp32/include/hal/spi_flash_ll.h:119
#1  spi_flash_hal_erase_block (host=0x3ffbe388 <s_timer_lock>, start_address=<optimized out>) at /Users/yangzhen/esp/esp-idf/components/hal/spi_flash_hal_iram.c:61
#2  0x4008408d in timer_insert (timer=0x0, without_update_alarm=<optimized out>) at /Users/yangzhen/esp/esp-idf/components/esp_timer/src/esp_timer.c:246
#3  0x400d6ba6 in timer_process_alarm (dispatch_method=ESP_TIMER_TASK) at /Users/yangzhen/esp/esp-idf/components/esp_timer/src/esp_timer.c:341
#4  timer_task (arg=<optimized out>) at /Users/yangzhen/esp/esp-idf/components/esp_timer/src/esp_timer.c:388

======================== THREADS INFO =========================
[... 67 lines cut ...]
#0  0x4000bff0 in ?? ()
#1  0x400991d8 in spi_flash_ll_set_hold (hold_n=0, dev=0x800991d8) at /Users/yangzhen/esp/esp-idf/components/hal/esp32/include/hal/spi_flash_ll.h:403
#2  spi_flash_hal_device_config (host=0x3ffc4778 <s_correction_us>) at /Users/yangzhen/esp/esp-idf/components/hal/spi_flash_hal_common.inc:52
#3  0x400988e5 in cpu_ll_set_breakpoint (pc=1, id=1073639052) at /Users/yangzhen/esp/esp-idf/components/hal/esp32/include/hal/cpu_ll.h:77
#4  cpu_hal_set_breakpoint (/builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/gdb/gdb/inline-frame.c:155: internal-error: void inline_frame_this_id(frame_info*, void**, frame_id*): Assertion `frame_id_p (*this_id)' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Quit this debugging session? (y or n) [answered Y; input not from terminal]
/builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/gdb/gdb/inline-frame.c:155: internal-error: void inline_frame_this_id(frame_info*, void**, frame_id*): Assertion `frame_id_p (*this_id)' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Create a core file of GDB? (y or n) [answered Y; input not from terminal]
Traceback (most recent call last):
  File "/Users/yangzhen/esp/esp-idf/components/espcoredump/espcoredump.py", line 350, in <module>
    temp_core_files = info_corefile()
  File "/Users/yangzhen/esp/esp-idf/components/espcoredump/espcoredump.py", line 206, in info_corefile
    gdb.switch_thread(thr_id)
  File "/Users/yangzhen/esp/esp-idf/components/espcoredump/corefile/gdb.py", line 120, in switch_thread
    self._gdbmi_run_cmd_get_one_response('-thread-select %s' % thr_id, 'done', 'result')
  File "/Users/yangzhen/esp/esp-idf/components/espcoredump/corefile/gdb.py", line 79, in _gdbmi_run_cmd_get_one_response
    return self._gdbmi_run_cmd_get_responses(cmd, resp_message, resp_type, multiple=False)[0]
  File "/Users/yangzhen/esp/esp-idf/components/espcoredump/corefile/gdb.py", line 72, in _gdbmi_run_cmd_get_responses
    raise ESPCoreDumpError("Couldn't find response with message '{}', type '{}' in responses '{}'".format(
corefile.ESPCoreDumpError: Couldn't find response with message 'done', type 'result' in responses '[]'
```

sxsx2yzyz, 2022-04-30:
> and

I enabled too.Now i'm trying not to use esp_timer.

sxsx2yzyz, 2022-05-06:
I no longer use anything related to esp_timer, e.g. LEDC_LOW_SPEED_MODE for ledc, periodic_timer for esp_timer. no longer have this problem.

sxsx2yzyz, 2022-05-13:
Again 
```
espcoredump.py v0.4-dev
===============================================================
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3ffbbbf4, name: 'esp_timer', GDB name: 'process 1073462260'

================== CURRENT THREAD REGISTERS ===================
exccause       0x45 (InterruptWDTTimoutCPU0)
excvaddr       0x0
epc1           0x0
epc2           0x0
epc3           0x0
epc4           0x0
epc5           0x0
epc6           0x0
epc7           0x0
eps2           0x0
eps3           0x0
eps4           0x0
eps5           0x0
eps6           0x0
eps7           0x0
pc             0x400994e7          0x400994e7 <xPortEnterCriticalTimeout+59>
lbeg           0x40093014          1074343956
lend           0x40093030          1074343984
lcount         0xffffffff          4294967295
sar            0x17                23
ps             0x60025             393253
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x800842a1          -2146942303
a1             0x3ffbbad0          1073461968
a2             0x3ffbe388          1073472392
a3             0xb33fffff          -1287651329
a4             0xabab              43947
a5             0x60023             393251
a6             0x60020             393248
a7             0xcdcd              52685
a8             0xabab              43947
a9             0xffffffff          -1
a10            0x3                 3
a11            0x60023             393251
a12            0x60020             393248
a13            0x3f81e584          1065477508
a14            0x7be388            8119176
a15            0x3fffff            4194303

==================== CURRENT THREAD STACK =====================
#0  compare_and_set_native (set=0x3ffbbad0, compare=3007315967, addr=0x3ffbe388 <s_timer_lock>) at /Users/yangzhen/esp/esp-idf/components/esp_hw_support/include/soc/compare_set.h:25
#1  spinlock_acquire (timeout=<optimized out>, lock=0x3ffbe388 <s_timer_lock>) at /Users/yangzhen/esp/esp-idf/components/esp_hw_support/include/soc/spinlock.h:103
#2  xPortEnterCriticalTimeout (mux=0x3ffbe388 <s_timer_lock>, timeout=<optimized out>) at /Users/yangzhen/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:287
#3  0x400842a1 in xPortEnterCriticalTimeoutSafe (timeout=-1, mux=<optimized out>) at /Users/yangzhen/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:592
#4  vPortEnterCriticalSafe (mux=<optimized out>) at /Users/yangzhen/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:599
#5  timer_list_lock (timer_type=<optimized out>) at /Users/yangzhen/esp/esp-idf/components/esp_timer/src/esp_timer.c:307
#6  0x40084422 in timer_process_alarm (dispatch_method=ESP_TIMER_TASK) at /Users/yangzhen/esp/esp-idf/components/esp_timer/src/esp_timer.c:363
[... 190 lines cut ...]
.coredump.tasks.data 0x3ffba8d0 0x1bc RW 
.coredump.tasks.data 0x3ffdfc70 0x158 RW 
.coredump.tasks.data 0x3ffdfaa0 0x1c8 RW 
.coredump.tasks.data 0x3ffe762c 0x158 RW 
.coredump.tasks.data 0x3ffe7460 0x1c4 RW 
.coredump.tasks.data 0x3ffb60e4 0x158 RW 
.coredump.tasks.data 0x3ffdea00 0x1c4 RW 
.coredump.tasks.data 0x3ffe878c 0x158 RW 
.coredump.tasks.data 0x3ffe8560 0x224 RW 
.coredump.tasks.data 0x3ffea34c 0x158 RW 
.coredump.tasks.data 0x3ffea110 0x234 RW 
.coredump.tasks.data 0x3ffb7cd0 0x158 RW 
.coredump.tasks.data 0x3ffba4b0 0x1d8 RW 
.coredump.tasks.data 0x3fff8f20 0x158 RW 
.coredump.tasks.data 0x3fff8bf0 0x328 RW 
.coredump.tasks.data 0x3ffda6e0 0x158 RW 
.coredump.tasks.data 0x3ffda500 0x1d8 RW 
.coredump.tasks.data 0x3ffd6580 0x158 RW 
.coredump.tasks.data 0x3ffd5fa0 0x5d8 RW 
.coredump.tasks.data 0x3ffb5a9c 0x158 RW 
.coredump.tasks.data 0x3ffdc7f0 0x1d0 RW 

===================== ESP32 CORE DUMP END =====================
===============================================================
```

closed 2022-10-21 by sxsx2yzyz