espressif/esp-idf #8781: 明明包含了APP MAIN为什么还是报错了呢？REALLY DEFINE APP MAIN API，WHY FAILED

xishing, 2022-04-15:
H:\esp32\examples\20220331_32_voice_X>idf.py build
Executing action: all (aliases: build)
Running ninja in directory h:\esp32\examples\20220331_32_voice_x\build
Executing "ninja all"...
[1/5] Performing build step for 'bootloader'
ninja: no work to do.
[2/3] Linking CXX executable 20220331_C3_voice_X.elf
FAILED: 20220331_C3_voice_X.elf 
cmd.exe /C "cd . && H:\.espressif\tools\xtensa-esp32-elf\esp-2021r1-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address   @CMakeFiles\20220331_C3_voice_X.elf.rsp  -o 20220331_C3_voice_X.elf  && cd ."
h:/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/freertos/libfreertos.a(port_common.c.obj):(.literal.main_task+0x0): undefined reference to `app_main'
h:/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/freertos/libfreertos.a(port_common.c.obj): in function `main_task':   
H:/esp32/components/freertos/port/port_common.c:98: undefined reference to `app_main'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

Alvin1Zhang, 2022-04-15:
Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

o-marshmallow, 2022-04-18:
Hi @xishing,

Can you give more details about how your structured your project? Is it open-source? Ca you share it to let us have a look at the problem?

Dazza0, 2022-05-13:
@xishing Please double check that your project's `CMakeLists.txt` is pulling in source file containing your `app_main()` definition. You can check `./build/compile_commands.json` to see the your source file containing `app_main()` is being built in the first place.

Please feel free to reopen this issue if are still facing the same error.

closed 2022-05-13 by Dazza0