espressif/esp-idf #8786: I2C "i2c_master_cmd_begin" gives error 0x107 ESP_ERR_TIMEOUT

AAl-Hadi, 2022-04-17:
I had a project working fine. But I had to change my PC and had to reinstall everything again and pulled my project from Github.
I'm using ESP-IDF plugin with eclipse. The project was working fine before but now all I2C sensors I have are not working at all
I get timeout error when reaching "i2c_master_cmd_begin"

I used an Arduino I2C scanner and was working fine
I'm sure it's not a hw problem 

This is the code that was working fine and now is not working
`	
    i2c_cmd_handle_t cmd;

    cmd = i2c_cmd_link_create();
    ESP_ERROR_CHECK(i2c_master_start(cmd));
    ESP_ERROR_CHECK(i2c_master_write_byte(cmd, SCD30_I2C_ADDRESS|I2C_WRITE_BIT, I2C_MASTER_NACK));
    ESP_ERROR_CHECK(i2c_master_write_byte(cmd, address >> 8, I2C_MASTER_NACK));
    ESP_ERROR_CHECK(i2c_master_write_byte(cmd, address & 0xFF, I2C_MASTER_NACK));
    ESP_ERROR_CHECK(i2c_master_stop(cmd));
    ESP_ERROR_CHECK(i2c_master_cmd_begin(I2C_MASTER_NUM, cmd, 1000/portTICK_PERIOD_MS));
    i2c_cmd_link_delete(cmd);

    cmd = i2c_cmd_link_create();
    ESP_ERROR_CHECK(i2c_master_start(cmd));
    ESP_ERROR_CHECK(i2c_master_write_byte(cmd, SCD30_I2C_ADDRESS|I2C_READ_BIT, I2C_MASTER_NACK));

    ESP_ERROR_CHECK(i2c_master_read(cmd, data, size, 2));
    ESP_ERROR_CHECK(i2c_master_stop(cmd));
    ESP_ERROR_CHECK(i2c_master_cmd_begin(I2C_MASTER_NUM, cmd, 1000/portTICK_PERIOD_MS));
    i2c_cmd_link_delete(cmd);`

AxelLin, 2022-04-18:
1. Which esp-idf version was working, which esp-idf version is broken?
2. Which module are you using?

0xjakob, 2022-04-18:
@AAl-Hadi Have you checked the configuration of the I2C driver? Did you make sure that the I2C pins and also the I2C address didn't accidentally change somehow?

AAl-Hadi, 2022-04-18:
> 

1. Working : ESP-IDF v4.3.1-dirty. Failed: ESP-IDF v4.4-dirty
2. ESP32S2 on a custom board

AAl-Hadi, 2022-04-18:
@0xjakob I checked. Everything is fine. I still have the old binary and that is working fine. I also tried an Arduino scanner and that is working fine. I double checked the I2C configuration everything seems fine

o-marshmallow, 2022-04-18:
Hi @AAl-Hadi ,

I guess you are using the SCD30 temperature/humidity/CO2 sensor, if that's the case, can you please double-check your device address, or macro, `SCD30_I2C_ADDRESS`. the value must be `0xC2` in your case.
 
Else, do you have any logic analyzer or oscilloscope to probe the bus and check what is going on?

AAl-Hadi, 2022-04-18:
@o-marshmallow That's not the problem. since the fw was already working before. Yes the address is shifted.
I found what the problem is. It's from the SCD30 sensor itself. I have 4 I2C sensors soldered on board. After a lot of debugging I found out that the SCD30 is doing something to the bus probably. Once it's powered ON, I2C won't work anymore. The solution is to power it ON before enabling I2C bus. I don't know why this is happening and why was it no happening before. I still have the old binary and that is working fine. But the same FW doens't work when I flash it unless I power ON the sensor before enabling the I2C. I'd say the problem is solved but I'd appreciate more insights on why this happens!

Thank you all who posted

o-marshmallow, 2022-04-18:
@AAl-Hadi That's really odd. Devices are allowed to pull the SCL line low to perform clock stretching but this usually happens during a transfer, not on startup. I think in your case a logic analyzer would really help to debug the bus.

How did you configure the I2C bus in your application? Did you enabled internal pull-up resistors? 
Can you try temporarily connecting 4.7k pull-up resistors on SCL and SDA to check if that changes anything? (the internal pull-up have much higher value)

AAl-Hadi, 2022-04-19:
@o-marshmallow I remembered today that I had this problem before. I used this particular sensor with an ST32 MCU and was doing the same. I used a logic analyzer at the time and found out the sensor sends a noise signal once it's powered ON. Probably is the same right now. I don't have a logic analyzer to confirm. I still don't know why it started to happen after I changed my machine. I still have the old binary that powers ON the sensor after enabling the I2C and is still working fine. But the same FW on my new machine is not working! I know it's the exact same FW because I pulled it from GitHub with no changes.

closed 2022-04-19 by AAl-Hadi

o-marshmallow, 2022-04-19:
@AAl-Hadi Thanks for coming back with a partial solution :+1: