espressif/esp-idf #8787: CMake Error: Could not create symbolic link for...

xiaguangbo, 2022-04-17:
env：
esp-idf master

---

> Executing task: cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 .. <

-- Found Git: C:/Users/xiaguangbo/esp/.espressif/tools/idf-git/2.30.1/cmd/git.exe (found version "2.30.1.windows.1")
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/xiaguangbo/esp/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/xiaguangbo/esp/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Users/xiaguangbo/esp/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s2
-- Project sdkconfig file D:/xiaguangbo/project/esp/project/template-app/sdkconfig
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Found Python3: C:/Users/xiaguangbo/esp/.espressif/python_env/idf5.0_py3.8_env/Scripts/python.exe (found version "3.8.7") found components: Interpreter
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
CMake Error at C:/Users/xiaguangbo/esp/esp-idf/components/mbedtls/mbedtls/CMakeLists.txt:148 (message):
  Could not create symbolic link for:
  C:\Users\xiaguangbo\esp\esp-idf\components\mbedtls\mbedtls\library\error.c
  --> 系统无法将文件移到不同的磁盘驱动器。

Call Stack (most recent call first):
  C:/Users/xiaguangbo/esp/esp-idf/components/mbedtls/mbedtls/library/CMakeLists.txt:162 (link_to_source)

-- Configuring incomplete, errors occurred!
See also "D:/xiaguangbo/project/esp/project/template-app/build/CMakeFiles/CMakeOutput.log".
终端进程“C:\windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 ..”已终止，退出代码: 1。

Alvin1Zhang, 2022-04-18:
Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

ESP-Marius, 2022-04-18:
Hi @xiaguangbo, this seems to be an issue caused upstream by the mbedtls library. It tries to create a symlink between a file in your IDF folder (error.c) to your build folder, which are on two different drives (C: and D:) which Windows seems to be unable to do.

This is is something they changed recently, so building a IDF project on a different drive from the SDK works on release/v4.4 and older, but not on the latest master....

We'll look into this and see if we can find a fix. For now you can work around this by moving your project `D:/xiaguangbo/project/esp/` to the `C:` drive.

xiaguangbo, 2022-04-18:
@ESP-Marius That's exactly the problem, thanks

mahavirj, 2022-05-05:
Linking https://github.com/Mbed-TLS/mbedtls/issues/5751

xobs, 2022-07-14:
Hi, I've just run into this issue, which still seems to be a problem. Is this the only place it's documented that `esp-idf` must be installed on the C drive? Is there any way `esp-idf` could be fixed?

As a workaround, I've changed `set(command cmd.exe /c mklink /h ${link} ${target})` to `set(command cmd.exe /c copy ${target} ${link})` in `components/mbedtls/mbedtls/CMakeLists.txt:139`.

Either way, this is still a problem and I don't think this issue should be closed, mostly because people will keep running into it and opening duplicates.

xiaguangbo, 2022-07-14:
@xobs I reopened this question. This problem is described in 'Linking https://github.com/Mbed-TLS/mbedtls/issues/5751' mentioned by mahavirj

kolipakakondal, 2022-07-26:
This problem still exists on the master branch and v5.0. Please find the attached log. 
[console.txt](https://github.com/espressif/esp-idf/files/9187879/console.txt)
[CMakeOutput.log](https://github.com/espressif/esp-idf/files/9187880/CMakeOutput.log)

Reported by Denys

closed 2022-08-26 by espressif-bot

texruska, 2023-11-10:
This issue can still impact Linux users. I'm using Docker and fixed it with:

components/mbedtls/mbedtls/CMakeLists.txt:140
```
if (CMAKE_HOST_UNIX)
-            execute_process(COMMAND ln -s ${target} ${link}
+            execute_process(COMMAND cp --remove-destination ${target} ${link}
                 RESULT_VARIABLE result
                 ERROR_VARIABLE output)
```

supiiik, 2024-01-14:
> This issue can still impact Linux users. I'm using Docker with a mounted volume. I fixed it with:
> 
> components/mbedtls/mbedtls/CMakeLists.txt:140
> 
> ```
> if (CMAKE_HOST_UNIX)
> -            execute_process(COMMAND ln -s ${target} ${link}
> +            execute_process(COMMAND cp --remove-destination ${target} ${link}
>                  RESULT_VARIABLE result
>                  ERROR_VARIABLE output)
> ```

Fixed my problem too. Thanks

try-except, 2025-01-16:
> This issue can still impact Linux users. I'm using Docker with a mounted volume. I fixed it with:
>
> components/mbedtls/mbedtls/CMakeLists.txt:140
>
> ```
> if (CMAKE_HOST_UNIX)
> - [...]

This worked perfectly! thanks a lot