espressif/esp-idf #8789: ESP32-S3 USB Serial/JTAG controller does not support console/linenoise editing

davidzuhn, 2022-04-17:
## Environment

- Development Kit:   ESP32-S3-DevKitC-1
- Kit version (for WroverKit/PicoKit/DevKitC): v1.0
- Module or chip used:  ESP32-S3-WROOM-1
- IDF version (run ``git describe --tags`` to find it): v5.0-dev-2483-g3aeb80acb6
- Build System: idf.py
- Compiler version: xtensa-esp32s3-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System: macOS 12.3.1, Raspberry Pi OS (buster)
- Using an IDE?: No
- Power Supply: USB

## Problem Description

The program in examples/system/console/advanced, unmodified, works perfectly well when using the CP2102N device (/dev/cu.SLAB_USBtoUART) and does not work when using the USB Serial/JTAG controller port (/dev/cu.usbmodem1101).   

Using 'idf.py monitor -p /dev/cu.usbmodel1101', I see the expected boot messages and console output from the program, but then I get an error message:

```
This is an example of ESP-IDF console component.
Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.
Press Enter or Ctrl+C will terminate the console environment.

Your terminal application does not support escape sequences.
Line editing and history features are disabled.
On Windows, try using Putty instead.
```

and then at this point no input is processed (so the console doesn't function at all, instead of falling back to no editing or history).   

### Expected Behavior

The interactive console should work the same when using either the UART or the USB Serial/JTAG controller.

### Actual Behavior

The UART works fine, while the USB Serial controller does not (no prompt, no input)

### Steps to reproduce

1.  checkout & install esp-idf & run export.sh as usual
2. cd examples/system/console/advanced (unmodified)
3. idf.py set-target esp32s3
4. idf.py build 
6. connect to the UART port, and run 'idf.py flash monitor -p /dev/cu.SLAB_USBtoUART'
7. the application should work just fine, allowing commands to be entered (such as 'help' or 'tasks').  command line editing works just fine
8. disconnect from the UART port and connect to the USB port and run 'idf.py flash monitor -p /dev/cu.usbmodem1101'
9. at this point, the error message
```
 Your terminal application does not support escape sequences.
Line editing and history features are disabled.
On Windows, try using Putty instead.
```
is displayed after the boot message and the introduction "This is an example of ESP-IDF console component [etc]", and no further input processing takes place.   Note that this is the same terminal application (and shell instance) where this just worked (see steps 6 & 7).   The only difference is the use of the USB Serial/JTAG controller instead of the CP2102N UART.   

Note that I have run the same process on a Linux build host (Raspberry Pi 4, Raspberry Pi OS based on Debian buster, aarch64) and the same basic results occur.   The UART port works fine on the console, while the USB port does not.  The only difference is that while on macOS ctrl-] will exit the monitor cleanly, that does not occur on the Pi and killing the monitor.py process was the only way to end that connection.   

### Code to reproduce this issue

esp-idf/examples/system/console/advanced   (unmodified)

## Debug Logs

davidzuhn, 2022-04-17:
Note that the examples/system/console/basic program exhibits the same characteristics, working on UART and failing on USB Serial/JTAG controller.

Spritetm, 2022-04-20:
Two things here:
- The default configuration is to have the UART as the main input/output device, with the USB-serial-JTAG converter as the 'secondary' (output-only) device. You can change this in menuconfig: set Component config → ESP System Settings → Channel for console output to USB Serial/JTAG Controller.
- With that, the example won't compile because there's no USB-serial-JTAG support in it at all... that needs to be added.

With that being said, it's not hard to rework the example to add support for this... attached is a quick & dirty modification that works here.
[console_example_main.c.gz](https://github.com/espressif/esp-idf/files/8518889/console_example_main.c.gz)

chipweinberger, 2022-04-22:
For those that stuble on this later

Required calls:
```
#include "driver/usb_serial_jtag.h"
#include "esp_vfs_usb_serial_jtag.h"

static void initialize_console(void) {
   ....
    esp_vfs_dev_usb_serial_jtag_set_rx_line_endings(ESP_LINE_ENDINGS_CR);
    esp_vfs_dev_usb_serial_jtag_set_tx_line_endings(ESP_LINE_ENDINGS_CRLF);
    const usb_serial_jtag_driver_config_t cfg=USB_SERIAL_JTAG_DRIVER_CONFIG_DEFAULT();
    usb_serial_jtag_driver_install(&cfg);

    esp_vfs_usb_serial_jtag_use_driver();
    ....
}
```

igrr, 2022-05-16:
Support for usb_serial_jtag on C3 and S3 has been added to the "basic" console example: https://github.com/espressif/esp-idf/commit/3390d2a2d18c289b0b1e1cf121bf6670099641db.

The "advanced" example remains UART-only for the time being.

peter-marcisovsky, 2024-09-08:
The console examples have been refactored by [this commit](https://github.com/espressif/esp-idf/commit/82a4c1281708ff9c974e775f02bd626b1733990d) so now it is possible to use both, the advanced console example and the basic console example, with all types of console peripherals, supported by a target chip.

Closing the issue, as it has been resolved.

closed 2024-09-08 by peter-marcisovsky