espressif/esp-idf #8790: A suggestion on LCD peripheral

LucivorLin, 2022-04-18:
**Is your feature request related to a problem? Please describe.**

A clear and concise description of what the problem is. Ex. I'm always frustrated when I use LCD peripherals.
SPI master settings cannot be configured in function `esp_lcd_new_panel_io_spi()`, but I want to change TX to LSB first, to ensure the color of my LCD is correct.

**Describe the solution you'd like**

in `/esp-idf/components/esp_lcd/src/sep_lcd_panel_io_spi.c`,
```c
spi_device_interface_config_t devcfg = {
        .flags = SPI_DEVICE_HALFDUPLEX,
```
The `flag` can be set as a parameter to ensure the flexibility of the SPI master device driving the screen.

**Describe alternatives you've considered**

Give some extra functions of API to change these settings.

**Chinese**

LCD外设生成的SPI主设备难以更改flag参数，但有屏幕修改LSB/MSB等一些需求。

Alvin1Zhang, 2022-04-18:
Thanks for reporting.

suda-morris, 2022-04-18:
@LucivorLin Yes, we didn't expose all the underlying spi device flags to LCD IO layer by design (because not all spi flags are suitable for LCD). But left the space to extend them in the future by adding more flags into `esp_lcd_panel_io_spi_config_t.flags`. 
Which flag in your case want to have? `SPI_DEVICE_TXBIT_LSBFIRST` or `SPI_DEVICE_RXBIT_LSBFIRST` or `SPI_DEVICE_BIT_LSBFIRST`?

LucivorLin, 2022-04-18:
@suda-morris  `SPI_DEVICE_TXBIT_LSBFIRST` is I want.

suda-morris, 2022-04-18:
@LucivorLin 
[spi-lcd-support-transmit-lsb-first.patch.zip](https://github.com/espressif/esp-idf/files/8505139/spi-lcd-support-transmit-lsb-first.patch.zip)

Is this patch sufficient for you? Sorry I don't have an LCD at hand that acquires an LSB first transmission. 

Ways to apply the patch:
1. decompress the zip file, and get the `.patch` file (e.g. my_fix.patch)
2. cd $IDF_PATH , check out the master branch
3. `git apply my_fix.patch` or `git am my_fix.patch`

LucivorLin, 2022-04-18:
After apply the patch, what should I do?
Since the `flags` of `esp_lcd_new_panel_io_spi` didn't change, I would like to ensure my operation is correct.
```git
xxx@LAPTOP-XXXXXXXX MINGW64 /e/Softwares/ESP/Container/esp-idf (master)
$ git apply temp/spi-lcd-support-transmit-lsb-first.patch
error: can't open patch 'temp/spi-lcd-support-transmit-lsb-first.patch': Permission denied

xxx@LAPTOP-XXXXXXXX MINGW64 /e/Softwares/ESP/Container/esp-idf (master)
$ git am temp/spi-lcd-support-transmit-lsb-first.patch

xxx@LAPTOP-XXXXXXXX MINGW64 /e/Softwares/ESP/Container/esp-idf (master)

```
Should there be any reaction of the latter command line? @suda-morris

suda-morris, 2022-04-19:
@LucivorLin  The patch file extracted from the attached zip file is *0001-lcd-spi-lcd-support-transmit-lsb-first.patch*. Maybe you didn't decompress it successfully?

LucivorLin, 2022-04-19:
Now I successfully applied, and I added `.flags.lsb_first = 1` in `esp_lcd_panel_io_spi_config_t io_config`.
However, since I am new to SEP-IDF master branch, I cannot build the project as before, the error shows as follow:
```
Traceback (most recent call last):
  File "E:/Softwares/ESP/Container/esp-idf/tools/ldgen/ldgen.py", line 16, in <module>
    from ldgen.entity import EntityDB
  File "E:\Softwares\ESP\Container\esp-idf\tools\ldgen\ldgen\entity.py", line 12, in <module>
    from pyparsing import (Group, Literal, OneOrMore, ParseException, SkipTo, Suppress, White, Word, ZeroOrMore, alphas,
ImportError: cannot import name 'rest_of_line' from 'pyparsing' (e:\Softwares\ESP\Tools\python_env\idf4.4_py3.8_env\lib\site-packages\pyparsing.py)
[2/8] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D E:\ESP32_Projects\S2_projects\lvgl\build\bootloader\esp-idf\esptool_py && e:\Softwares\ESP\Tools\python_env\idf4.4_py3.8_env\Scripts\python.exe E:/Softwares/ESP/Container/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 E:/ESP32_Projects/S2_projects/lvgl/build/bootloader/bootloader.bin"
Bootloader binary size 0x5100 bytes. 0x1f00 bytes (28%) free.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```
@suda-morris

suda-morris, 2022-04-19:
the pyparsing dependency of ldgen has been upgraded on master branch. I guess you need to rerun the "install.sh" script to install the python environment needed for master branch. Currently you're using the idf 4.4 environment for compiling the master branch (`e:\Softwares\ESP\Tools\python_env\idf4.4_py3.8_env`)

LucivorLin, 2022-04-19:
Well, I tried using the patch in v4.4.
However, the LCD cannot work, I think it may be because of the priority of the operator in C language.
Is that better to change the code from
```c
spi_device_interface_config_t devcfg = {
        .flags = SPI_DEVICE_HALFDUPLEX | io_config->flags.lsb_first ? SPI_DEVICE_TXBIT_LSBFIRST : 0,
```
to
```c
spi_device_interface_config_t devcfg = {
        .flags = SPI_DEVICE_HALFDUPLEX | (io_config->flags.lsb_first ? SPI_DEVICE_TXBIT_LSBFIRST : 0),
```
?
@suda-morris

suda-morris, 2022-04-19:
@LucivorLin  Ah yes, my bad. Nice found. So with your fix, does that work now?

LucivorLin, 2022-04-19:
It works properly now.

closed 2022-04-19 by LucivorLin