espressif/esp-idf #8791: ESP32-S3: When SPIRAM_RODATA is turned on, the program restarts infinitely

FanhuaCloud, 2022-04-18:
## Environment

- Development Kit:      none
- Kit version (for WroverKit/PicoKit/DevKitC): none
- Module or chip used:  ESP32-S3
- IDF version (run ``git describe --tags`` to find it):
    // v5.0-dev-2482-g3aeb80acb
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     Windows
- (Windows only) PowerShell.
- Using an IDE?: Yes 
- Power Supply:        USB

## Problem Description

 When SPIRAM_RODATA is turned on, the program restarts infinitely

### Steps to reproduce
1. Set the sdkconfig
CONFIG_SPIRAM_FETCH_INSTRUCTIONS=y
CONFIG_SPIRAM_RODATA=y

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
## Debug Logs
```
I (168) cpu_start: App cpu up.
I (1118) spiram: SPI SRAM memory test OK
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375930
0x40375930: esp_restart_noos at C:/Espressif/frameworks/esp-idf-master/components/esp_system/port/soc/esp32s3/system_internal.c:151 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0110,len:0x18d0
load:0x403b6000,len:0xb70
load:0x403b9000,len:0x31c0
SHA-256 comparison failed:
Calculated: ca559ef1d2e158b3d1b4153ea97c21ce955de778bf2c81a6f48870f27a881aaf
Expected: 630c78a2d85e4401b990d43d39aab574cd653a23c0a4bdc22e11fd562bf1666c
Attempting to boot anyway...
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

Icarus113, 2022-04-18:
@FanhuaCloud 
Hi, this feature isn't ready for S3 yet. Will update this issue when it's ready on Master branch.

FanhuaCloud, 2022-05-16:
> @FanhuaCloud Hi, this feature isn't ready for S3 yet. Will update this issue when it's ready on Master branch.

Hello, how is the development of this feature currently, I found that it doesn't seem to be able to compile yet

Icarus113, 2022-06-29:
@FanhuaCloud  Hi, this feature now is available.

FanhuaCloud, 2022-06-29:
> @FanhuaCloud Hi, this feature now is available.

Thank you，I am already use it.

closed 2022-06-29 by FanhuaCloud