espressif/esp-idf #8795: No explicit conversion to esp_tls_error_handle_t in esp_http_client example

dmitrij9992905, 2022-04-18:
No explicit conversion to esp_tls_error_handle_t in esp_http_client example
`esp_err_t err = esp_tls_get_and_clear_last_error(evt->data, &mbedtls_err, NULL);`

As a result, the example cannot be compiled

As a solution, I propose to add explicit data conversion:
`esp_err_t err = esp_tls_get_and_clear_last_error((esp_tls_error_handle_t)evt->data, &mbedtls_err, NULL);`

laukik-hase, 2022-04-19:
Hello, @dmitrij9992905!

The change proposed by you makes sense. But, could you please mention the ESP-IDF version you are using? The example compiles fine on the `master` branch.

dmitrij9992905, 2022-04-20:
The ESP-IDF version was 4.3.1

laukik-hase, 2022-04-20:
Hello, @dmitrij9992905!

The example compiles fine on ESP-IDF `v4.3.1`. Are you building the example with some custom `CFLAGS`?

closed 2022-04-24 by espressif-bot