espressif/esp-idf #8796: ESP32 C3  PDM I2S RX

mrburns42, 2022-04-19:
Issue #7604  seems to state that the ESP32 C3  does not support PDM I2S RX inputs.   However, when I look at the Technical Reference Manual,  it states that both PDM RX and TX are available.    When I build with PlatformIO with a simple test case, the code builds but then runs a continual boot loop.    In the serial port messages it states:

 E (120) I2S: i2s_check_cfg_validity(1490): PDM does not support RX on this chip

 I looked at the Technical Reference Manual
https://www.espressif.com/sites/default/files/documentation/esp32-c3_technical_reference_manual_en.pdf

It states " ESP32-C3 I2S supports both TDM RX mode and PDM RX mode. Set I2S_RX_TDM_EN to enable TDM RX
mode, or set I2S_RX_PDM_EN to enable PDM RX mode." [Section 27.9.1]

Am I referencing out of date documentation?   Or am I overlooking something else as to why this will not work?

L-KAYA, 2022-04-19:
As far as I know, C3 does not support PDM RX, it's probably a mistake in the Technical Reference Manual, the datasheet of C3 also said it only supports PDM TX (chapter 3.4.5) .

L-KAYA, 2025-01-13:
Update:

ESP32-C3 only supports raw PDM data format on the RX direction. Because it has no PDM2PCM converter, which means we need to read the raw PDM data and then convert it. The raw PDM data format was not supported until the commit [9b779d8b](https://github.com/espressif/esp-idf/commit/9b779d8b3c1c7d7aff73ebe5ef36d03ae3d60141)

closed 2025-01-13 by Alvin1Zhang