espressif/esp-idf #8798: idf.py build esp32c3 “SHA-256 comparison failed”

QGB, 2022-04-19:
```
./install.sh all
cd examples/get-started/hello_world/
idf.py set-target esp32c3
idf.py menuconfig
idf.py build

In [409]: !esptool --chip esp32c3 --port COM16 erase_flash
esptool.py v3.2
Serial port COM16
Connecting....
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 60:55:f9:72:d7:48
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 12.1s
Hard resetting via RTS pin...

In [410]: !esptool --chip esp32c3 --port COM16 --baud 460800 write_flash -z 0x0 C:/test/ESP32/esp32c3/hello_world.bin
esptool.py v3.2
Serial port COM16
Connecting....
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 60:55:f9:72:d7:48
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00024fff...
Compressed 151344 bytes to 81670...
Wrote 151344 bytes (81670 compressed) at 0x00000000 in 2.7 seconds (effective 450.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

```
serial port 16
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3c020020,len:0x6d30
load:0x3fc8a400,len:0x14e4
load:0x40380000,len:0x7dd4
load:0x42000020,len:0x12a3c
load:0x40387dd4,len:0x248c
load:0x50000010,len:0x10
SHA-256 comparison failed:
Calculated: 25bb723fd3e6316efb5d65670397578000b02737a534bb645f9f9686e2b3e404
Expected: 00d516aa87430f022d0ad43c476dbb101231ed5a6834298fdbd7f5dca6b560bc
Attempting to boot anyway...
entry 0x40380322
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4038013e
SPIWP:0xee
mode:DIO, clock div:1
load:0x3c020020,len:0x6d30
load:0x3fc8a400,len:0x14e4
load:0x40380000,len:0x7dd4
load:0x42000020,len:0x12a3c
load:0x40387dd4,len:0x248c
load:0x50000010,len:0x10
SHA-256 comparison failed:
Calculated: 25bb723fd3e6316efb5d65670397578000b02737a534bb645f9f9686e2b3e404
Expected: 00d516aa87430f022d0ad43c476dbb101231ed5a6834298fdbd7f5dca6b560bc
Attempting to boot anyway...
entry 0x40380322
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380112
SPIWP:0xee
mode:DIO, clock div:1
load:0x3c020020,len:0x6d30
load:0x3fc8a400,len:0x14e4
load:0x40380000,len:0x7dd4
load:0x42000020,len:0x12a3c
load:0x40387dd4,len:0x248c
load:0x50000010,len:0x10
SHA-256 comparison failed:
Calculated: 25bb723fd3e6316efb5d65670397578000b02737a534bb645f9f9686e2b3e404
Expected: 00d516aa87430f022d0ad43c476dbb101231ed5a6834298fdbd7f5dca6b560bc
Attempting to boot anyway...
```

o-marshmallow, 2022-04-20:
Hi @QGB,

Which version of IDF are you using? Can you have a try flashing your board with `idf.py flash` ?

ESP-Marius, 2022-04-20:
@QGB

`hello_world.bin` sounds like your app binary, but a normal IDF project also has a `bootloader.bin`, as well as a `partition-table.bin`. To successfully boot it is actually the bootloader that should be flashed to 0x0, not the app binary.

As @o-marshmallow mentioned, using `idf.py flash` is the easiest way, as it takes care of all of this for you. If you still want to flash directly with `esptool.py` then you can take a look at `build/flash_project_args` which contains the commands needed (all the binaries and their flash offsets).  See https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/build-system.html#flash-arguments for more information about this.

QGB, 2022-04-20:
```
(Top) → Serial flasher config
                                  Espressif IoT Development Framework Configuration
[ ] Disable download stub
    Flash SPI mode (DIO)  --->
    Flash Sampling Mode (STR Mode)  --->
    Flash SPI speed (80 MHz)  --->
    Flash size (4 MB)  --->
[*] Detect flash size when flashing bootloader
    Before flashing (Reset to bootloader)  --->
    After flashing (Reset after flashing)  --->

entry 0x40380322
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3c020020,len:0x6d30
load:0x3fc8a400,len:0x14e4
load:0x40380000,len:0x7dd4
load:0x42000020,len:0x12a3c
load:0x40387dd4,len:0x248c
load:0x50000010,len:0x10
SHA-256 comparison failed:
Calculated: 15cb229b05ffc7246d7ffefe43c7aaceeda906fef214c27e56fac2068d3872b4
Expected: 269d8c8dcd3611ae87ee33e822d463d9bfcd38beb9460e8a8164c2efe9755cfe
Attempting to boot anyway...
entry 0x40380322
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403807c6
SPIWP:0xee
mode:DIO, clock div:1
load:0x3c020020,len:0x6d30
load:0x3fc8a400,len:0x14e4
load:0x40380000,len:0x7dd4
load:0x42000020,len:0x12a3c
load:0x40387dd4,len:0x248c
load:0x50000010,len:0x10
SHA-256 comparison failed:
Calculated: 15cb229b05ffc7246d7ffefe43c7aaceeda906fef214c27e56fac2068d3872b4
Expected: 269d8c8dcd3611ae87ee33e822d463d9bfcd38beb9460e8a8164c2efe9755cfe
Attempting to boot anyway...
entry 0x40380322

```

dobairoland, 2022-04-20:
Have you tried `idf.py flash` as it was suggested?

dobairoland, 2022-04-20:
I see. In that case you cannot connect the port to the docker image according to the note at the end of the https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-docker-image.html page.

At the end of `idf.py build` you can see: 
```
Project build complete. To flash, run this command:
/home/user/.espressif/python_env/idf5.0_py3.10_env/bin/python ../../../components/esptool_py/esptool/esptool.py ... 
```
Please pay attention to the arguments after `esptool.py`. You need all of those on your Windows machine in order to successfully flash. 

As others pointed out, the following doesn't look correct:
```
In [410]: !esptool --chip esp32c3 --port COM16 --baud 460800 write_flash -z 0x0 C:/test/ESP32/esp32c3/hello_world.bin
```
You can find the correct arguments above. Usually there are 3 binaries to flash. It looks like you are flashing your application at the offset of the bootloader.

---

I'd like to suggest to build and flash from Windows directly. It is a much easier way in comparison with the one you are following now.

closed 2022-04-20 by dobairoland

QGB, 2022-04-20:
flash success

```
!esptool -p COM16 -b 460800 --before default_reset --after hard_reset --chip esp32c3  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x0000 C:/test/ESP32/esp32c3/bootloader.bin 0x8000 C:/test/ESP32/esp32c3/partition-table.bin 0x10000 C:/test/ESP32/esp32c3/hello_world.bin

Error reading from serial device
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd6108,len:0x196c
load:0x403ce000,len:0x904
load:0x403cf600,len:0x2eac
SHA-256 comparison failed:
Calculated: 0e71be9e3c22a30b5d1a0b1177fbc3ff55925c9c8ee37adc4f5dc85acb7a3857
Expected: deea2d801983a9eb4c23c6412e305963ffa926801d656626857ab297f8c9835e
Attempting to boot anyway...
entry 0x403ce000
I (50) boot: ESP-IDF v5.0-dev-2491-g86013e918a 2nd stage bootloader
I (50) boot: compile time 03:38:38
I (50) boot: chip revision: 3
I (53) boot.esp32c3: SPI Speed      : 40MHz
I (58) boot.esp32c3: SPI Mode       : DIO
I (63) boot.esp32c3: SPI Flash Size : 4MB
I (68) boot: Enabling RNG early entropy source...
I (73) boot: Partition Table:
I (77) boot: ## Label            Usage          Type ST Offset   Length
I (84) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (91) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (99) boot:  2 factory          factory app      00 00 00010000 00100000
I (106) boot: End of partition table
I (111) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=06d30h ( 27952) map
I (125) esp_image: segment 1: paddr=00016d58 vaddr=3fc8a400 size=014e4h (  5348) load
I (129) esp_image: segment 2: paddr=00018244 vaddr=40380000 size=07dd4h ( 32212) load
I (144) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=12a3ch ( 76348) map
I (160) esp_image: segment 4: paddr=00032a64 vaddr=40387dd4 size=0248ch (  9356) load
I (163) esp_image: segment 5: paddr=00034ef8 vaddr=50000010 size=00010h (    16) load
I (169) boot: Loaded app from partition at offset 0x10000
I (172) boot: Disabling RNG early entropy source...
I (189) cpu_start: Pro cpu up.
I (198) cpu_start: Pro cpu start user code
I (198) cpu_start: cpu freq: 160000000 Hz
I (198) cpu_start: Application information:
I (201) cpu_start: Project name:     hello_world
I (206) cpu_start: App version:      1
I (210) cpu_start: Compile time:     Apr 20 2022 03:38:18
I (217) cpu_start: ELF file SHA256:  79416edfc35b1f2f...
I (222) cpu_start: ESP-IDF:          v5.0-dev-2491-g86013e918a
I (229) heap_init: Initializing. RAM available for dynamic allocation:
I (236) heap_init: At 3FC8C760 len 000338A0 (206 KiB): DRAM
I (242) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (249) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (256) spi_flash: detected chip: mxic
I (260) spi_flash: flash io: dio
I (264) sleep: Configure to isolate all GPIO pins in sleep state
I (271) sleep: Enable automatic switching of GPIO sleep configuration
I (278) cpu_start: Starting scheduler.
Hello world!
This is esp32c3 chip with 1 CPU core(s), WiFi/BLE, silicon revision 3, 4MB external flash
Minimum free heap size: 328252 bytes
Restarting in 10 seconds...
Restarting in 9 seconds...
Restarting in 8 seconds...
Restarting in 7 seconds...
Restarting in 6 seconds...
Restarting in 5 seconds...
Restarting in 4 seconds...

```

QGB, 2022-04-22:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6a8
load:0x403d0000,len:0x2358
SHA-256 comparison failed:
Calculated: 4bd624aff3257f126bd68f521ac53163cd7fe48cae8d3a77a847ef36aa76ad41
Expected: 2dd871cd993fccdf87b5ef3c78d949bd29ad8f4d97ecb130ff4297267fa0ddbe
Attempting to boot anyway...
entry 0x403ce000
I (85) cpu_start: Pro cpu up.
I (94) cpu_start: Pro cpu start user code
I (94) cpu_start: cpu freq: 160000000 Hz
I (94) cpu_start: Application information:
I (97) cpu_start: Project name:     hello_world
I (102) cpu_start: App version:      1
I (106) cpu_start: Compile time:     Apr 20 2022 03:38:18
I (113) cpu_start: ELF file SHA256:  79416edfc35b1f2f...
I (119) cpu_start: ESP-IDF:          v5.0-dev-2491-g86013e918a
I (125) heap_init: Initializing. RAM available for dynamic allocation:
I (132) heap_init: At 3FC8C760 len 000338A0 (206 KiB): DRAM
I (138) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (145) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (152) spi_flash: detected chip: mxic
I (156) spi_flash: flash io: dio
I (160) sleep: Configure to isolate all GPIO pins in sleep state
I (167) sleep: Enable automatic switching of GPIO sleep configuration
I (174) cpu_start: Starting scheduler.
Hello world!
This is esp32c3 chip with 1 CPU core(s), WiFi/BLE, silicon revision 3, 4MB external flash
Minimum free heap size: 328252 bytes
Restarting in 10 seconds...
Restarting in 9 seconds...
Restarting in 8 seconds...
Restarting in 7 seconds...
Restarting in 6 seconds...
Restarting in 5 seconds...
```

QGB, 2022-04-24:
不使用 idf build 出来的 `C:\Espressif\frameworks\esp-idf-v4.4\examples\get-started\hello_world\build\bootloader\bootloader.bin`.  而使用其他工程的 `bootloader.bin`
虽然可以成功运行，但在启动时会出现：
```

SHA-256 comparison failed:
Calculated: 68bf5beb6eee0d75a66d017ee85c093544b1ef1f666366045b50061c0a4dc49e
Expected: deea2d801983a9eb4c23c6412e305963ffa926801d656626857ab297f8c9835e
Attempting to boot anyway...
entry 0x403ce000
I (53) boot: ESP-IDF v5.0-dev-2491-g86013e918a 2nd stage bootloader
I (53) boot: compile time 03:38:38
```

QGB, 2022-04-24:
加入 arduino 组件时，必须先 删除 老 build 文件夹。不然 idf.py menuconfig "Arduino Configuration" 找不到

```

```