espressif/esp-idf #8799: i2c.h not found in component

Reiner1210, 2022-04-19:
Hello,

I update my IDF and now in my cutom component header the #include "driver/i2c.h" is no longer found.
With the IDF version from december all worked fine .....

In file included from /home/reiner/DATA/data/programm/esp/mycomponents/components/include/bmx80_helper.h:18,
                 from ../src/main.c:43:
/home/reiner/DATA/data/programm/esp/mycomponents/components/include/i2c_helper.h:18:10: fatal error: driver/i2c.h: No such file or directory
 #include "driver/i2c.h"
          ^~~~~~~~~~~~~~

Reiner1210, 2022-04-19:
I found it: I need this in my component:

REQUIRES "driver"

But why has it changed ??

o-marshmallow, 2022-04-20:
Hi @Reiner1210,

As you have found, it is now necessary to explicit the dependency on `driver` component. A refactoring is currently on-going in order to reduce dependencies between lower-level components (such as `hal` or `soc`) and higher-level components (such as `esp_system` or `driver`). One of the consequences of this is to remove `driver` as part of the common components, thus, any component using `driver` shall now explicit it with `REQUIRES` or `PRIV_REQUIRES`

o-marshmallow, 2022-04-21:
Closing this issue

closed 2022-04-21 by o-marshmallow