espressif/esp-idf #8800: Read bootloader ESP-IDF version from device

BoraBene, 2022-04-20:
Hi,

i tried to read the ESP-IDF version of the bootloader from the nvs. Sadly only the ESP-IDF version of the firmware is readable while using the application. The bootloader ESP-IDF is only mentioned once while starting the serial monitor. I used different ESP-IDF versions  ( mentioned in the images)  to proof if the firmware or the bootloader ESP-IDF is given back by the function esp_get_idf_version() .

bootloader ESP-IDF: 
[image]

Firmware ESP-IDF:
[image]

It would be nice to also read the ESP-IDF version of the bootloader from the nvs, so i can see which of my device uses a old version of the bootloader SDK.

KonstantinKondrashov, 2022-04-21:
Hi @BoraBene!
Yes, I agree it might be useful. Seems like we need to add a similar structure in bootloader https://github.com/espressif/esp-idf/blob/master/components/app_update/esp_app_desc.c#L14-L42  and then we will know from which offset we need to read bootloader to extract info about bootloader. The new func like `esp_ota_get_bootloader_description()` will return this structure.

chipweinberger, 2022-04-22:
Yes i've wanted this feature too.

I resorted to writing the bootloader version to its own partition, during bootloader boot. Not ideal! but works.

BoraBene, 2022-04-22:
Thanks, maybe i could use it as a workaround right now. 
Hopefully Espressif is including this in the future, would be a lot easier! :)

chipweinberger, 2022-04-22:
This might help you:

bootloader_ver.h
`void jboot_write_version_needed();`

bootloader_ver.c

```
#include <stdbool.h>
#include "string.h"

#include <esp_err.h>
#include "esp_log.h"
#include "esp_idf_version.h"

#include "bootloader_init.h"
#include "bootloader_utility.h"
#include "bootloader_common.h"

#include "jcommon/system/pd_partition_defines.h"
#include "jcommon/peripherals/pd_gpio.h"

#include "hal/gpio_ll.h"
#include "esp_rom_gpio.h"

#include "bootloader_utils.h"

#include "bootloader_ver.h"

// forward declare
// these are found in "bootloader_flash_priv.h"
#define FLASH_SECTOR_SIZE 0x1000
esp_err_t bootloader_flash_read(size_t src_addr, void *dest, size_t size, bool allow_decrypt);
esp_err_t bootloader_flash_write(size_t dest_addr, void *src, size_t size, bool write_encrypted);
esp_err_t bootloader_flash_erase_sector(size_t sector);

static const char *TAG = "boot ver";

static uint32_t sentinelExpected =  0x4A534F4E; // "JSON" in hex

static uint32_t fourByteAligned(uint32_t a){
    return (a + 3) & ~0x03;
}

// prepend ESP-IDF
static uint32_t expectedBootVer(char* buf, uint32_t len) {

    // len must be 4 byte aligned
    if (len != fourByteAligned(len)){
        abort();
    }

    const char* a = "{";
    const char* b = "\"jBootVer\":\"0.1\"";
    const char* c = ",";
    const char* d = "\"compileDate\":\"" __DATE__ "\"";
    const char* e = ",";
    const char* f = "\"compileTime\":\"" __TIME__ "\"";
    const char* g = ",";
    const char* h = "\"ESP-IDF\":\"" IDF_VER "\"";
    const char* i = "}";
        
    int wrote = 0;
    strncpy(buf+wrote, a, len-wrote); wrote+=strlen(a);
    strncpy(buf+wrote, b, len-wrote); wrote+=strlen(b);
    strncpy(buf+wrote, c, len-wrote); wrote+=strlen(c);
    strncpy(buf+wrote, d, len-wrote); wrote+=strlen(d);
    strncpy(buf+wrote, e, len-wrote); wrote+=strlen(e);
    strncpy(buf+wrote, f, len-wrote); wrote+=strlen(f);
    strncpy(buf+wrote, g, len-wrote); wrote+=strlen(g);
[... 213 lines cut ...]
        const esp_partition_info_t *partition = &partitions[i];

        char label[sizeof(partition->label) + 1] = {0};

        if (partition->type == PART_TYPE_DATA) {

            // partition->label is not null-terminated string.
            strncpy(label, (char *)&partition->label, sizeof(label) - 1);

            if (bootloader_common_label_search(findLabel, label) == true) {

                posOut->offset = partition->pos.offset;
                posOut->size = partition->pos.size;

                ESP_LOGI(TAG, "found: %s, offset:0x%x size:%u", findLabel, posOut->offset,  posOut->size);
                found = true;
            } 
        }
    }

    bootloader_munmap(partitions);

    return found;
}
```

closed 2023-05-21 by espressif-bot

chipweinberger, 2023-05-21:
wow nice work @KonstantinKondrashov , looks perfect ❤️

I'm glad you guys found this important enough to do such a good job implementing it. 

Its very useful to be able to read the bootloader version programatically!

```
typedef struct {
    uint8_t magic_byte;         /*!< Magic byte ESP_BOOTLOADER_DESC_MAGIC_BYTE */
    uint8_t reserved[3];        /*!< reserved for IDF */
    uint32_t version;           /*!< Bootloader version */
    char idf_ver[32];           /*!< Version IDF */
    char date_time[24];         /*!< Compile date and time*/
    uint8_t reserved2[16];      /*!< reserved for IDF */
} esp_bootloader_desc_t;
```

```
/**
 * @brief Returns the description structure of the bootloader.
 *
 * @param[in] bootloader_partition Pointer to bootloader partition.
 *                                 If NULL, then the current bootloader is used (the default location).
 *                                 offset = CONFIG_BOOTLOADER_OFFSET_IN_FLASH,
 *                                 size = CONFIG_PARTITION_TABLE_OFFSET - CONFIG_BOOTLOADER_OFFSET_IN_FLASH,
 * @param[out] desc     Structure of info about bootloader.
 * @return
 *  - ESP_OK                Successful.
 *  - ESP_ERR_NOT_FOUND     Description structure is not found in the bootloader image. Magic byte is incorrect.
 *  - ESP_ERR_INVALID_ARG   Arguments is NULL.
 *  - ESP_ERR_INVALID_SIZE  Read would go out of bounds of the partition.
 *  - or one of error codes from lower-level flash driver.
 */
esp_err_t esp_ota_get_bootloader_description(const esp_partition_t *bootloader_partition, esp_bootloader_desc_t *desc);
```

jonnydman, 2023-11-30:
I know it's done, but for some reason I can't seem to find it under `esp_ota_ops.h`. is it implemented in a different place?
I'm using ESP-IDF version 5.1.1