espressif/esp-idf #8801: ota flash writing crashed

achao1104, 2022-04-20:
idf4.3.2 esp32s2

the dump log is wired, where is wrong?

Re-enable cpu cache.
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40087f0c: fffac61e a1006136 3fb1e13f
0x40087f0c: gpio_intr_disable at D:/project/esp32/idf-v4.3.2/components/driver/gpio.c:188 (discriminator 2)

Core  0 register dump:
PC      : 0x40087f12  PS      : 0x00060033  A0      : 0x40027279  A1      : 0x3ffce840
0x40087f12: gpio_set_level at D:/project/esp32/idf-v4.3.2/components/driver/gpio.c:236

0x40027279: _xt_lowint1 at D:/project/esp32/idf-v4.3.2/components/freertos/port/xtensa/xtensa_vectors.S:1105

A2      : 0x3ffd4549  A3      : 0x00000000  A4      : 0x8000d89b  A5      : 0x4002f9b8
0x4002f9b8: _frxt_int_enter at D:/project/esp32/idf-v4.3.2/components/freertos/port/xtensa/portasm.S:119

A6      : 0x112cef69  A7      : 0x3ffdae40  A8      : 0x800277c8  A9      : 0x00000001
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x80033209  A13     : 0x3ffdadd0
A14     : 0x3f402000  A15     : 0x3ffdae59  SAR     : 0x00000002  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x80033209  LEND    : 0x3ffdadd0  LCOUNT  : 0x40027159
0x40027159: _xt_user_exc at D:/project/esp32/idf-v4.3.2/components/freertos/port/xtensa/xtensa_vectors.S:627

Core  0 was running in ISR context:
EPC1    : 0x40034137  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000
0x40034137: uart_hal_write_txfifo at D:/project/esp32/idf-v4.3.2/components/hal/uart_hal_iram.c:35

Backtrace:0x40087f0f:0x3ffce840 0x40027276:0x3ffce860 0x4000d898:0x3ffdae60 0x40028af5:0x3ffdae80 0x4003d80e:0x3ffdaea0 0x4003d71d:0x3ffdaed0 0x4003d232:0x3ffdaef0 0x400289b1:0x3ffdaf70 0x400a25ed:0x3ffdafd0 0x400a1429:0x3ffdaff0 0x4008fae9:0x3ffdb010 0x4008fd8c:0x3ffdb3d0 0x40094d70:0x3ffdb400 0x4008b080:0x3ffdb420 0x4010605b:0x3ffdb470
0x40087f0f: gpio_intr_disable at D:/project/esp32/idf-v4.3.2/components/driver/gpio.c:189 (discriminator 2)

0x40027276: _xt_lowint1 at D:/project/esp32/idf-v4.3.2/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x40028af5: delay_us at D:/project/esp32/idf-v4.3.2/components/spi_flash/spi_flash_os_func_app.c:148

0x4003d80e: spi_flash_chip_generic_wait_idle at D:/project/esp32/idf-v4.3.2/components/spi_flash/spi_flash_chip_generic.c:373

0x4003d71d: spi_flash_chip_generic_page_program at D:/project/esp32/idf-v4.3.2/components/spi_flash/spi_flash_chip_generic.c:229

0x4003d232: spi_flash_chip_generic_write at D:/project/esp32/idf-v4.3.2/components/spi_flash/spi_flash_chip_generic.c:254

0x400289b1: esp_flash_write at D:/project/esp32/idf-v4.3.2/components/spi_flash/esp_flash_api.c:749

0x400a25ed: esp_partition_write at D:/project/esp32/idf-v4.3.2/components/spi_flash/partition.c:461

0x400a1429: esp_ota_write at D:/project/esp32/idf-v4.3.2/components/app_update/esp_ota_ops.c:250

0x4008fae9: ota_excute at D:\project\esp32\project\solardevice-dog\build/../main/otautil.c:134
 (inlined by) ota_check at D:\project\esp32\project\solardevice-dog\build/../main/otautil.c:303

0x4008fd8c: ota_upgrade at D:\project\esp32\project\solardevice-dog\build/../main/otautil.c:378

0x40094d70: server_send_pending_operation at D:\project\esp32\project\solardevice-dog\build/../main/bizutil.c:793

0x4008b080: app_main at D:\project\esp32\project\solardevice-dog\build/../main/main.c:278 (discriminator 15)

0x4010605b: main_task at D:/project/esp32/idf-v4.3.2/components/freertos/port/port_common.c:133

backtrace:

D:\project\esp32\project\solardevice-dog>xtensa-esp32-elf-addr2line -pfiaC -e D:\project\esp32\project\solardevice-dog\build\solardevice.elf 0x40087f0f:0x3ffce840 0x40027276:0x3ffce860 0x4000d898:0x3ffdae60 0x40028af5:0x3ffdae80 0x4003d80e:0x3ffdaea0 0x4003d71d:0x3ffdaed0 0x4003d232:0x3ffdaef0 0x400289b1:0x3ffdaf70 0x400a25ed:0x3ffdafd0 0x400a1429:0x3ffdaff0 0x4008fae9:0x3ffdb010 0x4008fd8c:0x3ffdb3d0 0x40094d70:0x3ffdb400 0x4008b080:0x3ffdb420 0x4010605b:0x3ffdb470
0x40087f0f: gpio_intr_disable at D:/project/esp32/idf-v4.3.2/components/driver/gpio.c:189 (discriminator 2)
0x40027276: _xt_lowint1 at D:/project/esp32/idf-v4.3.2/components/freertos/port/xtensa/xtensa_vectors.S:1105
0x4000d898: ?? ??:0
0x40028af5: delay_us at D:/project/esp32/idf-v4.3.2/components/spi_flash/spi_flash_os_func_app.c:148
0x4003d80e: spi_flash_chip_generic_wait_idle at D:/project/esp32/idf-v4.3.2/components/spi_flash/spi_flash_chip_generic.c:373
0x4003d71d: spi_flash_chip_generic_page_program at D:/project/esp32/idf-v4.3.2/components/spi_flash/spi_flash_chip_generic.c:229
0x4003d232: spi_flash_chip_generic_write at D:/project/esp32/idf-v4.3.2/components/spi_flash/spi_flash_chip_generic.c:254
0x400289b1: esp_flash_write at D:/project/esp32/idf-v4.3.2/components/spi_flash/esp_flash_api.c:749
0x400a25ed: esp_partition_write at D:/project/esp32/idf-v4.3.2/components/spi_flash/partition.c:461
0x400a1429: esp_ota_write at D:/project/esp32/idf-v4.3.2/components/app_update/esp_ota_ops.c:250
0x4008fae9: ota_excute at D:\project\esp32\project\solardevice-dog\build/../main/otautil.c:134
 (inlined by) ota_check at D:\project\esp32\project\solardevice-dog\build/../main/otautil.c:303
0x4008fd8c: ota_upgrade at D:\project\esp32\project\solardevice-dog\build/../main/otautil.c:378
0x40094d70: server_send_pending_operation at D:\project\esp32\project\solardevice-dog\build/../main/bizutil.c:793
0x4008b080: app_main at D:\project\esp32\project\solardevice-dog\build/../main/main.c:278 (discriminator 15)
0x4010605b: main_task at D:/project/esp32/idf-v4.3.2/components/freertos/port/port_common.c:133

mahavirj, 2022-04-20:
@achao1104 

Please refer to documentation related to this issue at https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/storage/spi_flash_concurrency.html#iram-safe-interrupt-handlers

>  0x40087f0c: gpio_intr_disable at D:/project/esp32/idf-v4.3.2/components/driver/gpio.c:188 (discriminator 2)

This address point to flash cache region for ESP32-S2. So it appears like there is an interrupt handler registered with flag `ESP_INTR_FLAG_IRAM` that calls a function residing in flash cache. This will not work in case flash cache itself is disabled, e.g. in this case its disabled due to flash program operation.

Please check your application for this. If problem persists, please share your application code.

achao1104, 2022-04-21:
Thanks for your answer!! 
Due to lack of uart, we use timer to simulate a uart to print log, as you said, I guess when it is executing OTA (write to flash) and printing log at the same time, it will cause the crash. How can I improve it?

void IRAM_ATTR timer_group0_isr(void *para)
{
    timer_spinlock_take(TIMER_GROUP_0);
    uint32_t timer_intr = timer_group_get_intr_status_in_isr(TIMER_GROUP_0);
    if (timer_intr & TIMER_INTR_T1)
    {
    	if(txbitsidx<8)
    	{
    		if(txdataidx<txdatalen)
    			gpio_set_level(GPIO_UART_SIMULATE,(txdata[txdataidx]>>txbitsidx)&0x01);
    		txbitsidx++;
    	}else if((txbitsidx>=8)&&(txbitsidx<=8))
    	{
    		gpio_set_level(GPIO_UART_SIMULATE,1);
    		txbitsidx++;
    	}else
    	{
    		txbitsidx=0;
    		txdataidx++;
    		if(txdataidx>=txdatalen){
    			timer_pause(TIMER_GROUP_0,TIMER_1);
    			txdone=1;
    		}
    		else
    			gpio_set_level(GPIO_UART_SIMULATE,0);
    	}
        timer_group_clr_intr_status_in_isr(TIMER_GROUP_0, TIMER_1);
        timer_group_enable_alarm_in_isr(TIMER_GROUP_0, TIMER_1);
    }
    timer_spinlock_give(TIMER_GROUP_0);
}
static void example_tg0_timer_init(double timer_interval_usec)
{
    /* Select and initialize basic parameters of the timer */
    timer_config_t config = {
        .divider = TIMER_DIVIDER,
        .counter_dir = TIMER_COUNT_UP,
        .counter_en = TIMER_PAUSE,
        .alarm_en = TIMER_ALARM_EN,
        .auto_reload = 1,
    }; // default clock source is APB
    timer_init(TIMER_GROUP_0, TIMER_1, &config);

    /* Timer's counter will initially start from value below.
       Also, if auto_reload is set, this value will be automatically reload on alarm */
    timer_set_counter_value(TIMER_GROUP_0, TIMER_1, 0x00000000ULL);

    /* Configure the alarm value and the interrupt on alarm. */
    timer_interval_usec=timer_interval_usec * TIMER_SCALE/1000000;
    timer_set_alarm_value(TIMER_GROUP_0, TIMER_1, timer_interval_usec);
    timer_enable_intr(TIMER_GROUP_0, TIMER_1);
    timer_isr_register(TIMER_GROUP_0, TIMER_1, timer_group0_isr,(void *) TIMER_1, ESP_INTR_FLAG_IRAM, NULL);

    //timer_start(TIMER_GROUP_0, timer_idx);
}

void uart2_tx(uint8_t *data,int datalen)
{
	if(datalen<=0)
		return;
	txdata=data;
	txdatalen=datalen;
	txbitsidx=0;
	txdataidx=0;
	txdone=0;
	timer_set_counter_value(TIMER_GROUP_0, TIMER_1, 0x00000000ULL);
	//timer_pause(TIMER_GROUP_0,TIMER_1);
	gpio_set_level(GPIO_UART_SIMULATE,0);
	timer_start(TIMER_GROUP_0, TIMER_1);
	uint32_t maxtimeout=0;
	while((!txdone)&&(maxtimeout<50*1000))
	{
		delay_ms(20);
		maxtimeout+=20;
	}
}

mahavirj, 2022-04-21:
@achao1104 

Simple solution here could be to remove `ESP_INTR_FLAG_IRAM` during timer ISR registration:

```
timer_isr_register(TIMER_GROUP_0, TIMER_1, timer_group0_isr,(void *) TIMER_1, 0, NULL);
```

Please note that this also means timer interrupt won't be active whilst flash cache is disabled. If you would like to keep it enabled, please refactor its ISR to use all data/code from internal memory only. Please refer to documentation pointer I shared earlier for this.

achao1104, 2022-04-21:
thanks a lot.

achao1104, 2022-04-22:
sorry for repopen, but just clarify some questions to study how to read the dump log.

1) 0x40087f0c is in icache address region, that is why you said the following code resides in flash cache, is it right?
0x40087f0c: gpio_intr_disable at D:/project/esp32/idf-v4.3.2/components/driver/gpio.c:188 (discriminator 2)
2) The following is the isr program, and I didn't see prefix IRAM_ATTR in the front of such as gpio_set_level, however, it looks the manual said besides codes with prefix IRAM_ATTR reside in ram, there are some other codes such as freertos always reside in ram, but I don't know if gpio related api such as gpio_set_level is in or not ram region.
void IRAM_ATTR timer_group0_isr(void *para)
{
    timer_spinlock_take(TIMER_GROUP_0);
    uint32_t timer_intr = timer_group_get_intr_status_in_isr(TIMER_GROUP_0);
    if (timer_intr & TIMER_INTR_T1)
    {
    	if(txbitsidx<8)
    	{
    		if(txdataidx<txdatalen)
    			gpio_set_level(GPIO_UART_SIMULATE,(txdata[txdataidx]>>txbitsidx)&0x01);
    		txbitsidx++;
    	}else if((txbitsidx>=8)&&(txbitsidx<=8))
    	{
    		gpio_set_level(GPIO_UART_SIMULATE,1);
    		txbitsidx++;
    	}else
    	{
    		txbitsidx=0;
    		txdataidx++;
    		if(txdataidx>=txdatalen){
    			timer_pause(TIMER_GROUP_0,TIMER_1);
    			txdone=1;
    		}
    		else
    			gpio_set_level(GPIO_UART_SIMULATE,0);
    	}
        timer_group_clr_intr_status_in_isr(TIMER_GROUP_0, TIMER_1);
        timer_group_enable_alarm_in_isr(TIMER_GROUP_0, TIMER_1);
    }
    timer_spinlock_give(TIMER_GROUP_0);
}

mahavirj, 2022-04-22:
> 0x40087f0c is in icache address region, that is why you said the following code resides in flash cache, is it right?

Yes. This address belongs  to flash code region and its accessed through flash cache.

https://github.com/espressif/esp-idf/blob/079b5b1857242d6288478c62dc36a843718882e9/components/esp_system/ld/esp32s2/memory.ld.in#L74-L76

> there are some other codes such as freertos always reside in ram

Some core components (including `freertos`) contains linker fragments (`linker.lf`) for default placement of code/data regions to specific memory. You may find more information on this from https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-guides/linker-script-generation.html

> but I don't know if gpio related api such as gpio_set_level is in or not ram region.

You can check for relevant symbol address in generated map file in `build` directory.

We also have a tool that can help to give list of symbols from specific component library and their section placement:

```
$ idf_size.py --archive_details  libdriver.a  build/generic_gpio.map 
...<snip>...
Symbols from section: .flash.text
	periph_module_enable             : 1172
	gpio_config                      :  524
	uart_flush_input                 :  456
	uart_wait_tx_done                :  365
	gpio_set_level                   :  231
	gpio_isr_handler_add             :  213
	gpio_set_intr_type               :  209
	gpio_output_disable              :  193
	rtc_gpio_deinit                  :  192
	gpio_isr_handler_remove          :  190
	gpio_output_enable               :  188
	gpio_input_enable                :  150
	gpio_pullup_en                   :  148
	gpio_pulldown_en                 :  144
	gpio_pullup_dis                  :  144
	gpio_intr_enable                 :  142
```

Hope this helps!

closed 2022-04-24 by achao1104