espressif/esp-idf #8808: example adc\dma_read return ESP_ERR_INVALID_STATE

hard-cat, 2022-04-21:
## Environment

- Development Kit:      ESP32-LyraT
- IDF version v4.4
- Build System: idf.py
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch2) 8.4.0
- Operating System:    Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: No
- Power Supply:        USB

The result of reading adc_digi_read_bytes always returns 103 except for the first read.

LOG
`I (27) boot: ESP-IDF v4.4-dirty 2nd stage bootloader
I (27) boot: compile time 15:53:25
I (27) boot: chip revision: 1
I (30) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (37) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (46) boot.esp32: SPI Flash Size : 2MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00100000
I (90) boot: End of partition table
I (94) boot_comm: chip revision: 1, min. application chip revision: 0
I (101) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=093c4h ( 37828) map
I (123) esp_image: segment 1: paddr=000193ec vaddr=3ffb0000 size=02380h (  9088) load
I (127) esp_image: segment 2: paddr=0001b774 vaddr=40080000 size=048a4h ( 18596) load
I (137) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=173b8h ( 95160) map
I (172) esp_image: segment 4: paddr=000373e0 vaddr=400848a4 size=07634h ( 30260) load
I (185) esp_image: segment 5: paddr=0003ea1c vaddr=50000000 size=00010h (    16) load
I (191) boot: Loaded app from partition at offset 0x10000
I (191) boot: Disabling RNG early entropy source...
I (205) cpu_start: Pro cpu up.
I (205) cpu_start: Starting app cpu, entry point is 0x400810c4
I (0) cpu_start: App cpu up.
I (219) cpu_start: Pro cpu start user code
I (219) cpu_start: cpu freq: 160000000
I (219) cpu_start: Application information:
I (224) cpu_start: Project name:     adc
I (228) cpu_start: App version:      v4.4-dirty
I (233) cpu_start: Compile time:     Apr 21 2022 15:53:17
I (239) cpu_start: ELF file SHA256:  41a0b8d3492f8fa1...
I (245) cpu_start: ESP-IDF:          v4.4-dirty
I (251) heap_init: Initializing. RAM available for dynamic allocation:
I (258) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (264) heap_init: At 3FFB2C80 len 0002D380 (180 KiB): DRAM
I (270) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (276) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (283) heap_init: At 4008BED8 len 00014128 (80 KiB): IRAM
I (290) spi_flash: detected chip: gd
I (293) spi_flash: flash io: dio
W (297) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (312) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (322) gpio: GPIO[35]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (332) ADC DMA: adc_pattern[0].atten is :0
I (332) ADC DMA: adc_pattern[0].channel is :7
I (342) ADC DMA: adc_pattern[0].unit is :0
I (342) TASK:: ret is 0, ret_num is 256
I (362) TASK:: ret is 103, ret_num is 256
I (372) TASK:: ret is 103, ret_num is 256
I (382) TASK:: ret is 103, ret_num is 256
I (392) TASK:: ret is 103, ret_num is 256
I (402) TASK:: ret is 103, ret_num is 256
I (412) TASK:: ret is 103, ret_num is 256
I (422) TASK:: ret is 103, ret_num is 256
I (432) TASK:: ret is 103, ret_num is 256
I (442) TASK:: ret is 103, ret_num is 256
I (452) TASK:: ret is 103, ret_num is 256
I (462) TASK:: ret is 103, ret_num is 256
I (472) TASK:: ret is 103, ret_num is 256
I (482) TASK:: ret is 103, ret_num is 256
I (492) TASK:: ret is 103, ret_num is 256
I (502) TASK:: ret is 103, ret_num is 256
I (512) TASK:: ret is 103, ret_num is 256
I (522) TASK:: ret is 103, ret_num is 256
I (532) TASK:: ret is 103, ret_num is 256
I (542) TASK:: ret is 103, ret_num is 256
I (552) TASK:: ret is 103, ret_num is 256
I (562) TASK:: ret is 103, ret_num is 256
I (572) TASK:: ret is 103, ret_num is 256`

hard-cat, 2022-04-21:
I replaced the free space check code in the ring buffer in file ringbuf.c

based on code signatures:
  CheckItemFitsFunction_t xCheckItemFits; //Function to check if item can currently fit in ring buffer
and
GetCurMaxSizeFunction_t xGetCurMaxSize; //Function to get current free size

the function and the variable determine the free space in the ring buffer, so I did this:

    portENTER_CRITICAL_ISR(&pxRingbuffer->mux);
    //if (pxRingbuffer->xCheckItemFits(xRingbuffer, xItemSize) == pdTRUE) {   <--- old 
    if (pxRingbuffer->xGetCurMaxSize >= xItemSize) {                            <---- new
        pxRingbuffer->vCopyItem(xRingbuffer, pvItem, xItemSize);
        xReturn = pdTRUE;
        //Check if the free semaphore should be returned to allow other tasks to send
        if (prvGetFreeSize(pxRingbuffer) > 0) {
            xReturnSemaphore = pdTRUE;
        }
    } else {
        xReturn = pdFALSE;
    }

the error is gone, but so far I haven't checked if the measurement is working correctly.

hard-cat, 2022-04-21:
applied these corrections, the delay in response to a change in voltage at the input of the ADC disappeared.
https://github.com/espressif/esp-idf/issues/8485

it seems to me that setting .sample_freq_hz is not applied, and the sample is always constant....

elidupree, 2022-05-07:
I'm having the same sample_freq_hz issue; I reported it as #8874 before I eventually sound this issue while digging for a workaround.

Any chance you've found a workaround for this? I've tried to apply the suggested workarounds in #6691 and #3325 (which might be related because it uses the i2s timer on ESP32?), but all I've managed to do so far is cause panics…

hard-cat, 2022-05-07:
I did not find a workaround, moreover, I see duplication of results in the conversion results. the number of takes depends on the sample rate setting.

elidupree, 2022-05-07:
Oh wow. Could the duplication be related to #3686? I've likewise observed that the samples don't arrive in the channel order 4,5,6,7,4,5,6,7 as I expected (instead, I get multiple of the same channel in a row), but since the documentation doesn't actually say what you're expected to get, I didn't know if that was wrong