espressif/esp-idf #8809: which ESP-IDF Versions support CONFIG_MBEDTLS_SSL_PROTO_TLS1_3 ?

franz-ms-muc, 2022-04-21:
**Is your feature request related to a problem? Please describe.**

CONFIG_MBEDTLS_SSL_PROTO_TLS1_3 needed for a Project. 
it seems ESP-IDF 5.0 already has it. 
it seems ESP-IDF 4.4 does not have it. 

**Describe the solution you'd like**

have CONFIG_MBEDTLS_SSL_PROTO_TLS1_3

mahavirj, 2022-04-22:
@franz-ms-muc 

Yes, `TLS1.3`  will be supported in ESP-IDF v5.0 release. You may find more information on this at https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/mbedtls.html#mbed-tls-support-in-esp-idf. Since, v3.x release of mbedTLS involves breaking changes, we will not be making this available in ESP-IDF v4.x series.

Additional references:
- https://github.com/espressif/esp-idf/issues/8515
- https://github.com/Mbed-TLS/mbedtls/blob/development/docs/architecture/tls13-support.md

Hope this helps!

mahavirj, 2022-04-25:
Closing. Please feel free to lets us know if there are any further questions on this topic.

closed 2022-04-25 by mahavirj

franz-ms-muc, 2022-04-25:
is there a Release Date for ESP-IDF 5.0 ? 

Thanks !

franz-ms-muc, 2022-04-25:
Other Question: if I use wolfssl with ESP-IDF, is there a Difference in TLS1.3 Support ? or is it same Story: only 5.0 Support TLS1.3 ?

mahavirj, 2022-04-26:
> is there a Release Date for ESP-IDF 5.0 ?

Please expect v5.0-RC release towards end of May or early June timeframe.

> Other Question: if I use wolfssl with ESP-IDF, is there a Difference in TLS1.3 Support ?  or is it same Story: only 5.0 Support TLS1.3 ?

You can use wolfSSL based TLS1.3 support with current ESP-IDF stable release as well. Please go through https://github.com/espressif/esp-wolfssl once, especially note about licensing.