espressif/esp-idf #8810: [efuse_table_gen.py] Field at USER_DATA, EFUSE_BLK3, 0, 256  intersected

chipweinberger, 2022-04-22:
Given that EFUSE_BLK3 is "User Data" it seems weird that using it in a custom table results in an error?

**esp_efuse_custom_table.csv**
```
JSERIAL_NUMBER,                  EFUSE_BLK3,    0,  32,   
JMODEL_NUMBER,                   EFUSE_BLK3,    32, 10,  
JHARDWARE_REV,                   EFUSE_BLK3,    42, 10,  
```

**Logs: idf.py efuse-custom-table**
``
Executing action: efuse-custom-table
Running ninja in directory /jamcorder_app/build

Executing "ninja efuse-custom-table"...

[1/1] cd /Volumes/...app/esp_efuse_custom_table.csv -t esp32s3 --max_blk_len 25

FAILED: esp-idf/efuse/CMakeFiles/efuse-custom-table 

cd /jamcorder_app/build/esp-idf/efuse && 
/.espressif/python_env/idf4.4_py3.9_env/bin/python
/esp-idf/components/efuse/efuse_table_gen.py 
/esp-idf/components/efuse/esp32s3/esp_efuse_table.csv
/jamcorder_app/esp_efuse_custom_table.csv -t esp32s3 --max_blk_len 256

Parsing efuse CSV input file /esp-idf/components/efuse/esp32s3/esp_efuse_table.csv ...
Verifying efuse table...
Parsing efuse CSV input file /jamcorder_app/esp_efuse_custom_table.csv ...
Verifying efuse table...
Field at USER_DATA, EFUSE_BLK3, 0, 256  intersected with  JSERIAL_NUMBER, EFUSE_BLK3, 0, 32
Max number of bits in BLK 256
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
charlesweinberger@Chips-Macbook-Pro-2 jamcorder_app % 
```

chipweinberger, 2022-04-22:
This person also hit this issue, but no reply:
https://esp32.com/viewtopic.php?f=13&p=95612

chipweinberger, 2022-04-22:
I Imagine, this could be solved by modifying `/esp-idf/components/efuse/esp32s3/esp_efuse_table.csv `, but I really should not have to do that. 

I like to keep my components original (and easily updateable with git) unless I really need to modify them.

davidzuhn, 2022-04-22:
That's my post from esp32.com referenced above, and I came here to report the bug.   Finding this report, I _strongly_ agree with Chip/Charles(?), in that we should be able to use EFUSE_BLK3 without having to remove entries from the table in the esp-idf repository.

KonstantinKondrashov, 2022-04-22:
Hi @chipweinberger and @davidzuhn!

[BLOCK3 is described in csv file as](https://github.com/espressif/esp-idf/blob/master/components/efuse/esp32s3/esp_efuse_table.csv#L158-L160)

To resolve this issue you need to describe your fields in `esp_efuse_custom_table.csv` as:

```
USER_DATA.JSERIAL_NUMBER,                  EFUSE_BLK3,    0,  32,   
USER_DATA.JMODEL_NUMBER,                   EFUSE_BLK3,    32, 10,  
USER_DATA.JHARDWARE_REV,                   EFUSE_BLK3,    42, 10,  
```
(not need to modify /esp-idf/components/efuse/esp32s3/esp_efuse_table.csv).

Final eFuse name will be like `USER_DATA_JSERIAL_NUMBER`.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/efuse.html#structured-efuse-fields

chipweinberger, 2022-04-22:
Thanks for the answer. That's actually rather elegant :)

Some suggestions:

Surprisingly the esp-idf example code does not do it that way:

https://github.com/espressif/esp-idf/blob/master/examples/system/efuse/main/esp_efuse_custom_table.csv

And imo eFuse Manager docs are not clear enough (I read it before posting). It should have this error string and example on there. - like a little FAQ.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/system/efuse.html

KonstantinKondrashov, 2022-04-22:
>  It should have this error string and example on there. - like a little FAQ.

Thanks for the suggestion, I'll improve it and consider extending the example app to show how to use this feature.

davidzuhn, 2022-04-23:
This is good information to have.   Thanks.   

Modifying the table in the examples/system/efuse directory to include the USER_DATA prefix does get me further along.   But then the data shown from 'idf.py show-efuse-table' doesn't make sense anymore:

```
$ idf.py show-efuse-table
[specific values not shown]
Used bits in efuse table:
EFUSE_BLK0 
[0 31] [0 0] [2 13] [15 16] [18 18] [20 30] [32 38] [32 38] [40 52] [57 60] [68 70] [80 111] [116 121] [124 157] [159 159] 

EFUSE_BLK1 
[0 122] [186 191] 

EFUSE_BLK10 
[0 255] 

EFUSE_BLK2 
[0 129] [132 254] 

EFUSE_BLK3 
[0 255] [56 93] [200 247] 

[more blocks not shown]

Note: Not printed ranges are free for using. (bits in EFUSE_BLK0 are reserved for Espressif)
```

In EFUSE_BLK3, the entire range of [0 255] is shown as used, and then subsets of that range as well. 

If this is the new mandatory way to access bits in EFUSE_BLK3, then please update the documentation and create a new example (since the same CSV file won't work for ESP32 and ESP32-S3).  As I read the text "The field_name supports structured format using ." it does not imply that this is necessary.   

It also seems unusual to me that this is a syntax change from the way that ESP32 worked, and that there is no backport, so I can't use this new style of definition with an older design.  I have existing devices with ESP32 and a new design intended to make use of ESP32-S3, and I hope to minimize the amount of difference between the two designs.  Now it looks like I need to have two custom efuse tables (with no difference in semantic content) just because the syntax for the fuse name has changed.    

In addition, this is forcing a naming convention on the device codebase.   From the manual: "EFUSE_BLK4 or EFUSE_BLK_KEY0 can be used as key (for secure_boot or flash_encryption) or for user purposes;".   Testing shows that I can create entries like:

```
KEY0.COUNTRYCODE,              EFUSE_BLK4,    16, 16,    ISO Country code
SYS_DATA_PART2.SPEED_SETTING,  EFUSE_BLK10,   0,  16,    Comment goes here
```

but if I'm not using this as a key, then I've got a 'KEY0' in the name (ESP_EFUSE_KEY0_COUNTRYCODE) only because I'm forced to by the hierarchical syntax.   I guess I don't mind 'USER_DATA' because that's what it is, but 'KEY0'...'KEY5' or 'SYS_DATA_PART2' seem unnecessarily specific to something that may not be the case.   If I'm not creating a key but instead using it for more user data (per the manual), why do I have to have KEY0 in the name?     Similarly, ESP_EFUSE_SYS_DATA_PART2_SPEED_SETTING is cumbersome as opposed to the (ESP32 style fuse name) ESP_EFUSE_SPEED_SETTING.

KonstantinKondrashov, 2022-04-25:
Hi @davidzuhn!
> In EFUSE_BLK3, the entire range of [0 255] is shown as used, and then subsets of that range as well.
EFUSE_BLK3 
[0 255] [56 93] [200 247] 

BLOCK3 has fields:
USER_DATA with len=256
USER_DATA.FIELD1 wit len=38
USER_DATA.MAC_CUSTOM with len=48
It prints these ranges correctly. All of these fields can be used in the code, their ranges are not free.

> It also seems unusual to me that this is a syntax change from the way that ESP32 worked

The syntax has not changed, it has been extended to support nested fields.
ESP32 does not have a field for whole block3 in the efuse common table this is why you are able to define fields in customer table directly. The common efuse table for ESP32-S3 chip has fields for each efuse blocks, which leads to an error like `Field at USER_DATA, EFUSE_BLK3, 0, 256  intersected with ...` when you adding a fields that reside in EFUSE_BLK3.

I agree with the rest of your comment, but can not resolve it easily. The efuse_table_gen.py checks efuse fields wraping/crossing/unique names.

There may be a way to do this, do it manually, skipping the common table checks with the custom table. Run the command only on the custom table (see below).

```
esp/esp-idf/components/efuse/efuse_table_gen.py  esp/esp-idf/examples/system/efuse/main/esp_efuse_custom_table.csv -t esp32s3
```

chipweinberger, 2022-05-03:
Closing this, since my issue is resolved with the dot syntax. @davidzuhn, feel free to open new issue with your concerns.

closed 2022-05-03 by chipweinberger