espressif/esp-idf #8815: ESP32-S3 vs ESP32 inconsistent SPI results

ThuanLeUte, 2022-04-22:
## Environment

Module used: ESP32-S3-WROOM-2
IDF version: V4.4
Build System: idf.py
Compiler version : xtensa-esp32-elf-gcc
Operating System: Linux

## Problem Description

We are replacing ESP32 with ESP32S3 in some designs. ESP32S3 SPI Master in mode 0 does not read the MISO signals correctly.

**Note: We are using the same code and SDK (V4.4) on the ESP32 and ESP32S3. But only ESP32S3 have the issue with SPI.**

**Expected Behavior**
SPI-master of ESP32S3 in mode 0 works like ESP32 with SPI-Slaves (ST25R95).

**Actual Behavior**

Data read from the ESP32S3 is one bit shift left compare with the actual data.

For example:  The actual data from ST25R95 is 0x55 but the ESP32S3 read data is 0xAA.

Please check at the pictures.

[image]
[image]
[image]
[image]
[image]

ThuanLeUte, 2022-04-25:
I think this is the issue from the ESP32S3. Let espressif confirm on that

ElieSKickMaker, 2022-04-26:
Well, like you, I've tested on the ESP32 and it's working. I don't have the spi shifted bug..

linhkienthuduc21, 2022-05-29:
I have faced the same issue with ESP32S3 + CR95HF. The expectation data (measured on logic analyzer is 0x0E, but ESP32S3 rx 0x1C). I have try so many things but it seem not working

bdecherf, 2022-10-13:
I have the same issue with esp32s3, but it works setting the frequency at 2MHz.

idea--list, 2022-11-01:
I am porting my code (idf 4.4.2) from ESP32 (Rev 3) to ESP32S3. I am logging data of an SPI sensor at 512Hz.
The ESP32 logs reliably for many hours, while the ESP32S3 works fine for ca 1 hour, and then it hangs within the first 3 hours. Always. I already set the SPI bus frequency to 100K only... and i do not have an oscilloscope to check the lines.

Do you think this issue may be related to my observation? Is this bug relevant to the ST25R95 only or is it a general MISO line issue on the ESP32S3?

sergiosider, 2023-08-06:
Well, we are in 2023., August... faced the same problem, no answer from ESPRESSIF.

Looking to the osciloscope it looks that ST25R95 is "giving one shift left" , like it has received a false clk before. Could be a glitch on ESP32S3 ?

dexter323i, 2025-03-22:
I have the same issue with ESP32-S3. I have an SPI master with 1MHz clock, SPI Mode2, and reads 3 bytes from a Slave. (It sends 0, 0, 0 on the MOSI line) The master is a closed device, I cannot modify it. My project is with ESP32-S3 is to create a Slave for this Master. My code is adjusted for Arduino framework, but it is based on the official ESP32-S3 SPI Slave example code:

```
#include <Arduino.h>
#include <stdio.h>
#include <stdint.h>
#include <stddef.h>
#include <string.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

#include "esp_log.h"
#include "driver/spi_slave.h"
#include "driver/gpio.h"

#define PIN_SPI_SS      10
#define PIN_SPI_SCK     12
#define PIN_SPI_MOSI    11
#define PIN_SPI_MISO    13
#define GPIO_HANDSHAKE  (gpio_num_t)2

#define RCV_HOST    SPI2_HOST

//Called after a transaction is queued and ready for pickup by master. We use this to set the handshake line high.
void my_post_setup_cb(spi_slave_transaction_t *trans)
{
    gpio_set_level(GPIO_HANDSHAKE, 1);
}

//Called after transaction is sent/received. We use this to set the handshake line low.
void my_post_trans_cb(spi_slave_transaction_t *trans)
{
    gpio_set_level(GPIO_HANDSHAKE, 0);
}

// Define static send buffer with 3 bytes.
WORD_ALIGNED_ATTR uint8_t sendbuf[4] = {0x01, 0x01, 0x01, 0x00};
// Buffer for receiving 3 bytes.
WORD_ALIGNED_ATTR uint8_t recvbuf[4] = {0};
esp_err_t ret;
spi_slave_transaction_t t;

void setup() {
        pinMode(PIN_SPI_MISO, OUTPUT);
        pinMode(PIN_SPI_MOSI, INPUT_PULLDOWN);
        pinMode(PIN_SPI_SCK, INPUT);
        pinMode(PIN_SPI_SS, INPUT);

        Serial.begin(115200);

        delay(2000);

        //Configuration for the SPI bus
        spi_bus_config_t buscfg = {
            .mosi_io_num = PIN_SPI_MOSI,
            .miso_io_num = PIN_SPI_MISO,
            .sclk_io_num = PIN_SPI_SCK,
            .quadwp_io_num = -1,
            .quadhd_io_num = -1,
        };

        //Configuration for the SPI slave interface
        spi_slave_interface_config_t slvcfg = {
            .spics_io_num = PIN_SPI_SS,
            .flags = 0,
            .queue_size = 3,
            .mode = 2,
            .post_setup_cb = my_post_setup_cb,
            .post_trans_cb = my_post_trans_cb
        };

        //Configuration for the handshake line
        gpio_config_t io_conf = {
            .pin_bit_mask = (1 << GPIO_HANDSHAKE),
            .mode = GPIO_MODE_OUTPUT,
            .intr_type = GPIO_INTR_DISABLE
        };

        //Configure handshake line as output
        gpio_config(&io_conf);
        //Enable pull-ups on SPI lines so we don't detect rogue pulses when no master is connected.
        gpio_set_pull_mode((gpio_num_t)PIN_SPI_MOSI, GPIO_PULLUP_ONLY);
        gpio_set_pull_mode((gpio_num_t)PIN_SPI_SCK, GPIO_PULLUP_ONLY);
        gpio_set_pull_mode((gpio_num_t)PIN_SPI_SS, GPIO_PULLUP_ONLY);

        //Initialize SPI slave interface
        ret = spi_slave_initialize(RCV_HOST, &buscfg, &slvcfg, SPI_DMA_CH_AUTO);
        assert(ret == ESP_OK);

        memset(&t, 0, sizeof(t));
}

void loop() {
        //Clear receive buffer, set send buffer to something sane
        memset(recvbuf, 0, sizeof(recvbuf));

        //Set up a transaction of 128 bytes to send/receive
        t.length = 4 * 8;
        t.tx_buffer = sendbuf;
        t.rx_buffer = recvbuf;
        /* This call enables the SPI slave interface to send/receive to the sendbuf and recvbuf. The transaction is
        initialized by the SPI master, however, so it will not actually happen until the master starts a hardware transaction
        by pulling CS low and pulsing the clock etc. In this specific example, we use the handshake line, pulled up by the
        .post_setup_cb callback that is called as soon as a transaction is ready, to let the master know it is free to transfer
        data.
        */
        ret = spi_slave_transmit(RCV_HOST, &t, portMAX_DELAY);
}
```

It is strange bit shift only happens after a bit is sent normally. E.g. "0, 0, 1" arrives perfectly!

Bytes "1, 1, 1" sent:
[image]

"0, 1, 5" sent:
[image]

"5, 3, 7" sent:
[image]

wanckl, 2025-04-17:
@dexter323i @ThuanLeUte Hi, It is known issue that SPI don't working with `ST25R95` family NFC chips, because the **slave** not totally follow the SPI standard timing (It change the MISO line half cycle earlier).

And any other slave devices who change the MISO line half cycle earlier will also meet issues working with esp32s2/3.

Guess 2 possible solutions are:
- Add a fifo chip between MISO line, which let MISO signal delay for half spi clock cycle.
- Or add 2 devices for mode 0 and mode 2, manually control CS line then write via mode 0 but read via mode 2.

---
From the latest idf we can see spi driver introduce a new config https://github.com/espressif/esp-idf/blob/e9bdd395995f60878e01cece10525987bacc8a5f/components/hal/include/hal/spi_types.h#L82

Which indicat that in latest chips, it support to config the sample phases to adopt more devices, good news.

wanckl, 2025-04-25:
@ThuanLeUte Hi, Dose your issue any update?

DB8KN, 2025-08-20:
Hi folks,
anyone make any progress regarding thsi issue?
I just ran into the same issue with _ESP32-S3-WROOM-2_ and the _SI4463_ Transceiver. As with the _ST25R95_, it seems the device does not exactly conform to the SPI standard [si4463 forum post](https://community.silabs.com/s/question/0D5Vm000009CPtLKAW/si4463-c2a-receive-bit-broken?language=en_US).

So i did some digging around inside the esp code.

1. The `spi_sampling_point_t` that @wanckl proposed does not seem to apply to the ESP32-S3 as it does not seem to have this feature. 
 https://github.com/espressif/esp-idf/blob/758939caecb16e5542b3adfba0bc85025517db45/components/hal/esp32s3/include/hal/spi_ll.h#L731-L734
2. The `spi_hal_timing_conf_t` struct has members `int timing_miso_delay` and `int timing_dummy` which could be used to manipulate the sampling point. 
https://github.com/espressif/esp-idf/blob/8c750b088c7cd857d079c0eeb495da199b359461/components/hal/include/hal/spi_hal.h#L72-L82
The ESP32-S3s [Technical Reference Manual](https://www.espressif.com/sites/default/files/documentation/esp32-s3_technical_reference_manual_en.pdf) indeed specifies the corresponding config registers starting from pages 1146 and 1166 and following. However, the function that should implement setting these registers is empty:
https://github.com/espressif/esp-idf/blob/758939caecb16e5542b3adfba0bc85025517db45/components/hal/esp32s3/include/hal/spi_ll.h#L881-L892
I have attempted to implement this and related missing functions, and though i seem to have been able to somewhat change the sampling point and get different levels of data corruption, i have not yet been able to reach a point where there are less, not more bit errors.
I am not 100% sure yet about how the dummy clock cycles are supposed to work. I'll start digging into this next.

If anyone else is currently working on this or related issues let me know :)

mattdontek, 2025-09-10:
We have ran into this problem also.  The code which runs fine on the ESP32 doesn't work on the ESP32S3.  A workaround is to set the clock speed higher to reduce the issue, but it gives me little confidence that the SPI on the S3 works reliably.  It sounds like whoever was assigned to look at this is blaming the chips that the S3 is connected to rather than fixing the issue that wasn't present in the ESP32.

wanckl, 2026-09-15:
Hi folks,    

esp32 vs s3 spi master sample point is known design different, can't say it is a bug but what software can adjust is limited.

except chips who support `spi_sampling_point_t`.

For chips who don't support sample point:
`spi_ll_set_miso_delay` is deal for case that slave output data late, but ST25R95 is earlier, it take effect but maybe sampled to 2nd bit.