espressif/esp-idf #8819: [ESP32-S3][eFuse] OPTIONAL_UNIQUE_ID has no documentation. Why does it exist?

chipweinberger, 2022-04-22:
in esp_efuse_table.csv, there a reference to an optional 128-bit ID.

```
# SYS_DATA_PART1 BLOCK# - System configuration
#######################
    OPTIONAL_UNIQUE_ID,                   EFUSE_BLK2,    0,  128,     Optional unique 128-bit ID
```

But there is no documentation for this:
1. am I supposed to program this myself?
2. how is this ID used by the system? [I dont see it used anywhere](https://github.com/espressif/esp-idf/search?p=1&q=OPTIONAL_UNIQUE_ID)
3. will ESP-IDF ever get rid of this? I don't want to depend on it, only to find out it has been removed in v4.5

igrr, 2022-04-23:
Currently it is used by esp-rainmaker: https://github.com/espressif/esp-rainmaker/blob/f57a5ec62d6430a4f517de39a9121fc533f7f17d/components/esp_rainmaker/src/core/esp_rmaker_claim.c#L286

We don't have plans to remove this unique ID.

chipweinberger, 2022-04-23:
Thanks =) My google skills failed I should have googled ESP_EFUSE_OPTIONAL_UNIQUE_ID, not just OPTIONAL_UNIQUE_ID.

closed 2022-04-23 by chipweinberger

Emill, 2023-03-23:
In what way is it "optional"? Does optional mean it could for some devices be all-zero? Or does optional just mean it's optional to use the ID in the application?

Also, what bit in EFUSE_WR_DIS does actually protect this field? Is it bit 20 per the Technical Reference Manual, or is it bit 21 per mem_definition.py in espefuse repository?

When looking at that file, why does the other BLOCK2 fields as specified in https://github.com/espressif/esp-idf/blob/master/components/efuse/esp32s3/esp_efuse_table.csv#L154 not match with https://github.com/espressif/esptool/blob/master/espefuse/efuse/esp32s3/mem_definition.py#L232, regarding the bit positions?

Regarding the link to esp-rainmaker, it says in the comments in the code it's for ESP32-S2. But its TRM does not say this field exists...