espressif/esp-idf #8820: DAC audio plays back at incorrect speed in ESP-IDF 4.4

phoddie, 2022-04-22:
## Environment

- Development Kit: M5Stack
- Module or chip used:  ESP32
- IDF version: v4.4
- Build System: CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System: macOS
- Using an IDE?: No

## Problem Description

Audio playback using `i2s_` API to DAC plays back at about 2 half-steps higher than expected. For example, a B-flat tone plays back correctly under ESP-IDF 4.3. Under ESP-IDF 4.4, the same audio plays back as a C natural (two half steps higher).
  
## Code to reproduce this issue

Summary of implementation:

```c
	i2s_config_t i2s_config = {
		.mode = I2S_MODE_MASTER | I2S_MODE_TX | I2S_MODE_DAC_BUILT_IN,
		.sample_rate = 11025,
		.bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
		.channel_format = I2S_CHANNEL_FMT_ONLY_RIGHT,
		.communication_format = I2S_COMM_FORMAT_STAND_MSB,
		.intr_alloc_flags = 0,
		.dma_buf_count = 2,
		.dma_buf_len = 1024,
		.use_apll = false
	};
	i2s_driver_install(0, &i2s_config, 0, NULL);
	i2s_set_dac_mode(I2S_DAC_CHANNEL_BOTH_EN);

	while (moreAudioToPlay)
		i2s_write(0, buffer, 4096, &bytes_written, portMAX_DELAY);
```

## Other information

This issue was first noticed by @meganetaaan. His [report](https://github.com/Moddable-OpenSource/moddable/issues/881#issuecomment-1103943678) includes a video that shows the behavior under ESP-IDF 4.3 and 4.4 so you can easily compare.

L-KAYA, 2022-04-24:
`I2S_CHANNEL_FMT_ONLY_RIGHT` means using only one channel, but `I2S_DAC_CHANNEL_BOTH_EN` enabled two channel, they need to be matched.
Please try to use `I2S_CHANNEL_FMT_ALL_RIGHT` with `I2S_DAC_CHANNEL_BOTH_EN` or `I2S_CHANNEL_FMT_ONLY_RIGHT` with `I2S_DAC_CHANNEL_RIGHT_EN`.

phoddie, 2022-04-26:
@L-KAYA – I made a mistake above. Apologies. The code is using `I2S_CHANNEL_FMT_ONLY_RIGHT` with `I2S_DAC_CHANNEL_RIGHT_EN` on M5Stack Fire. I also tried with `I2S_CHANNEL_FMT_ALL_RIGHT` with `I2S_DAC_CHANNEL_BOTH_EN` but the result is the same.

We are currently pinned to commit `c29343`. I also tried the latest from the `v4.4` tag. That does not play audio at all. just a pop.

L-KAYA, 2025-02-28:
Now dac has a separate driver, the issue will be closed since it is out of dated. If the issue above still exist on the new driver, please feel free to open a new issue, thanks!

closed 2025-02-28 by Alvin1Zhang