espressif/esp-idf #8821: Update CMake

josesimoes, 2022-04-23:
**Is your feature request related to a problem? Please describe.**

Need to be able to user newer CMake version.

**Describe the solution you'd like**

Either update CMake to a newer version (current one is 20.3) or allow overriding it from the PS1 export script.

**Describe alternatives you've considered**

**Additional context**

CMake should be using the latest official version or, at least, one closer to it. The one being distributed with IDF 4.4.1 it's way outdated.

As an alternative it should be possible to easily override this. Currently overriding this is only possible by using the python utility in tools. It should be possible to pass the parameter to the PS1 script.

igrr, 2022-04-23:
Thank you for the reminder @josesimoes. We were going to upgrade the bundled CMake version in relation to https://github.com/espressif/esp-idf/issues/4162. We will speed up the process.

josesimoes, 2022-04-23:
Thanks! looking forward to it as currently I have to go around this with a bunch of hacks.

Nevertheless, please consider the suggestion to be able to pass the prefer-system option in export.ps1

igrr, 2022-04-23:
In fact, I'm not sure why CMake is currently marked as "required" in tools/tools.json when installing on Windows. On macOS and Linux it is "optional". This allows the developer to use their own version of CMake. (Same for ninja.)
We'll see about making cmake and ninja also "optional" on Windows.

closed 2022-05-19 by espressif-bot

frobertopereira, 2022-10-30:
Hi. please, to update Cmake, where can I find the idf_tools.py file?
I'm running the instructions from this tutorial https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-macos-setup.html and got stucked in the update step.
Sorry if it is the wrong topic or too much basic to ask here.