espressif/esp-idf #8822: [docs] esp_efuse_write_field_blob() does not specify which bits are writes

chipweinberger, 2022-04-23:
When you see code that looks like:

uint64_t buf = 0x31
esp_efuse_write_field_blob(EFUSE_MINE, &buf, 10);

or 

uint8_t buf = 0x01
esp_efuse_write_field_blob(EFUSE_MINE, &buf, 3);

Does it write the open portion of the bits (i.e. all zeros) or the bottom portion.

The docs should make this clear.

[image]

chipweinberger, 2022-04-23:
For those that find this thread, this is my understanding:

Xtensa and RISC-V are little endian. 
So while we write numbers like this 0x28FA71, the *bytes* are written in mirrored format. 0x71FA28.

Easier to see in binary:
0b00101000,11111010,01110001 = 0x28FA71

The fuses that get written, are the bytes mirrored.
0b01110001,11111010,00101000

So, if we write 13 bits. We write the first 13 bits of that^ mirrored number:
0b01110001,11111

So to summarize.
- We *arent* simply writing the top bits or the bottom bits of the number
-  We write the lower bytes, but if the # of bits to write is not divisible by 8, we then write only the upper bits within that leftover byte
- easiest solution is to just avoid writing bits not divisble by 8, and then we can say it writes the lowest bytes.

Maybe a  `esp_efuse_write_lowest_x_bits(esp_efuse_desc_t* fuse, (uint64_t)number, (uint8_t) numBits);` helper function would make this more clear.

chipweinberger, 2022-04-23:
@KonstantinKondrashov, does this all sound right?

chipweinberger, 2022-04-23:
like this:

```
#include "esp_efuse.h"

static const char* TAG = "pd efuse";

esp_err_t pd_efuse_write_lowest_x_bits(const esp_efuse_desc_t* field[],  uint64_t value,  uint8_t numBits)
{
    esp_err_t err;

    esp_efuse_desc_t tmp = *(field[0]);
    tmp.bit_count = 1;

    esp_efuse_desc_t* tmpPtr = &tmp;

    for(uint8_t i = 0; i < numBits; i++){

        uint64_t bit = value & (1ULL << i);

        // where to write?
        tmp.bit_start = tmp.bit_start + i;

        if (bit > 0) {
            if (esp_efuse_write_field_bit(&tmpPtr) != ESP_OK){
                return err;
            }
        }
    }

    pd_err_success(error);
    return;
}

uint64_t pd_efuse_read_lowest_x_bits(const esp_efuse_desc_t* field[], 
                                                     uint8_t numBits)
{
    esp_efuse_desc_t tmp = *(field[0]);
    tmp.bit_count = 1;

    esp_efuse_desc_t* tmpPtr = &tmp;

    uint64_t result = 0;

    for(uint8_t i = 0; i < numBits; i++){

        // where to read?
        tmp.bit_start = tmp.bit_start + i;

        bool bit =  esp_efuse_read_field_bit(&tmpPtr);

        if (bit) {
            result |= (1ULL << i);
        }
    }

    return result;
}

```

KonstantinKondrashov, 2022-04-25:
Hi @chipweinberger!
> uint64_t buf = 0x3111
esp_efuse_write_field_blob(EFUSE_MINE, &buf, 10);

You will get only 10 bits written (0x0111).

So while we write numbers like this 0x28FA01 (0b00101000,11111010,00000001).
uint32_t buf = 0x28FA01;
esp_efuse_write_field_blob(EFUSE_MINE, &buf, 24);
It is written by bytes in the order:
byte addr 0: value 0x01 [7...0 bits]
byte addr 1: value 0xFA [15..8 bits]
byte addr 2: value 0x28 [23..16 bits]
At the end in eFuse registers you will see the same order of bits 0b00101000,11111010,00000001.

Your code will not work for efuse blocks which has encoding scheme (3/4 or RS). Do not use it.

chipweinberger, 2022-04-25:
Thanks for the answer @KonstantinKondrashov! So If you  did:

```
uint32_t buf = 0x28FA01; // 0b00101000,11111010,00000001
esp_efuse_write_field_blob(EFUSE_MINE, &buf, 20); // cut off the last 4 bits
```

The fuses would be written like this:
0b1000,11111010,00000001.

> Your code will not work for efuse blocks which has encoding scheme

I've tested it (with RAM fuses) works without issue. I wrap it in a batch. 

```
esp_efuse_batch_write_begin()
pd_efuse_write_lowest_x_bits(EFUSE_TEST,  255,  3);
esp_efuse_batch_write_commit()
```

Am I misunderstanding something? I find my code easier to understand than `write_field_blob()`, so I would prefer to use it.

KonstantinKondrashov, 2022-04-26:
@chipweinberger!
I will update the doc to make the order of bits clear. 

> The fuses would be written like this:
0b1000,11111010,00000001.

Yes, it is correct.

> Am I misunderstanding something?

No, it is ok to use it in batch mode. There is a requirement for eFuses to reduce amount of burn operations this is why we need to wrap burn APIs in esp_efuse_batch_write...().

pd_efuse_write_lowest_bits -> esp_efuse_write_field_bit ->  esp_efuse_write_field_blob(). You just wrap the efuse APIs. Your funcs give the same result (writes bits in order from lowest bit to greatest bit).

chipweinberger, 2022-04-26:
Ah, I did not realize esp_efuse_read_field_blob() called esp_efuse_write_field_blob(). That kind of defeats the purpose of my simplification =)

Thanks for the information @KonstantinKondrashov!

closed 2022-04-26 by chipweinberger