espressif/esp-idf #8823: Some packets cannot be captured in sniffer mode

wgcod, 2022-04-24:
I'm very sorry to reply to the issue submitted before, https://github.com/espressif/esp-idf/issues/5375.
I repeated the problem and tested some results as follows:
[image]
[image]
[image]
now reset
[image]
[image]
reset again
[image]
[image]
[image]

It seems that 'rx state' is incremented, and the others are 0.
Data type packets cannot be captured in the process. They are displayed as QoS data (2:8) with other packet capture tools (rtl8812 or mt7612). However, with the transmission of QoS packets, esp32 returns a large number of misc type packets. Possible reasons?

jack0c, 2022-05-20:
@wgcod to summarize, ESP32 in your side can't receive packets from AP6212A, but from @zhangyanjiaoesp side, it can.
However, with the transmission of QoS packets, esp32 returns a large number of misc type packets.
I think the reasons may be:
1. FCS error, hardware will treat them as misc type packets
2. length error(from the signal part), hardware will treat them as misc type packets

If you have many AP6212A modules, I suggest test every one. And if you can send the bad AP6212A to @zhangyanjiaoesp , it will be the best.

wgcod, 2022-05-30:
However, packets that cannot be received by esp32 can be received by other packet capturing tools. Screenshots of some packets are in the https://github.com/espressif/esp-idf/issues/5375. There are no length errors and FCS errors. And several ap6212a have been tested for the same problem.Can I send you an AP6212A?@zhangyanjiaoesp

zhangyanjiaoesp, 2022-05-30:
> However, packets that cannot be received by esp32 can be received by other packet capturing tools. Screenshots of some packets are in the #5375. There are no length errors and FCS errors. And several ap6212a have been tested for the same problem.Can I send you an AP6212A?@zhangyanjiaoesp

Yes, you can send me an AP6212A, and I will try to reproduce the issue on my side. You can send email to sales@espressif.com, in the email, please describe this issue briefly and provide the github issue link, then my colleague will tell you how to mail the AP6212A ?

closed 2023-03-03 by Alvin1Zhang