espressif/esp-idf #8826: esp-idf blink example fails to build

taguniversalmachine, 2022-04-24:
I am using the ESP32-S3-DevKitC-1 with MacOS 11.6.2 (20G314)

Working from commit 45c1d1cba212b2012d53a55fcf9d338a959c2ece and attempting to build blink example project, I get a failure message in the build.

```
[831/839] Building CXX object esp-idf/...df_asio.dir/asio/asio/src/asio.cpp.obj
ninja: build stopped: subcommand failed.
```

The process seems to find all the tools it needs and seems to be making it most of the way through, I did see this error:
```
../main/main.c: In function 'app_main':
../main/main.c:17:5: error: implicit declaration of function 'tcpip_adapter_init'; did you mean 'spi_flash_init'? [-Werror=implicit-function-declaration]
     tcpip_adapter_init();
     ^~~~~~~~~~~~~~~~~~
     spi_flash_init
```

Attached is full terminal output.

[build_log.txt](https://github.com/espressif/esp-idf/files/8550344/build_log.txt)

igrr, 2022-04-25:
@taguniversalmachine could you please give a link to the example you are building? Blink example in esp-idf doesn't contain calls to tcpip_adapter_init.

The issue likely is that you are trying to build some code written for an earlier version of esp-idf with the master branch of esp-idf. In the master branch, a major release (v5.0) is in development, it contains a number of breaking changes. Among these is removal of the legacy tcpip_adapter APIs. We have a migration guide on this topic: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/tcpip-adapter.html

I think the issue can most likely be solved by switching to an example from the esp-idf copy which you have installed. These examples should work without issues.

taguniversalmachine, 2022-04-25:
Ok I followed the command-line instructions and that worked. I think maybe Eclipse was downloading some other version. Thanks!

closed 2022-04-25 by taguniversalmachine