espressif/esp-idf #8829: Crash during WIFI init, probably related to phy calibration, under very specific conditions

mws-rmain, 2022-04-25:
## Environment:
- Development Kit: none
- Module or chip used: ESP32-WROOM-32U
- IDF version: v4.4
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: external 3.3V

## Problem Description
Bootloader crashes & reboots (most of the time - intermittently may proceed with boot process) under the following conditions:
- sdkconfig has 'ESP_PHY_CALIBRATION_AND_DATA_STORAGE=y'
- partitions.csv does NOT contain a phy_init entry (so there is no defined storage location for phy calibration data)
- I'm using a WROOM-32U module, so antenna is connected to U.FL connector & may be removed. If antenna is present, the crash occurs almost every time. If the antenna is disconnected, WIFI init will complete without crashing.
- under the same conditions, setting 'ESP_PHY_CALIBRATION_AND_DATA_STORAGE=n' in sdkconfig allows wifi init to complete 100% of the time.

I believe the root cause is a poorly handled error condition during PHY calibration. If the system attempts to store PHY calibration data, and no storage is available, the system will crash & reboot instead. If no antenna is connected, PHY calibration code may (?) take a different path that does not attempt to store calibration data.

Clearly this was a mistake on my part to leave 'ESP_PHY_CALIBRATION_AND_DATA_STORAGE=y', when the phy_init entry was removed from the partitions.csv file. However, I would _expect_ the system would simply revert to recalibrating the PHY on each bootup, NOT cause a crash, especially since the resulting crash terminated error logging without recording the cause of the crash (i.e., "no 'phy_init' partition found", etc.)

I'm not including debug logs, as the crash terminates logging before logging the relevant cause.

mws-rmain, 2022-04-25:
Update:
My comments above regarding 'ESP_PHY_CALIBRATION_AND_DATA_STORAGE=n' are incorrect. Disabling this feature results in sdkconfig containing the line "# CONFIG_ESP_PHY_CALIBRATION_AND_DATA_STORAGE is not set" (so the variable is not defined.

Also, I've determined this clearing this setting alone DOES NOT fix the problem.

Below is the log output when antenna is attached (crashes & reboots):
```
I (13) boot: ESP-IDF v4.4 2nd stage bootloader
I (13) boot: compile time 10:43:06
I (13) boot: chip revision: 1
I (14) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (21) boot.esp32: SPI Speed      : 40MHz
I (26) boot.esp32: SPI Mode       : DIO
I (30) boot.esp32: SPI Flash Size : 4MB
I (35) boot: Enabling RNG early entropy source...
I (40) boot: Partition Table:
I (44) boot: ## Label            Usage          Type ST Offset   Length
I (51) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (59) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (66) boot:  2 bootloader       factory app      00 00 00010000 000ea000
I (74) boot:  3 nvsUser          WiFi data        01 02 000fa000 00004000
I (81) boot:  4 nvsFactory       WiFi data        01 02 000fe000 00002000
I (89) boot:  5 sas_app          OTA app          00 10 00100000 00300000
I (96) boot: End of partition table
I (100) boot: Defaulting to factory image
I (105) boot_comm: chip revision: 1, min. application chip revision: 0
I (112) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=18c40h (101440) map
I (158) esp_image: segment 1: paddr=00028c68 vaddr=3ffb0000 size=03950h ( 14672) load
I (163) esp_image: segment 2: paddr=0002c5c0 vaddr=40080000 size=03a58h ( 14936) load
I (170) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=6fee8h (458472) map
I (338) esp_image: segment 4: paddr=0009ff10 vaddr=40083a58 size=12348h ( 74568) load
I (369) esp_image: segment 5: paddr=000b2260 vaddr=50000000 size=00010h (    16) load
I (379) boot: Loaded app from partition at offset 0x10000
I (380) boot: Disabling RNG early entropy source...
I (391) cpu_start: Pro cpu up.
I (391) cpu_start: Starting app cpu, entry point is 0x40081330
I (0) cpu_start: App cpu up.
I (408) cpu_start: Pro cpu start user code
I (408) cpu_start: cpu freq: 160000000
I (408) cpu_start: Application information:
I (412) cpu_start: Project name:     sas-bl
I (417) cpu_start: App version:      v2.0.00_20211222-dirty
I (423) cpu_start: Compile time:     Apr 25 2022 10:42:52
I (429) cpu_start: ELF file SHA256:  d7ddb605c6ffd8b9...
I (435) cpu_start: ESP-IDF:          v4.4
I (440) heap_init: Initializing. RAM available for dynamic allocation:
I (447) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (453) heap_init: At 3FFB9D18 len 000262E8 (152 KiB): DRAM
I (460) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (466) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (472) heap_init: At 40095DA0 len 0000A260 (40 KiB): IRAM
I (480) spi_flash: detected chip: generic
I (483) spi_flash: flash io: dio
I (488) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
E (498) SAS-bl: SAS-bl v3.0.00 Built Apr 25 2022 @10:45:15
I (518) io_config: ioConfig()
I (518) iop_driver: setIOP_DO(0,0,0x80=>0x00)
I (518) iop_driver: setIOP_DO(0,0,0x 1=>0x00)
I (518) io_config: ioConfig() OP_MASK:200A4D034
I (528) gpio: GPIO[2]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (538) gpio: GPIO[4]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (548) gpio: GPIO[5]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (558) gpio: GPIO[12]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (558) gpio: GPIO[14]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (568) gpio: GPIO[15]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (578) gpio: GPIO[18]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (588) gpio: GPIO[21]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (598) gpio: GPIO[23]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (608) gpio: GPIO[33]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
[... 1426 lines cut ...]
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
CONFIG_SEMIHOSTFS_HOST_PATH_MAX_LEN=128
# End of deprecated options
```

igrr, 2022-04-25:
> sdkconfig has 'ESP_PHY_CALIBRATION_AND_DATA_STORAGE=y'
> partitions.csv does NOT contain a phy_init entry (so there is no defined storage location for phy calibration data)

I think there is no contradiction here. Phy_init partition is used to store PHY initialization data. In the absence of such partition, initialization data can be stored in the application binary. Calibration data, on the other hand,  is stored in NVS, which you probably have in the partition table.

Does this issue occur on custom hardware? If yes, does it also occur if you try to run the same code on a development kit? I'm suspecting that a brownout reset might be the cause.

mws-rmain, 2022-04-28:
Thanks for the response, Ivan.
Yes, this is custom hardware. The 3V3 is generated by a buck regulator good for 2A, so capacity should not be an issue (though dynamic response might still be, but see trace image below).
Also, resets occur when the 3V3 reg. supplied from an external 12V 18W power supply, so USB power capacity is not the issue.

I verified there was a 100mV drop for approximately 25us when the reset occurs (on a 200MHz scope). 
I think it is unlikely to be a brownout reset, since the drop is already 'leveling out' (see below).
[image]

In the above image:
- CH1 is pin 3 of the ESP-WROOM-32U module (ENABLE);
- CH4 is pin 2 of the ESP-WROOM-32U module (VDD33).

I did also try disabling the brownout reset (even adding a delay with vTaskDelay()), but the reset persisted.
```
//    WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0);

    ESP_ERROR_CHECK( esp_wifi_start() );

    // disable power saving to ensure near-realtime with no 'anomalous delays'
	esp_wifi_set_ps( WIFI_PS_NONE );

//    vTaskDelay( pdMS_TO_TICKS( 5 ) );

//    WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 1);
```
I can confirm:
- with 'Store phy calibration data in NVS' DISABLED ('# CONFIG_ESP_PHY_CALIBRATION_AND_DATA_STORAGE is not set' in sdkconfig), the reset will always occur if the antenna is connected;
- with the same setting, if the antenna is removed, the boot proceeds normally.
- with 'Store phy calibration data in NVS' ENABLED ('# CONFIG_ESP_PHY_CALIBRATION_AND_DATA_STORAGE=y' in sdkconfig), the reset DOES NOT OCCUR, even if the antenna is connected. Note that this is on the same hardware, with only this change in software.

My partitions.csv does contain an NVS entry - not that it matters since configuring to NOT store PHY calibration in NVS is causing the problem.

I loaded the code on a standard devkit (older ESP-WROOM-32 module with PCB antenna), and it did not reset.
If I can find my defective devkit, I'll replace the module with one from our last production run & report results ASAP.

Please let me know if you have any other suggestions.

mws-rmain, 2022-04-29:
Just a thought: Does the current esp_idf code actually perform a 'phy calibration' when used on a module with PCB antenna? Or is the calibration of the PCB antenna static enough that it is hard-coded? This might explain why code that resets on the 'WROOM-32U' module (with a u.fl connector for external 'unknown' antenna), when the same code works fine on the 'WROOM-32' module (with a 'standard' PCB antenna).

Also, I ran a test where I used an output signal to synchronize traces to code, and found that the 100mV power glitch I showed above was occurring approximately 8ms after invoking esp_wifi_start(). The delay I had added during an initial test that included disabling BOR was not long enough to encompass 8ms. I re-tested with a 10ms delay, and the reset still occurred, but immediately after I re-enabled BOR, even though the 3V3 had fully recovered at that point (actually, it was above 3.3V by tens of millivolts). Does the brownout detector have some hysteresis? 

Regardless, the glitches visible on the externally accessible 3V3 rail do not appear to be excessive enough to warrant a brownout. Is the voltage monitored by the brownout circuit the same voltage, or something internal?

I'm still planning on testing with a current WROOM-32U module on a devkit board - results TBD.

mws-rmain, 2022-05-02:
I've determined at least some of my issue was due to BOR that occurred on another MCU that is part of this design. This other MCU appears to have significant variance in the brownout threshold between parts, so identical code will behave differently on multiple boards. My solution is to disable BO monitoring immediately after booting.

I _could_ re-enable brownout monitoring, if there was:
- some method that would confirm that PHY calibration was complete (ideally, positive confirmation of completion, not just a delay); and
- some confirmation that power consumption during PHY calibration was 'worst case' (meaning the same or worse conditions would NOT occur during normal WiFi operations, and so run the risk of causing a brown-out).

Failing that, I'll just have to leave brownout monitoring disabled.

Thanks for your help.

closed 2022-05-17 by Alvin1Zhang