espressif/esp-idf #8836: [v4.4][ESP32-S3] esp_restart() conflicts with bootloader GPIO outputs

chipweinberger, 2022-04-26:
Using v4.4.1

I use a custom booloader that with GPIOs configured as outputs for LEDs and buzzer.

bootloader_start.c
```
static int select_partition_number(bootloader_state_t *bs)
{
    // turn on Red LED
    esp_rom_gpio_pad_select_gpio(37);
    gpio_ll_output_enable(&GPIO, 37);
    gpio_ll_set_level(&GPIO, 37, 1);
    ....
}
```

It works as expected after a power reset (unplug VCC).

However, when doing esp_restart(), the bootloader is unable to change the LED color.  It is as if the bootloader is unable to use the GPIO as an output.

The GPIOs seem to stays the same value as whatever the app last set them to.

chipweinberger, 2022-04-26:
I've tried calling this but no success.

```
ledc_timer_rst(LEDC_LOW_SPEED_MODE, LEDC_TIMER_1);
ledc_timer_rst(LEDC_LOW_SPEED_MODE, LEDC_TIMER_2);
esp_restart();
```

negativekelvin, 2022-04-26:
Did you try esp_rom_gpio_connect_out_signal

chipweinberger, 2022-04-26:
No. How do I use that? Not familiar with the arguments.

chipweinberger, 2022-04-26:
That worked!

closed 2022-04-26 by chipweinberger