espressif/esp-idf #8837: Harmonic Distortion on low amplitude music

wouter-burns, 2022-04-26:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROOM-32U
- IDF version: v4.4.1
- Build System: idf.py
- Compiler version: 8.4.0
- Operating System: Windows
- Using an IDE?: Espressif IDE V2.4.2
- Power Supply: external 3.3V

## Problem Description

There's audible distortion on low volumes when bluetooth absolute volume is not enabled. This seems to be an issue with SBC decoding, as the same song on an equally low volume does not reproduce this distortion when _Bluetooth Absolute Volume_ is implemented on the sink.

Test signal is a sine of 440 Hz, white line is the bluetooth signal (green is unused)

Bluetooth absolute volume **not** implemented:
Original signal amplitude: -46 dB
Harmonic distortion peaks: -66 dB (audible)
[image]

Bluetooth absolute volume implemented:
Original signal amplitude: -46 dB
Harmonic distortion peaks: -108 dB
[image]

### Expected Behavior
The harmonic peaks should not increase on lower volume SBC streams

### Actual Behavior
The distortion is audible and increases on lower volumes

### Steps to reproduce

1. A2DP sink example
2. Play some music (without bluetooth absolute volume) at -40 dB FS (1% volume)

### Code to reproduce this issue

Changes made from the A2DP example to simulate setting the absolute volume to 1%:

```cpp
// bt_app_core.c

static void bt_i2s_task_handler(void *arg)
{
    size_t item_size     = 0;
    size_t bytes_written = 0;

    for (;;) {
        uint8_t *data = (uint8_t *) xRingbufferReceive(s_databuf, &item_size, (portTickType) portMAX_DELAY);
        if (item_size != 0) {
            for (int i = 0; i < item_size; i += 2) {
                int16_t *temp = (int16_t *) &data[i];
                *temp /= 127;
            }
            i2s_write(0, data, item_size, &bytes_written, portMAX_DELAY);

            vRingbufferReturnItem(s_databuf, (void *) data);
        }
    }
}

// bt_app_av.c

// remove functions:
// volume_set_by_local_host
// volume_change_simulation

```

## Debug Logs

## Other items if possible

**How spectrum analysis was done**
1. Play a sine at 1% volume when connected to the ESP32
2. Put the data on I2S bus
3. I2S data is converted to a Dante Stream
4. Analyse the Dante stream using Rightmark

This seems to be an issue with SBC decoding, as we have a bluetooth player with the same interface, (bluetooth in, I2S out). When we put this signal on a Dante Stream, and analyse this, the signal looks as expected: (also with absolute volume disabled)
[image]

wouter-burns, 2022-04-27:
1. _you played some music at -40 dB FS (1% volume) on source, did you? And why did you divide PCM data by 127 on sink before analyze?_
I did 2 separate tests: 1 with the source set at 0 dB FS, and then dividing the PCM data by 127 (to simulate a bluetooth absolute volume of 1).
The second test is with the source at -40 dBFS but no division of the PCM data. This results in both cases in the same sine amplitude, but the noise is different (spectrum analysis pictures)
The reason I divide the volume myself is because if both connected devices support bluetooth absolute volume, the source will always send the data at full volume, it's up to the sink to do the division. Maybe the bluetooth library implements this, but it didn't seem to work in my case.

2. _Did you do the same thing (divide PCM data by 127) on your compare devices?_
The divide by 127 is the code that resulted in the clean signal, because it received the sinus at full volume. The 'noisy' signal is the result of not dividing, but just giving a low volume signal

4. _-46 dB(or -40dB) is not loud， why did you want reduce it again?_
This is to reliably reproduce an issue I have with quiet parts in a song. It didn't sound right to mee, with a lot of high frequency noise. The low-amplitude sine is to reproduce this easily.

5. _We can do volume adjustment in I2S module, Why did you change the data?_
Because I didn't realise this was possible, is it a function in the AVRCP API?

blueMoods, 2022-04-27:
I have understand your issues, I think the way to decrease the volume by divide 127 is not suitable.

Maybe you can do it at I2S module instead of Bluetooth module. @houhanyan@espressif.com could you please give some advice?

wouter-burns, 2022-04-27:
@blueMoods, just to avoid any confusion:

The distortion happens when I don **not** change any of the code in the example.

And thank you for your quick responses!

blueMoods, 2022-04-29:
@wouter-burns  Thanks for your remind

`This seems to be an issue with SBC decoding, as we have a bluetooth player with the same interface, (bluetooth in, I2S out). When we put this signal on a Dante Stream, and analyse this, the signal looks as expected: (also with absolute volume disabled)`
What is your bluetooth player? we want to compaire it with ESP32.

wouter-burns, 2022-05-02:
The bluetooth player we compared it with is the **Sunitec BM153**.

wouter-burns, 2022-05-16:
Hi, an update from my side. I have been able to do some more tests, on a new PCB revision of our application. The distortion is non-existent now, so I will close this issue.

Original Code:
```cpp
// bt_app_core.c

static uint8_t s_absoluteVolume; // set by the volume command
static void bt_i2s_task_handler(void *arg)
{
    size_t item_size     = 0;
    size_t bytes_written = 0;
    for (;;) {
 		uint8_t *data = (uint8_t *) xRingbufferReceive(s_databuf, &item_size, (portTickType) portMAX_DELAY);
 		if (item_size != 0) {
 			for (int i = 0; i < item_size; i += 2) {
 				int16_t temp = *((int16_t *) (data + i));
 				temp /= 127;
 				temp *= s_absoluteVolume;
 				*((int16_t *) (data + i)) = temp;
 			}
 			i2s_write_expand(BT_I2S_ID, data, item_size, 16, 32, &bytes_written, portMAX_DELAY);
 			vRingbufferReturnItem(s_databuf, (void *) data);
 		}
 	}
}
````

The bug in this part of the code is that I first divide the PCM data by 127, and then multiplying the code by the absolute volume setting (this is a value between 0-127 according to the spec). By first dividing, I lose a lot of accuracy of the PCM value (effectively decimating the 8 LS bits.

For some reason, the code didn't build correctly and was not flashed to the ESP, so I was under the impression that fix didn't do anything.

Improved code:
```cpp
// bt_app_core.c

static uint8_t s_absoluteVolume; // set by the volume command
static void bt_i2s_task_handler(void *arg)
{
    size_t item_size     = 0;
    size_t bytes_written = 0;
    for (;;) {
 		uint8_t *data = (uint8_t *) xRingbufferReceive(s_databuf, &item_size, (portTickType) portMAX_DELAY);
 		if (item_size != 0) {
 			for (int i = 0; i < item_size; i += 2) {
 				int32_t temp = *((int16_t *) (data + i));
 				temp *= s_absoluteVolume;
 				temp /= 127;
 				*((int16_t *) (data + i)) = (int16_t) temp;
 			}
 			i2s_write_expand(BT_I2S_ID, data, item_size, 16, 32, &bytes_written, portMAX_DELAY);
 			vRingbufferReturnItem(s_databuf, (void *) data);
 		}
 	}
}
````

closed 2022-05-16 by wouter-burns