espressif/esp-idf #8840: Release v4.4 cannot use std::map in header file.

lanistor, 2022-04-26:
We are using c++ in our esp32 project, when we use `<map>` of libc++, error occurs:

```
r/Hardware/main/modules/include/../common/utils/./time.h:1:10: fatal error: map: No such file or directory
 #include <map>
          ^~~~~
compilation terminated.
```
When we change `time.h` to `time.hpp`, the error will disappear.
Source code is simple:
```
#include <map>
#include <string>
```

Root CmakeList.txt
```
cmake_minimum_required(VERSION 3.5)

if (NOT DEFINED PROJECT_NAME)
    include_directories(main/include)
    include_directories(main/modules/include)
    include($ENV{IDF_PATH}/tools/cmake/project.cmake)
    project(test)
else()
    message(FATAL_ERROR "LV PORT ESP32: This must be a project's main CMakeLists.txt.")
endif()
```

lanistor, 2022-04-26:
> 您能否发布构建系统报告的完整错误？它通常从失败的编译命令的命令行开始。
> 
> > 当我们将 time.h 更改为 time.hpp 时，错误就会消失。
> 
> 这很可能意味着您的`time.h`标头隐藏了标准 time.h 标头，其中定义了 C 标准时间函数（https://en.cppreference.com/w/c/chrono）。`time.h`包含在许多 C 文件中。`map`编译 C 代码时 header 不可用，因此会发生错误。
> 
> 我认为您应该考虑仅将项目的包含标头添加到组件中，而不是整个构建（包括所有 IDF 组件）。基本上，相反`include_directories`，尝试使用`target_include_directories`将包含目录附加到特定目标。
> 
> 避免与标准 C 库头文件的名称冲突也是一种解决方案。

Thanks for reply. I changed `time.h` to `time_utils.h`, the error is still there.

lanistor, 2022-04-26:
> Could you please post the complete error reported by the build system? It usually starts with the command line of the compilation command which has failed.
> 
> > When we change time.h to time.hpp, the error will disappear.
> 
> It most likely means that your `time.h` header hides the standard time.h header where C standard time functions are defined (https://en.cppreference.com/w/c/chrono). `time.h` is included from a number of C files. `map` header is not available when compiling C code, therefore the error occurs.
> 
> I think you should consider adding your project's include headers to your components only, not to the whole build (which includes all IDF components). Basically, instead `include_directories`, try using `target_include_directories` to attach include directories to the specific targets.
> 
> Avoiding collision with names of standard C library headers can also be a solution.

Thanks for your reply. I changed `time.h` to `time_utils.h`, the error is still there.

igrr, 2022-04-26:
Thanks for trying this! There seems to be no issue with the files you have shown in the first post (time.h and CMakeLists.txt). Unfortunately many details about your project are still unclear: where is time.h included from, how are the components structured. Could you please attach (or upload to Github) some small project which demonstrates the issue? I'm afraid it will be difficult to help you troubleshoot this issue without having some code to reproduce it.

lanistor, 2022-04-26:
@igrr I had resolved it, the reason is i named a file as `utils.h`, after i changed it to `xx_utils.h`, the problem resolved.

Thanks a lot.

closed 2022-04-26 by lanistor