espressif/esp-idf #8842: Any way to enable encryption along with FATFS?

Huckies, 2022-04-26:
Hi there, I'm using a ESP32-S3-DevKitM-1.
A flash encryption in development mode is enabled, as well as the secure boot V2. Here is the summary.
```
=== Run "summary" command ===
EFUSE_NAME (Block) Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Config fuses:
DIS_ICACHE (BLOCK0)                                Disables ICache                                    = False R/W (0b0)
DIS_DCACHE (BLOCK0)                                Disables DCache                                    = False R/W (0b0)
DIS_DOWNLOAD_ICACHE (BLOCK0)                       Disables Icache when SoC is in Download mode       = False R/W (0b0)
DIS_DOWNLOAD_DCACHE (BLOCK0)                       Disables Dcache when SoC is in Download mode       = False R/W (0b0)
DIS_FORCE_DOWNLOAD (BLOCK0)                        Disables forcing chip into Download mode           = False R/W (0b0)
DIS_CAN (BLOCK0)                                   Disables the TWAI Controller hardware              = False R/W (0b0)
DIS_APP_CPU (BLOCK0)                               Disables APP CPU                                   = False R/W (0b0)
FLASH_TPUW (BLOCK0)                                Configures flash startup delay after SoC power-up, = 0 R/W (0x0)
                                                    unit is (ms/2). When the value is 15, delay is 7.
                                                   5 ms                                              
DIS_LEGACY_SPI_BOOT (BLOCK0)                       Disables Legacy SPI boot mode                      = True R/W (0b1)
UART_PRINT_CHANNEL (BLOCK0)                        Selects the default UART for printing boot msg     = UART0 R/W (0b0)
FLASH_ECC_MODE (BLOCK0)                            Configures the ECC mode for SPI flash             
   = 16-byte to 18-byte mode R/W (0b0)
DIS_USB_DOWNLOAD_MODE (BLOCK0)                     Disables USB OTG download feature in UART download = False R/W (0b0)
                                                    boot mode                                        
UART_PRINT_CONTROL (BLOCK0)                        Sets the default UART boot message output mode     = Enabled R/W (0b00)
FLASH_TYPE (BLOCK0)                                Selects SPI flash type                             = 4 data lines R/W (0b0)
FLASH_PAGE_SIZE (BLOCK0)                           Sets the size of flash page                        = 0 R/W (0b00)
FLASH_ECC_EN (BLOCK0)                              Enables ECC in Flash boot mode                     = False R/W (0b0)
FORCE_SEND_RESUME (BLOCK0)                         Forces ROM code to send an SPI flash resume comman = False R/W (0b0)
                                                   d during SPI boot                                 
BLOCK_USR_DATA (BLOCK3)                            User data                                         
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 

Efuse fuses:
WR_DIS (BLOCK0)                                    Disables programming of individual eFuses          = 109055233 R/W (0x06800d01)
RD_DIS (BLOCK0)                                    Disables software reading from BLOCK4-10           = 12 R/- (0b0001100)

Identity fuses:
SECURE_VERSION (BLOCK0)                            Secure version (used by ESP-IDF anti-rollback feat = 0 R/W (0x0000)
                                                   ure)                                              
MAC (BLOCK1)                                       Factory MAC Address                               
   = 60:55:f9:f5:c1:00 (OK) R/W 
WAFER_VERSION (BLOCK1)                             WAFER version                                      = 1 R/W (0b001)
PKG_VERSION (BLOCK1)                               ??? Package version                                = ESP32-S3 R/W (0x0)
BLOCK1_VERSION (BLOCK1)                            ??? BLOCK1 efuse version                           = 5 R/W (0b101)
OPTIONAL_UNIQUE_ID (BLOCK2)                        ??? Optional unique 128-bit ID                    
   = 43 9f b4 d0 9d 1c cc f7 a0 ca 58 aa f4 9f 8a 6b R/W 
BLOCK2_VERSION (BLOCK2)                            ??? Version of BLOCK2                              = 7 R/W (0b111)
CUSTOM_MAC (BLOCK3)                                Custom MAC Address                                
   = 00:00:00:00:00:00 (OK) R/W 

Security fuses:
SOFT_DIS_JTAG (BLOCK0)                             Software disables JTAG by programming odd number o = 7 R/W (0b111)
                                                   f 1 bit(s). JTAG can be re-enabled via HMAC periph
                                                   eral                                              
HARD_DIS_JTAG (BLOCK0)                             Hardware disables JTAG permanently                 = True R/W (0b1)
DIS_DOWNLOAD_MANUAL_ENCRYPT (BLOCK0)               Disables flash encryption when in download boot mo = False R/W (0b0)
                                                   des                                               
SPI_BOOT_CRYPT_CNT (BLOCK0)                        Enables encryption and decryption, when an SPI boo = Enable R/W (0b001)
                                                   t mode is set. Enabled when 1 or 3 bits are set,di
                                                   sabled otherwise                                  
SECURE_BOOT_KEY_REVOKE0 (BLOCK0)                   Revokes use of secure boot key digest 0            = False R/W (0b0)
SECURE_BOOT_KEY_REVOKE1 (BLOCK0)                   Revokes use of secure boot key digest 1            = True R/W (0b1)
SECURE_BOOT_KEY_REVOKE2 (BLOCK0)                   Revokes use of secure boot key digest 2            = True R/W (0b1)
KEY_PURPOSE_0 (BLOCK0)                             KEY0 purpose                                       = SECURE_BOOT_DIGEST0 R/- (0x9)
KEY_PURPOSE_1 (BLOCK0)                             KEY1 purpose                                       = USER R/W (0x0)
KEY_PURPOSE_2 (BLOCK0)                             KEY2 purpose                                       = XTS_AES_256_KEY_1 R/- (0x2)
KEY_PURPOSE_3 (BLOCK0)                             KEY3 purpose                                       = XTS_AES_256_KEY_2 R/- (0x3)
KEY_PURPOSE_4 (BLOCK0)                             KEY4 purpose                                       = USER R/W (0x0)
KEY_PURPOSE_5 (BLOCK0)                             KEY5 purpose                                       = USER R/W (0x0)
SECURE_BOOT_EN (BLOCK0)                            Enables secure boot                                = True R/W (0b1)
[... 79 lines cut ...]
composite_no_cam,app,ota_3,0x300000,1M,
controller,app,ota_4,0x400000,1M,
xinput,app,ota_5,0x500000,1M,
storage,data,fat,0x600000,2M,encrypted
*******************************************************************************
```
I attempt to store some files on ``storage`` sector using FATFS and the standard file operating functions. The initial data is provided by the ``profile`` app (compiled into binary) and used by other apps.

Something wrong happens after my turning on flash encryption. More specifically, if I build all the binaries of apps and manually encrypt them, merge them into an integrated binary (8MB in size) and them flash it, the ``storage`` partition can still be initialized normally but it can no longer be mounted using ``esp_vfs_fat_spiflash_mount`` and returns an error code ``0xffffffff``.

I followed the guide here [https://docs.espressif.com/projects/esp-idf/en/v4.4.1/esp32s3/security/flash-encryption.html#manually-encrypting-files](url)
```
espsecure.py encrypt_flash_data --aes_xts --keyfile ./HucKeys_encryption_key.bin --address 0x30000 --output HucKeys_profile_en.bin ./build/HucKeys.bin && \
espsecure.py encrypt_flash_data --aes_xts --keyfile ./HucKeys_encryption_key.bin --address 0x100000 --output HucKeys_boot_en.bin ./build/HucKeys.bin && \
...
espsecure.py encrypt_flash_data --aes_xts --keyfile ./HucKeys_encryption_key.bin --address 0x0 --output HucKeys_bootloader_en.bin ./build/bootloader/bootloader.bin && \
espsecure.py encrypt_flash_data --aes_xts --keyfile ./HucKeys_encryption_key.bin --address 0x1f000 --output HucKeys_partition_table_en.bin ./build/partition_table/partition-table.bin && \
[... cut ...]
esptool.py -p /dev/cu.usbmodem111201 -b 921600 --no-stub --before default_reset --after hard_reset --chip esp32s3 write_flash --flash_mode dio --flash_freq 80m --flash_size 8MB --verify 0x0 HucKeys_en.bin
```

But if I use ``idf.py encrypted-flash`` (uart download mode is set to ``Enabled``) and let the ESP32-S3 do the encryption job, ``storage`` can be accessed like unencrypted.

So what's the way to correctly access an encrypted fat partition, while using binary files manually encrypted before flashing?

igrr, 2022-04-26:
@Huckies could you please clarify how you are generating the initial content of the `storage` partition?

If you are encrypting all partitions on the host and generating a merged binary, I think you should also encrypt the storage partition binary and include it into the merge_bin arguments list.
If you don't include the storage partition binary when calling esptool.py merge_bin, you will simply overwrite the filesystem with empty space (0xff).

Huckies, 2022-04-26:
> @Huckies could you please clarify how you are generating the initial content of the `storage` partition?
>
> If you are encrypting all partitions on the host and generating a merged binary, I think you [...]

Thanks for reply so fast. I set a task to check if partition ``storage`` is formatted (by checking whether the last two bytes of sector 1 is 0x55 and 0xAA) if not, format and fill it with data from a constant array compiled into the .bin file.

Here is the functions I used in initializing:
``esp_partition_find_first()`` to locate the ``storage`` partition.
``wl_mount()``
``wl_erase_range()`` to erase the whole ``storage`` partition.
Finally use ``wl_write()`` to fill some blocks from start, one by one.

I understand that ``--merge_bin`` simply fills ``0xFF`` to untouched bytes, since the operations I listed above is always a success. It's the failure in mounting FATFS to read files leads me here.

Huckies, 2022-04-26:
So I'm going to need somehow create a binary file with garbage bytes and encrypt it before calling ``"merge_bin"`` to tell the ESP32-S3 this is an encrypted partition? Or it'll be treated unencrypted even with the ``encrypted`` flag in partition table?

Huckies, 2022-04-27:
@igrr Is there any possibility that the flash encryption leads to a longer period physical writing data? I've set wear leveling to mode 'safety' so there might be a chance to get the writing procedure broke and causing a damaged partition, finally leads to the failure of ``esp_vfs_fat_spiflash_mount()`` ?

igrr, 2022-04-27:
Thank you for clarification. I think what you are trying to do should work, in principle. The process is pretty complicated though, so pinpointing where an issue might be is not trivial. I would recommend to split the process into two parts and find out which part is not behaving as expected. Then split again, repeating until the issue is located.

In this case, you seem to have two stages: 1) writing the initial partition image into the storage partition, and 2) mounting the partition in another app.

Here's what you can do to diagnose each part independently:

1. For the first part, after the partition is filled with the initial data, don't reboot into the other app. Use esptool.py to read back the storage partition content, then use espsecure.py to decrypt it. Compare it with the original data you were trying to write.
2. For the 2nd part, try writing the initial storage partition content together with the rest of the partitions. I.e. encrypt the initial image of the storage partition, add it to the merge_bin arguments, and flash it along with other binaries. Then run the application which tries to mount the filesystem.

Once you know which of the two parts is not working correctly, we can make the next step to troubleshoot this.

Huckies, 2022-04-27:
I'm almost 100% sure it is the first stage that caused the problem.
The reason why ``idf.py encrypted-flash`` succeeds is the former program has formatted ``storage`` partition once, so it's invariable to format it again which largely improves the probability of success.

By the way when I use ``esptool.py --no-stub read_flash 0x600000 0x200000 storage_dump.bin`` it returns
```
Serial port /dev/cu.usbserial-11120
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: 
Enabling default SPI flash mode...

A fatal error occurred: Failed to read flash block (result was 01090000: CRC or checksum was invalid)
```

igrr, 2022-04-27:
> I'm almost 100% sure it is the first stage that caused the problem.

Could you share both binaries (what was expected to be in flash, and what was actually in flash) based on my previous suggestion to read back and decrypt the flash contents? This will help figure out why the result is incorrect.

Huckies, 2022-04-27:
@igrr Sorry I didn't make myself clear, it fails with command ``esptool.py --no-stub read_flash 0x600000 0x200000 storage_dump.bin`` so I don't make it to read back the binaries.

Huckies, 2022-04-28:
@radimkarnis 
```
esptool.py v3.3-dev
Found 3 serial ports
Serial port /dev/cu.wlan-debug
/dev/cu.wlan-debug failed to connect: [Errno 16] could not open port /dev/cu.wlan-debug: [Errno 16] Resource busy: '/dev/cu.wlan-debug'
Serial port /dev/cu.usbserial-11120
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.105 No serial data received.
.TRACE +0.057 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.011 Read 1 bytes: c0
TRACE +0.000 Read 63 bytes: 
    0108040007071220 00000000c0c00108 | ....... ........
    0400070712200000 0000c0c001080400 | ..... ..........
    0707122000000000 c0c0010804000707 | ... ............
    122000000000c0c0 01080400070712   | . .............
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.006 Read 1 bytes: 20
TRACE +0.000 Read 47 bytes: 
    00000000c0c00108 0400070712200000 | ............. ..
    0000c0c001080400 0707122000000000 | ........... ....
    c0c0010804000707 122000000000c0   | ......... .....
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.000 Received full packet: 010804000707122000000000

Detecting chip type...TRACE +0.000 command op=0x14 data len=0 wait_response=1 timeout=3.000 data=
TRACE +0.000 Write 10 bytes: c00014000000000000c0
TRACE +0.006 Read 1 bytes: c0
TRACE +0.000 Read 33 bytes: 
    0114180000000000 f100000001090002 | ................
    0300000c09000000 0000000000000000 | ................
    c0                                | .
TRACE +0.000 Received full packet: 
    0114180000000000 f100000001090002 | ................
    0300000c09000000 0000000000000000 | ................
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=4cf1ce3f
TRACE +0.000 Write 14 bytes: c0000a0400000000004cf1ce3fc0
TRACE +0.005 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010a04000000000000000000c0
TRACE +0.000 Received full packet: 010a04000000000000000000
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=00100040
TRACE +0.000 Write 14 bytes: c0000a04000000000000100040c0
TRACE +0.005 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010a04000900000000000000c0
TRACE +0.000 Received full packet: 010a04000900000000000000
 ESP32-S3
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=44700060
TRACE +0.000 Write 14 bytes: c0000a04000000000044700060c0
[... 648 lines cut ...]
TRACE +0.000 Read 13 bytes: 010704005560000000000000c0
TRACE +0.000 Received full packet: 010704005560000000000000
Running stub...
TRACE +0.000 command op=0x06 data len=8 wait_response=1 timeout=0.050 data=0000000078893740
TRACE +0.000 Write 18 bytes: 
    c000060800000000 0000000000788937 | .............x.7
    40c0                              | @.
TRACE +0.004 Read 1 bytes: e0
TRACE +0.000 Read invalid data: e0
TRACE +0.000 Remaining data in serial buffer: 
Traceback (most recent call last):
  File "/Users/huckies/esp-idf/components/esptool_py/esptool/esptool.py", line 5347, in <module>
    _main()
  File "/Users/huckies/esp-idf/components/esptool_py/esptool/esptool.py", line 5340, in _main
    main()
  File "/Users/huckies/esp-idf/components/esptool_py/esptool/esptool.py", line 4670, in main
    esp = esp.run_stub()
  File "/Users/huckies/esp-idf/components/esptool_py/esptool/esptool.py", line 887, in run_stub
    p = self.read()
  File "/Users/huckies/esp-idf/components/esptool_py/esptool/esptool.py", line 413, in read
    return next(self._slip_reader)
StopIteration
```

Please see above. Although I've set ``UART ROM download mode`` to ``(Enabled (not recommended))`` any commands without ``--no-stub`` option fails and prints an error message ends with ``StopIteration``.