espressif/esp-idf #8843: erase-otadata failed

chegewara, 2022-04-26:
Master branch, just updated.
```
Executing action: erase-otadata
Serial port /dev/ttyUSB0
Connecting...
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP32
Running ninja in directory /home/chegewara/esp/master/examples/get-started/hello_world/build
Executing "ninja erase-otadata"...
ninja: error: unknown target 'erase-otadata'
ninja failed with exit code 1
```

dobairoland, 2022-04-27:
Hi @chegewara. Are you running this in the unmodified `get-started/hello_world` example? Ota commands should not work there because there is no OTA partition defined in that project. 

But unfortunately, there is another issue which would you get for an OTA project. We try to fix this soon. The `master` branch is affected. v4.4 should work fine. I'm sorry for the inconvenience.

KaeLL, 2022-04-27:
> there is another issue which would you get for an OTA project. We try to fix this soon. The master branch is affected. 

@dobairoland could you link or detail the issue?

dobairoland, 2022-04-27:
It is not tracked on github. We've noticed that issuing for example `idf.py erase-otadata` prints only a couple of lines but nothing happens. I need to investigate more. I'll link the fix here and keep open this one until then.

dobairoland, 2022-05-20:
The fix is in progress. It should make into v5.0. No released versions are affected. The issue is present only in the master branch.

closed 2022-05-29 by espressif-bot