espressif/esp-idf #8845: assert failed: esp_dport_access_int_init dport_access.c:178 (res == pdTRUE)

vogelfreiheit, 2022-04-27:
## Environment

- Development Kit:      ESP32-Wrover-Kit, ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROOM-32 and ESP32-PICO-D4|ESP32
- IDF version (run ``git describe --tags`` to find it): assert failed: esp_dport_access_int_init dport_access.c:178 (res == pdTRUE)
- Build System:        CMake/PlatformIO
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):

PACKAGES: 
 - framework-espidf 3.40302.0 (4.3.2) 
 - tool-cmake 3.16.4 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - tool-ninja 1.7.1 
 - toolchain-esp32ulp 1.22851.191205 (2.28.51) 
 - toolchain-xtensa-esp32 8.4.0+2021r2-patch3

- Operating System:     Linux
- Using an IDE?: Yes (Pio)
- Power Supply:        USB

## Problem Description

Boot fails with the following log:

```
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7548
ho 0 tail 12 room 4
load:0x40078000,len:14912
load:0x40080400,len:3688
entry 0x4008067c
␛[0;32mI (29) boot: ESP-IDF 4.3.2 2nd stage bootloader␛[0m
␛[0;32mI (29) boot: compile time 03:05:53␛[0m
␛[0;32mI (29) boot: chip revision: 1␛[0m
␛[0;32mI (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0␛[0m
␛[0;32mI (39) boot.esp32: SPI Speed      : 40MHz␛[0m
␛[0;32mI (44) boot.esp32: SPI Mode       : DIO␛[0m
␛[0;32mI (49) boot.esp32: SPI Flash Size : 4MB␛[0m
␛[0;32mI (53) boot: Enabling RNG early entropy source...␛[0m
␛[0;32mI (58) boot: Partition Table:␛[0m
␛[0;32mI (62) boot: ## Label            Usage          Type ST Offset   Length␛[0m
␛[0;32mI (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000␛[0m
␛[0;32mI (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000␛[0m
␛[0;32mI (84) boot:  2 factory          factory app      00 00 00010000 00100000␛[0m
␛[0;32mI (92) boot: End of partition table␛[0m
␛[0;32mI (96) boot_comm: chip revision: 1, min. application chip revision: 0␛[0m
␛[0;32mI (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=19498h (103576) map␛[0m
␛[0;32mI (149) esp_image: segment 1: paddr=000294c0 vaddr=3ffb0000 size=04168h ( 16744) load␛[0m
␛[0;32mI (156) esp_image: segment 2: paddr=0002d630 vaddr=40080000 size=029e8h ( 10728) load␛[0m
␛[0;32mI (161) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=79d3ch (499004) map␛[0m
␛[0;32mI (344) esp_image: segment 4: paddr=000a9d64 vaddr=400829e8 size=15404h ( 87044) load␛[0m
␛[0;32mI (380) esp_image: segment 5: paddr=000bf170 vaddr=50000000 size=00010h (    16) load␛[0m
␛[0;32mI (392) boot: Loaded app from partition at offset 0x10000␛[0m
␛[0;32mI (392) boot: Disabling RNG early entropy source...␛[0m
␛[0;32mI (404) cpu_start: Pro cpu up.␛[0m
␛[0;32mI (404) cpu_start: Starting app cpu, entry point is 0x40082d68␛[0m
␛[0;32mI (390) cpu_start: App cpu up.␛[0m
␛[0;32mI (420) cpu_start: Pro cpu start user code␛[0m
␛[0;32mI (420) cpu_start: cpu freq: 160000000␛[0m
␛[0;32mI (420) cpu_start: Application information:␛[0m
␛[0;32mI (425) cpu_start: Project name:     XXX␛[0m
␛[0;32mI (430) cpu_start: App version:      d4871cf-dirty␛[0m
␛[0;32mI (436) cpu_start: Compile time:     Apr 27 2022 03:05:34␛[0m
␛[0;32mI (442) cpu_start: ELF file SHA256:  069c75255dd8cbd2...␛[0m
␛[0;32mI (448) cpu_start: ESP-IDF:          4.3.2␛[0m
␛[0;32mI (453) heap_init: Initializing. RAM available for dynamic allocation:␛[0m
␛[0;32mI (460) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM␛[0m
␛[0;32mI (466) heap_init: At 3FFDB1D8 len 00004E28 (19 KiB): DRAM␛[0m
␛[0;32mI (472) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM␛[0m
␛[0;32mI (478) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM␛[0m
␛[0;32mI (485) heap_init: At 40097DEC len 00008214 (32 KiB): IRAM␛[0m
␛[0;32mI (492) spi_flash: detected chip: gd␛[0m
␛[0;32mI (495) spi_flash: flash io: dio␛[0m
␛[0;32mI (500) cpu_start: Starting scheduler on PRO CPU.␛[0m

assert failed: esp_dport_access_int_init dport_access.c:178 (res == pdTRUE)

Backtrace:0x40083067:0x3ffe7bd0 0x4008ac91:0x3ffe7bf0 0x400924d5:0x3ffe7c10 0x400dfa7a:0x3ffe7d30 0x40149998:0x3ffe7d60

```

### Expected Behavior

Boot completes into the actual application code.

### Actual Behavior

Failure to boot.

VojtechBartoska, 2022-04-27:
Hello @vogelfreiheit,

thanks for your contribution.

Do you use only ESP-IDF or you are working with Arduino as ESP-IDF component?

vogelfreiheit, 2022-04-27:
Only ESP-IDF.

I confirmed this is related to extensive program memory being used. For example, statically defined arrays.
Before hand-off to the application, the bootloader tries to accommodate the image in RAM... and somehow it fails to properly do so.

It's weird because it sometimes manifests and then it doesn't, depending on compilation options and so forth.
145k is where things got strange.

The error message however isn't helpful at all.

I also verified that with malloc it happens too, but obviously you can catch the error and handle it as you see fit.

Alvin1Zhang, 2022-10-20:
Thanks for reporting, would you please help share the test code so that we could recreate the issue? Thanks.

closed 2023-03-08 by Alvin1Zhang