espressif/esp-idf #8847: ESP-LCD: swap_color_bytes doesn't work on ESP32-S3

FanhuaCloud, 2022-04-27:
## Environment

- Development Kit:     none
- Kit version (for WroverKit/PicoKit/DevKitC): none
- Module or chip used:  ESP32-S3
- IDF version (run ``git describe --tags`` to find it):
    // v5.0-dev-2585-ga82e6e63d
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:    Windows
- (Windows only) environment type: PowerShell.
- Using an IDE?: Yes Vscode
- Power Supply:        USB

## Problem Description
I am currently using the st7796 screen, which has similar commands to the 7789, when I use the official i80 demo, as long as I turn off lvgl's swap color, the color of the screen will look weird
//Detailed problem description goes here.

### Expected Behavior
The screen is normal
### Actual Behavior
The colors displayed on the screen are strange (see image below)
[image]

### Steps to reproduce

1. use this example：https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd/i80_controller
2. set screen to st7789
3. start

suda-morris, 2022-04-28:
@FanhuaCloud  I just checked st7789 (8bit data width) with that example and works well. By default, the example utilizes the LVGL to swap the color bytes, i.e. the `CONFIG_LV_COLOR_16_SWAP` option. But ESP32S3's LCD controller has a hardware feature that can help to do that in hardware. You can enable that feature by 
```
--- a/examples/peripherals/lcd/i80_controller/main/i80_controller_example_main.c
+++ b/examples/peripherals/lcd/i80_controller/main/i80_controller_example_main.c
@@ -157,6 +157,7 @@ void app_main(void)
         .user_ctx = &disp_drv,
         .lcd_cmd_bits = EXAMPLE_LCD_CMD_BITS,
         .lcd_param_bits = EXAMPLE_LCD_PARAM_BITS,
+        .flags.swap_color_bytes = true,
     };
     ESP_ERROR_CHECK(esp_lcd_new_panel_io_i80(i80_bus, &io_config, &io_handle));
```

Don't forget to turn off the `CONFIG_LV_COLOR_16_SWAP` as well.

If you're using ST7796 with 16bit data width, then you don't have to enable the "swap bytes" feature either by `swap_color_bytes` flag or the `CONFIG_LV_COLOR_16_SWAP`.

FanhuaCloud, 2022-04-28:
oh thank you, i tried similar code before, i realized i may have written the code wrong, everything works fine so far

FanhuaCloud, 2022-04-28:
[image]

closed 2022-04-28 by FanhuaCloud