espressif/esp-idf #8848: Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

diplfranzhoepfinger, 2022-04-27:
i get a 
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

with the following code: 
https://github.com/DiplomInformatikerFranzHoepfinger/test_kafka/commit/4bdefa1371b4d955d1944f5b0cc650a724dda285

diplfranzhoepfinger, 2022-04-27:
going deeper i guess that this for-loop is the Reason: 

https://github.com/DiplomInformatikerFranzHoepfinger/test_kafka/blob/master/TestKafka/components/librdkafka/rdkafka_conf.c#L2152

diplfranzhoepfinger, 2022-04-27:
with:

DiplomInformatikerFranzHoepfinger/test\_kafka@f8ac9c5

i get a printout: 

```
I (317) kafka: prop->name =  builtin.features
```

which means that the Access to 

rd\_kafka\_properties\[0\].name

is valid.

diplfranzhoepfinger, 2022-04-27:
extending this a bit: 

gives:

```shell
I (319) kafka: prop->name =  builtin.features
I (329) kafka: prop->name =  client.id
I (329) kafka: prop->name =  client.software.name
I (329) kafka: prop->name =  client.software.version
I (339) kafka: prop->name =  metadata.broker.list
I (349) kafka: prop->name =  bootstrap.servers
I (349) kafka: prop->name =  message.max.bytes
I (359) kafka: prop->name =  message.copy.max.bytes
I (359) kafka: prop->name =  receive.message.max.bytes
```

which means: 

```c
    ESP_LOGI(TAG, "prop->name =  %s", rd_kafka_properties[0].name);
    ESP_LOGI(TAG, "prop->name =  %s", rd_kafka_properties[1].name);
    ESP_LOGI(TAG, "prop->name =  %s", rd_kafka_properties[2].name);
    ESP_LOGI(TAG, "prop->name =  %s", rd_kafka_properties[3].name);
    ESP_LOGI(TAG, "prop->name =  %s", rd_kafka_properties[4].name);
    ESP_LOGI(TAG, "prop->name =  %s", rd_kafka_properties[5].name);
    ESP_LOGI(TAG, "prop->name =  %s", rd_kafka_properties[6].name);
    ESP_LOGI(TAG, "prop->name =  %s", rd_kafka_properties[7].name);
    ESP_LOGI(TAG, "prop->name =  %s", rd_kafka_properties[8].name);
```

seems to be valid as well ...

diplfranzhoepfinger, 2022-04-27:
replace this with a for-loop, and the LoadProhibited is back; 

see: DiplomInformatikerFranzHoepfinger/test\_kafka@a880455

diplfranzhoepfinger, 2022-04-27:
crazy: 

in a non-loop config it works: 

DiplomInformatikerFranzHoepfinger/test\_kafka@dddaeaa

KaeLL, 2022-04-27:
[Good luck](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html#loadprohibited-storeprohibited)

Alvin1Zhang, 2022-04-28:
Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

diplfranzhoepfinger, 2022-04-28:
Sorry, here are the missing Information: 

## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROVER]
- IDF version (run ``git describe --tags`` to find it):
    // v4.4
- Build System:         [CMake & idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)
- Operating System:     [Linux]
- Using an IDE?: [Yes eclipse]
- Power Supply:         [USB]

## Problem Description

see above.

log output: 

### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. clone https://github.com/DiplomInformatikerFranzHoepfinger/test_kafka
2. ...

### Code to reproduce this issue

```cpp
```

## Debug Logs

```
Executing action: monitor
Running idf_monitor in directory /home/franz/git/franz/test_kafka/TestKafka
Executing "/home/franz/.espressif/python_env/idf4.4_py3.8_env/bin/python /home/franz/esp-idf-v4.4/tools/idf_monitor.py -p /dev/ttyUSB0 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 0 --print_filter  /home/franz/git/franz/test_kafka/TestKafka/build/TestKafka.elf -m '/home/franz/.espressif/python_env/idf4.4_py3.8_env/bin/python' '/home/franz/esp-idf-v4.4/tools/idf.py'"...
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6612
load:0x40078000,len:14780
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (27) boot: ESP-IDF v4.4-dirty 2nd stage bootloader
I (27) boot: compile time 20:48:24
I (27) boot: chip revision: 3
I (30) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (38) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 2MB
I (51) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00100000
I (90) boot: End of partition table
I (94) boot_comm: chip revision: 3, min. application chip revision: 0
I (101) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=2948ch (169100) map
I (171) esp_image: segment 1: paddr=000394b4 vaddr=3ffb0000 size=028bch ( 10428) load
I (175) esp_image: segment 2: paddr=0003bd78 vaddr=40080000 size=042a0h ( 17056) load
I (184) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=2b760h (178016) map
I (250) esp_image: segment 4: paddr=0006b788 vaddr=400842a0 size=073ach ( 29612) load
I (262) esp_image: segment 5: paddr=00072b3c vaddr=50000000 size=00010h (    16) load
I (268) boot: Loaded app from partition at offset 0x10000
I (268) boot: Disabling RNG early entropy source...
I (282) cpu_start: Pro cpu up.
I (283) cpu_start: Starting app cpu, entry point is 0x4008118c
0x4008118c: call_start_cpu1 at /home/franz/esp-idf-v4.4/components/esp_system/port/cpu_start.c:156

I (0) cpu_start: App cpu up.
I (297) cpu_start: Pro cpu start user code
I (297) cpu_start: cpu freq: 160000000
I (297) cpu_start: Application information:
I (301) cpu_start: Project name:     TestKafka
I (306) cpu_start: App version:      dddaeaa-dirty
I (312) cpu_start: Compile time:     Apr 27 2022 21:11:43
I (318) cpu_start: ELF file SHA256:  c24aefacc9c0dd41...
I (324) cpu_start: ESP-IDF:          v4.4-dirty
I (329) heap_init: Initializing. RAM available for dynamic allocation:
I (336) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (342) heap_init: At 3FFB3780 len 0002C880 (178 KiB): DRAM
I (349) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (355) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (361) heap_init: At 4008B64C len 000149B4 (82 KiB): IRAM
I (369) spi_flash: detected chip: generic
I (372) spi_flash: flash io: dio
W (376) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (391) cpu_start: Starting scheduler on PRO CPU.
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400865a1  PS      : 0x00060033  A0      : 0x80086e75  A1      : 0x3ffb6980  
0x400865a1: taskSelectHighestPriorityTaskSMP at /home/franz/esp-idf-v4.4/components/freertos/tasks.c:3424

A2      : 0x00000005  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3ffb2c7c  
A6      : 0x00000001  A7      : 0x3ffb2c84  A8      : 0x00000002  A9      : 0x3ffb5e94  
A10     : 0x3ffb2e0c  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00060f23  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000048  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x4008659e:0x3ffb69800x40086e72:0x3ffb69a0 0x400853b4:0x3ffb69c0 0x40085366:0xa5a5a5a5  |<-CORRUPTED
0x4008659e: taskSelectHighestPriorityTaskSMP at /home/franz/esp-idf-v4.4/components/freertos/tasks.c:3410 (discriminator 1)

0x40086e72: vTaskSwitchContext at /home/franz/esp-idf-v4.4/components/freertos/tasks.c:3520 (discriminator 3)

0x400853b4: _frxt_dispatch at /home/franz/esp-idf-v4.4/components/freertos/port/xtensa/portasm.S:436

0x40085366: _frxt_int_exit at /home/franz/esp-idf-v4.4/components/freertos/port/xtensa/portasm.S:231

ELF file SHA256: c24aefacc9c0dd41

Rebooting...
```

igrr, 2022-04-28:
This does look a bit like a stack overflow. Could you please try enabling [CONFIG_FREERTOS_WATCHPOINT_END_OF_STACK](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-freertos-watchpoint-end-of-stack) option and see if you get [stack overflow](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html#unhandled-debug-exception) reported at run time? If yes, please check your stack size.

diplfranzhoepfinger, 2022-04-28:
> [Good luck](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html#loadprohibited-storeprohibited)

thanks !

diplfranzhoepfinger, 2022-04-28:
> This does look a bit like a stack overflow. Could you please try enabling [...]

yes, doing that now !

diplfranzhoepfinger, 2022-04-28:
i set 

```
CONFIG_FREERTOS_WATCHPOINT_END_OF_STACK=y
```

deleted build folder, built and tested again. 

```
Executing action: monitor
Running idf_monitor in directory /home/franz/git/franz/test_kafka/TestKafka
Executing "/home/franz/.espressif/python_env/idf4.4_py3.8_env/bin/python /home/franz/esp-idf-v4.4/tools/idf_monitor.py -p /dev/ttyUSB0 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 0 --print_filter  /home/franz/git/franz/test_kafka/TestKafka/build/TestKafka.elf -m '/home/franz/.espressif/python_env/idf4.4_py3.8_env/bin/python' '/home/franz/esp-idf-v4.4/tools/idf.py'"...
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6612
load:0x40078000,len:14780
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (27) boot: ESP-IDF v4.4-dirty 2nd stage bootloader
I (27) boot: compile time 09:57:02
I (27) boot: chip revision: 3
I (30) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (38) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 2MB
I (51) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00100000
I (90) boot: End of partition table
I (94) boot_comm: chip revision: 3, min. application chip revision: 0
I (101) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=2949ch (169116) map
I (171) esp_image: segment 1: paddr=000394c4 vaddr=3ffb0000 size=028bch ( 10428) load
I (175) esp_image: segment 2: paddr=0003bd88 vaddr=40080000 size=04290h ( 17040) load
I (184) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=2b790h (178064) map
I (250) esp_image: segment 4: paddr=0006b7b8 vaddr=40084290 size=07488h ( 29832) load
I (262) esp_image: segment 5: paddr=00072c48 vaddr=50000000 size=00010h (    16) load
I (268) boot: Loaded app from partition at offset 0x10000
I (268) boot: Disabling RNG early entropy source...
I (282) cpu_start: Pro cpu up.
I (283) cpu_start: Starting app cpu, entry point is 0x4008118c
0x4008118c: call_start_cpu1 at /home/franz/esp-idf-v4.4/components/esp_system/port/cpu_start.c:156

I (0) cpu_start: App cpu up.
I (297) cpu_start: Pro cpu start user code
I (297) cpu_start: cpu freq: 160000000
I (297) cpu_start: Application information:
I (301) cpu_start: Project name:     TestKafka
I (306) cpu_start: App version:      922f2af
I (311) cpu_start: Compile time:     Apr 28 2022 09:57:39
I (317) cpu_start: ELF file SHA256:  e1954e9d49425bbb...
I (323) cpu_start: ESP-IDF:          v4.4-dirty
I (329) heap_init: Initializing. RAM available for dynamic allocation:
I (336) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (342) heap_init: At 3FFB3780 len 0002C880 (178 KiB): DRAM
I (348) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (354) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (361) heap_init: At 4008B718 len 000148E8 (82 KiB): IRAM
I (368) spi_flash: detected chip: generic
I (372) spi_flash: flash io: dio
W (376) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (390) cpu_start: Starting scheduler on PRO CPU.
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400865dd  PS      : 0x00060033  A0      : 0x80086eb0  A1      : 0x3ffb69e0  
0x400865dd: taskSelectHighestPriorityTaskSMP at /home/franz/esp-idf-v4.4/components/freertos/tasks.c:3424

A2      : 0x00000005  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3ffb2c7c  
A6      : 0x00000001  A7      : 0x3ffb2c84  A8      : 0x00000002  A9      : 0x3ffb5eec  
A10     : 0x3ffb2e0c  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00060f23  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000048  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x400865da:0x3ffb69e00x40086ead:0x3ffb6a00 0x400853f0:0x3ffb6a20 0x400853a2:0xa5a5a5a5  |<-CORRUPTED
0x400865da: taskSelectHighestPriorityTaskSMP at /home/franz/esp-idf-v4.4/components/freertos/tasks.c:3410 (discriminator 1)

0x40086ead: vTaskSwitchContext at /home/franz/esp-idf-v4.4/components/freertos/tasks.c:3520 (discriminator 3)

0x400853f0: _frxt_dispatch at /home/franz/esp-idf-v4.4/components/freertos/port/xtensa/portasm.S:436

0x400853a2: _frxt_int_exit at /home/franz/esp-idf-v4.4/components/freertos/port/xtensa/portasm.S:231

ELF file SHA256: e1954e9d49425bbb

Rebooting...
```

diplfranzhoepfinger, 2022-04-28:
putting the code in a extra Task does also not solve it: 

DiplomInformatikerFranzHoepfinger/test\_kafka@458ba8c

igrr, 2022-04-28:
Have tried to build your project, but looks like some files are missing from git. I got the following error:
```
Include directory
  '/tmp/test_kafka/TestKafka/components/librdkafka/include/include'
  is not a directory.
```
Should there be something in this directory?

igrr, 2022-04-28:
Okay, I think I see the reason. Indeed it is related to a stack overflow pretty early in startup process, and our debugging information is currently not sufficient to locate the issue.

librdkafka allocates a whole bunch of variables in the thread-local storage area of each thread (even in threads which don't use librdkafka). Here's the snippet from the map file:
```
                0x000000003f4285f8                _thread_local_start = ABSOLUTE (.)
 *(.tdata)
 *(.tdata.*)
 .tdata.rd_kafka_thread_name
                0x000000003f4285f8       0x40 esp-idf/librdkafka/liblibrdkafka.a(rdkafka.c.obj)
                0x000000003f4285f8                rd_kafka_thread_name
 *(.tbss)
 *(.tbss.*)
 .tbss.ret$14524
                0x000000003f428638       0x20 esp-idf/librdkafka/liblibrdkafka.a(rdkafka.c.obj)
 .tbss.reti$15608
                0x000000003f428658        0x4 esp-idf/librdkafka/liblibrdkafka.a(rdkafka_broker.c.obj)
 .tbss.ret$15607
                0x000000003f42865c      0x400 esp-idf/librdkafka/liblibrdkafka.a(rdkafka_broker.c.obj)
                0x000000003f428a5c                _thread_local_end = ABSOLUTE (.)
```
In source code it is `static RD_TLS char ret[4][RD_KAFKA_NODENAME_SIZE];` and other RD_TLS variables.

So that's 1124 bytes in the TLS area. Since the TLS area is allocated on the stack, this reduces available stack space in every task by 1124 bytes. This leads to a crash very early in the startup process.

We'll add some checks to catch such issues and provide better diagnostics.

diplfranzhoepfinger, 2022-04-28:
Thanks for this VERY GOOD Support. 
so which Stack should i increase ? 
in sdkconfig ? 

feel free to use my Repo for testing. 

Thanks, 
Franz

diplfranzhoepfinger, 2022-04-28:
> Have tried to build your project, but looks like some files are missing from git. I got the following error:
> 
> ```
> Include directory
>   '/tmp/test_kafka/TestKafka/components/librdkafka/include/include'
>   is not a directory.
> ```
> 
> Should there be something in this directory?

hi, 

see now, i made commits to all 3 branches fixing this Error.

igrr, 2022-04-28:
Well, i have increased sizes of stacks of all tasks which were in use and it worked. But stack sizes of other tasks (Wi-Fi, bluetooth, or any other task which you add to the application later) also needs to be increased.

1. delete sdkconfig file
2. Put the following into `sdkconfig.defaults` file:
   ```
    CONFIG_ESP_MAIN_TASK_STACK_SIZE=16000
    CONFIG_ESP_IPC_TASK_STACK_SIZE=4096
    CONFIG_ESP_TIMER_TASK_STACK_SIZE=4584
    CONFIG_FREERTOS_IDLE_TASK_STACKSIZE=2536
    CONFIG_FREERTOS_TIMER_TASK_STACK_DEPTH=3048
    CONFIG_FREERTOS_WATCHPOINT_END_OF_STACK=y
   ```
3. build and flash the project again
4. (optional) recommend keeping sdkconfig in `.gitignore` and check in only `sdkconfig.defaults`

diplfranzhoepfinger, 2022-04-28:
Hello, 

tested your suggestion, 
committed it to https://github.com/DiplomInformatikerFranzHoepfinger/test_kafka/commit/7f2d78af8e4c825e4f34c3e782ee0c4c86b4720c

but i still get: 
```
Executing action: monitor
Running idf_monitor in directory /home/franz/git/franz/test_kafka/TestKafka
Executing "/home/franz/.espressif/python_env/idf4.4_py3.8_env/bin/python /home/franz/esp-idf-v4.4/tools/idf_monitor.py -p /dev/ttyUSB0 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 0 --print_filter  /home/franz/git/franz/test_kafka/TestKafka/build/TestKafka.elf -m '/home/franz/.espressif/python_env/idf4.4_py3.8_env/bin/python' '/home/franz/esp-idf-v4.4/tools/idf.py'"...
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
372) spi_flash: flash io: dio
ts Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6612
load:0x40078000,len:14780
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (27) boot: ESP-IDF v4.4-dirty 2nd stage bootloader
I (27) boot: compile time 14:40:50
I (27) boot: chip revision: 3
I (30) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (38) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 2MB
I (51) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00100000
I (90) boot: End of partition table
I (94) boot_comm: chip revision: 3, min. application chip revision: 0
I (101) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=294a4h (169124) map
I (171) esp_image: segment 1: paddr=000394cc vaddr=3ffb0000 size=028bch ( 10428) load
I (175) esp_image: segment 2: paddr=0003bd90 vaddr=40080000 size=04288h ( 17032) load
I (184) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=2b7c0h (178112) map
I (250) esp_image: segment 4: paddr=0006b7e8 vaddr=40084288 size=07498h ( 29848) load
I (262) esp_image: segment 5: paddr=00072c88 vaddr=50000000 size=00010h (    16) load
I (268) boot: Loaded app from partition at offset 0x10000
I (268) boot: Disabling RNG early entropy source...
I (282) cpu_start: Pro cpu up.
I (283) cpu_start: Starting app cpu, entry point is 0x40081194
0x40081194: call_start_cpu1 at /home/franz/esp-idf-v4.4/components/esp_system/port/cpu_start.c:156

I (0) cpu_start: App cpu up.
I (297) cpu_start: Pro cpu start user code
I (297) cpu_start: cpu freq: 160000000
I (297) cpu_start: Application information:
I (301) cpu_start: Project name:     TestKafka
I (306) cpu_start: App version:      7f2d78a
I (311) cpu_start: Compile time:     Apr 28 2022 14:41:24
I (317) cpu_start: ELF file SHA256:  86692a84f69826ae...
I (323) cpu_start: ESP-IDF:          v4.4-dirty
I (329) heap_init: Initializing. RAM available for dynamic allocation:
I (336) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (342) heap_init: At 3FFB3780 len 0002C880 (178 KiB): DRAM
I (348) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (354) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (361) heap_init: At 4008B720 len 000148E0 (82 KiB): IRAM
I (368) spi_flash: detected chip: generic
I (372) spi_flash: flash io: dio
W (376) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (390) cpu_start: Starting scheduler on PRO CPU.
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400865e5  PS      : 0x00060033  A0      : 0x80086eb8  A1      : 0x3ffb6dc0  
0x400865e5: taskSelectHighestPriorityTaskSMP at /home/franz/esp-idf-v4.4/components/freertos/tasks.c:3424

A2      : 0x00000005  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3ffb2c7c  
A6      : 0x00000001  A7      : 0x3ffb2c84  A8      : 0x00000002  A9      : 0x3ffb5eec  
A10     : 0x3ffb2e0c  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00060f23  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000048  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x400865e2:0x3ffb6dc00x40086eb5:0x3ffb6de0 0x400853f8:0x3ffb6e00 0x400853aa:0xa5a5a5a5  |<-CORRUPTED
0x400865e2: taskSelectHighestPriorityTaskSMP at /home/franz/esp-idf-v4.4/components/freertos/tasks.c:3410 (discriminator 1)

0x40086eb5: vTaskSwitchContext at /home/franz/esp-idf-v4.4/components/freertos/tasks.c:3520 (discriminator 3)

0x400853f8: _frxt_dispatch at /home/franz/esp-idf-v4.4/components/freertos/port/xtensa/portasm.S:436

0x400853aa: _frxt_int_exit at /home/franz/esp-idf-v4.4/components/freertos/port/xtensa/portasm.S:231

ELF file SHA256: 86692a84f69826ae

Rebooting...
```

diplfranzhoepfinger, 2022-04-28:
> I think increasing stack sizes like this is not a very good solution, it's just to show that the library can work, in principle. Better solution would be to add some platform-specific way for this library to use pthread local storage instead of RD_TLS attributes.

i agree. 

can you give some hint how i could modify it in a way that i use pthread local storage ?

diplfranzhoepfinger, 2022-04-28:
i need to learn more about "pthread local storage" vs. RD_TLS.

RD_TLS is defined: 
`#define RD_TLS __thread`

right ? 
and pthread local storage is this ? 
https://github.com/espressif/esp-idf/blob/master/components/pthread/pthread_local_storage.c

diplfranzhoepfinger, 2022-05-02:
@igrr i did do your suggestions,

https://github.com/espressif/esp-idf/issues/8848#issuecomment-1112040006

 but still similar error ... 

Thanks !

closed 2022-09-01 by diplfranzhoepfinger