espressif/esp-idf #8849: Incorrect semaphore behavior in FreeRTOS under esp-idf

matt-mjm, 2022-04-27:
## Environment

- Development Kit: WEMOS Lolin D32 (non-espressif tool)
- Kit version (for WroverKit/PicoKit/DevKitC): N/A
- Module or chip used:  ESP32-WROOM32
- IDF version (run ``git describe --tags`` to find it): v4.3 v4.4.1 v5.0-dev-2443-g047903c612
- Build System: idf.py / cmake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
when building on v5.0-dev-2443-g047903c612: xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2021r2-patch3) 8.4.0
when building on  v4.4.1: xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2021r2-patch3) 8.4.0
when building on v4.3: xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Windows 10 x86_64
- (Windows only) environment type: esp-idf PowerShell
- Using an IDE?: No
- Power Supply: USB

## Problem Description
We have identified a problem with the FreeRTOS Semaphore API under esp-idf. When creating two tasks which each attempt to give and take a mutex repeatedly inside a loop, one task (task A) can become blocked permanently during a take operation. Meanwhile, the other task (task B) is able to give and take the mutex inside itself repeatedly for over five seconds, completely starving out the first task of access.

### Procedure
 
The table below outlines the basic process of each of the two tasks in our sample code

| Task A | Task B |
| --- | --- |
| Take the mutex | Take the mutex |
| busywait (esp_rom_delay_us) | vTaskDelay |
| Give the semaphore | busywait (esp_rom_delay_us) |
| vTaskDelay | Give the semaphore |

### Notes
- The example code provided below should reproduce this problem on esp-idf v4.3, v4.4.1, and v5.0-dev-2443-g047903c612.
- Setting the menuconfig option **FREERTOS_UNICORE** causes the issue to disappear with our test code under all three esp-idf versions tested.
- Note that v4.3 is also affected by this problem in the case where the priority of task A is set above the priority of task B

### Affected Versions
All three tested esp-idf versions were affected, but not under the same conditions. The matrix below indicates the conditions under which the two different log outputs were generated.
- 1 means generated log output # 1 was generated (see logs below)
- 2 means generated log output # 2 was generated (see logs below)

| v5.0 | v4.4.1 | v4.3 | | |
| --- | --- | --- | --- | --- |
| 2 | 2 | 1 | Unpinned | Same Priority |
| 1 | 1 | 1 | | A higher priority than B |
| 2 | 2 | 2 | | B higher priority than A |
| 2 | 2 | 1 | Pinned to same core | Same Priority |
| 1 | 1 | 1 | | A higher priority than B |
| 2 | 2 | 2 | | B higher priority than A |
| 2 | 2 | 2 | Pinned to different core | Same Priority |
| 2 | 2 | 2 | | A higher priority than B |
| 2 | 2 | 2 | | B higher priority than A |
| 1 | 1 | 1 | FREERTOS_UNICORE | Same Priority |
| 1 | 1 | 1 | | A higher priority than B |
| 2 | 2 | 2 | | B higher priority than A |

### Expected Behavior
The intuitive expectation is that a "first-come-first-serve" system should exist for tasks of the same priority, with threads waiting the longest for the mutex being serviced first as the mutex becomes available.

Indeed, in FreeRTOS, semaphores are implemented as queues, which appear to offer this guarantee:
*Queues can have multiple readers, so it is possible for a single queue to have more than one task blocked on it waiting for data. [...] If the blocked tasks have equal priority, then the task that has been waiting for data the longest will be unblocked.*
Taken from: https://www.freertos.org/fr-content-src/uploads/2018/07/161204_Mastering_the_FreeRTOS_Real_Time_Kernel-A_Hands-On_Tutorial_Guide.pdf

It appears that this guarantee is being violated here: although A has been waiting to receive from the queue for many seconds and hundreds of queue sends and receives, xQueueSemaphoreTake in B continues to be serviced first before the same call in A.

```cpp
#define xSemaphoreGive( xSemaphore )    xQueueGenericSend( ...
#define xSemaphoreTake( xSemaphore, xBlockTime )    xQueueSemaphoreTake( ...
```

In our example, we would expect **Task A** to be able to intermittently acquire the mutex during the test period.

### Actual Behavior
After an initial period, **Task A** blocks forever while **Task B** takes and gives the semaphore to itself, starving task A entirely until the loop in **Task B** exits.

### Possible Explanations
During our testing with v5.0-dev-2443-g047903c612, we determined that xQueueGenericSend in Task B was indeed identifying that another task of equal priority was waiting to acquire the mutex (or, receive from the queue), and correctly yielding control to the scheduler. The scheduler, however, chose to once again schedule task B to run again, as tasks A and B are both of the same priority. Task B then called xQueueSemaphoreTake, which quickly acquired the mutex / dequeued with no need to block. Task A does run at the next opportunity, when Task B calls vTaskDelay, but it is too late; the mutex has already been taken, and it must block again and wait for another opportunity.

As v4.3 is also affected when the priority of task A is set above the priority of task B, some other unidentified mechanism may be involved in that case.

### Steps to reproduce
1. Create a fresh project with ``idf.py create-project``.
2. Paste the example code into a ``.c`` file.
4. Flash the image to an ESP32-WROOM32
5. Monitor the serial console
6. Compare the output to the attached logs / Observe that following the initial period, B repeatedly prints "Taken" while A doesn't get the semaphore until the loop in B exits
7. In the case that the problem cannot be reproduced, the attached binary can also be tested

### Code to reproduce this issue

```cpp
#include "esp_log.h"

#include "freertos/FreeRTOS.h"
#include "freertos/FreeRTOSConfig.h"
#include "freertos/task.h"
#include "freertos/semphr.h"

SemaphoreHandle_t sem;

// print is set to true while Task A is waiting for the mutex.
// Task B will print its "Taken" message only while Task A is
// waiting for the mutex, to avoid cluttering the console output
bool print = false;

static void task_a(void *arg) {
    static const char* TAG = "A";
    for (int i = 0; i < 3; i++) {
        ESP_LOGI(TAG, "Take");
        print = true;
        xSemaphoreTake(sem, portMAX_DELAY);
        print = false;
        ESP_LOGI(TAG, "Taken");
        esp_rom_delay_us(5000);
        ESP_LOGI(TAG, "Give");
        xSemaphoreGive(sem);
        vTaskDelay(pdMS_TO_TICKS(1000));
    }
    vTaskDelete(NULL);
}

static void task_b(void *arg) {
    static const char* TAG = "B";
    for (int i = 0; i < 500; i++) {
        xSemaphoreTake(sem, portMAX_DELAY);
        if (print) ESP_LOGI(TAG, "Taken");
        vTaskDelay(pdMS_TO_TICKS(10));
        esp_rom_delay_us(3546);
        xSemaphoreGive(sem);
    }
    vTaskDelete(NULL);
}

void app_main() {
    sem = xSemaphoreCreateMutex();

// For v4.4.1 and v5.0, PA and PB we have reproduced the issue with PA == PB == 1
// For v4.3, we have reproduced the issue with PA == PB == 1 and PA == 2, PB == 1
#define PA 1
#define PB 1

    // Unpinned (for versions 4.4 and 5.0)
    xTaskCreate(task_b, "taskb", 2048, NULL, tskIDLE_PRIORITY + PB, NULL);
    vTaskDelay(pdMS_TO_TICKS(100));
    xTaskCreate(task_a, "taska", 2048, NULL, tskIDLE_PRIORITY + PA, NULL);

    // Pinned to same core
    // xTaskCreatePinnedToCore(task_b, "taskb", 2048, NULL, tskIDLE_PRIORITY + PB, NULL, 1);
    // vTaskDelay(pdMS_TO_TICKS(100));
    // xTaskCreatePinnedToCore(task_a, "taska", 2048, NULL, tskIDLE_PRIORITY + PA, NULL, 1);

    // Pinned to different core (for version 4.3)
    // xTaskCreatePinnedToCore(task_b, "taskb", 2048, NULL, tskIDLE_PRIORITY + PB, NULL, 0);
    // vTaskDelay(pdMS_TO_TICKS(100));
    // xTaskCreatePinnedToCore(task_a, "taska", 2048, NULL, tskIDLE_PRIORITY + PA, NULL, 1);

    for (;;) {
        vTaskDelay(pdMS_TO_TICKS(1000));
    }
}
```

## Debug Logs
Note that A successfully takes the semaphore prior to B starting
Once B starts A is not able to take the semaphore again until B's loop has ended and its task has exited.
Then A is able to take the semaphore and complete its loop

#### Output 1 (Expected):
```
I (512) A: Take
I (522) A: Taken
I (522) A: Give
I (1522) A: Take
I (1522) A: Taken
I (1522) A: Give
I (2522) A: Take
I (2522) A: Taken
I (2522) A: Give
```
#### Output 2 (Actual):
```
I (510) A: Take
I (510) A: Taken
I (510) A: Give
I (1510) A: Take
I (1510) B: Taken
I (1520) B: Taken
... (Continue to loop "B: Taken")
I (5380) B: Taken
I (5390) B: Taken
I (5400) B: Taken
I (5410) A: Taken
I (5410) A: Give
I (6410) A: Take
I (6410) A: Taken
I (6410) A: Give
```

## Other items
[good-log.txt](https://github.com/espressif/esp-idf/files/8575744/good-log.txt)
[bad-log.txt](https://github.com/espressif/esp-idf/files/8575745/bad-log.txt)
[source.zip](https://github.com/espressif/esp-idf/files/8575750/source.zip)
[binaries.zip](https://github.com/espressif/esp-idf/files/8575752/binaries.zip)

nopnop2002, 2022-06-13:
There is no wait between xSemaphoreGive and xSemaphoreTake in task_b

```
static void task_b(void *arg) {
    static const char* TAG = "B";
    for (int i = 0; i < 500; i++) {
        xSemaphoreTake(sem, portMAX_DELAY);
        if (print) ESP_LOGI(TAG, "Taken");
        vTaskDelay(pdMS_TO_TICKS(10));
        esp_rom_delay_us(3546);
        xSemaphoreGive(sem);
    }
    vTaskDelete(NULL);
}
```

This is work.

```
static void task_b(void *arg) {
    static const char* TAG = "B";
    for (int i = 0; i < 500; i++) {
        xSemaphoreTake(sem, portMAX_DELAY);
        if (print) ESP_LOGI(TAG, "Taken");
        esp_rom_delay_us(3546);
        xSemaphoreGive(sem);
        vTaskDelay(pdMS_TO_TICKS(10));
    }
    vTaskDelete(NULL);
}
```

nopnop2002, 2022-06-14:
@acf-bwl 

Is it possible that the optimize of the compiler has an effect?

AxelLin, 2022-11-20:
@Dazza0 Any comment abut this issue? The starvation looks like a fundamental issue.

AxelLin, 2022-11-28:
@Dazza0 

It does not make sense there is no response for 7 months.
If you are not the correct assignee please speak up and forward this to proper developers.
As I said the starvation is a fundamental issue, I hope to get clarify if this is a bug or not.

@Alvin1Zhang @igrr

acf-bwl, 2022-11-28:
When we were originally investigating this issue, we also determined it was a fundamental issue, and that it might be very difficult to fix on the current Espressif multi-core FreeRTOS implementation. This was a pattern we used in parts of our code, and we have had to work around this issue. The Espressif developers have to decide what issues make sense to prioritize based on importance and ability to fix. Being a small customer and not paying for support means waiting for espressif to get around to it or fixing it ourselves. We do believe this is an important issue, though.

acf-bwl, 2022-12-05:
@Dazza0 Thank you for the detailed analysis. I can see you've put a lot of thought into this issue; it's much appreciated.

Note that from our testing, this can also happen with two tasks of the same priority as well, which was the biggest issue for us. We have a pattern we had been using in our code:

```
for (;;) {
    take mutex
    //  most of the time is spent in here waiting for I/O
    give mutex

    // allow other tasks of same or greater priority that are waiting, to access shared resource before taking mutex once again
}
```

> Reserving the next reader

I think some variation of this solution is the most workable. I do feel that some fix to this issue to bring it in line with POSIX semaphores / most other platforms semantics is inevitable eventually.

> Should "task Y" block and leave the message for "task A", waiting for "task A" to schedule back to read the message (i.e., strict FIFO order)?
> Or should "task Y" read the message immediately given that it would be more efficient as "task A" cannot immediately read the message (i.e., first come first serve model)?

Implementing the desired semantics strictly might be easiest until the problem is not reproducible anymore, and then loosening them up, but that's just my two cents ;)

AxelLin, 2022-12-06:
> @acf-bwl @AxelLin I've taken a look at this issue and have been able to recreate this locally on multiple releases. The issue you're seeing is one scenario of the more general "spinlock acquire race condition" previously described in [this comment](https://github.com/espressif/esp-idf/pull/10199#pullrequestreview-1194340219).
> 
> This issue is quite difficult to explain, so I'll provide a quick, summary here and leave the full explanations below.
> 
> * This bug essentially involves a lower priority task A on core X stealing a  (queue message/mutex/semaphore) from a higher priority blocked task B on core Y. It is a result of some single-core FreeRTOS assumptions no longer being true under SMP.

How about the starvation issue happened with "the sam priority" case?

Dazza0, 2022-12-06:
@AxelLin 

> How about the starvation issue happened with "the sam priority" case?

The race condition is the same, In this case the task B on the other core of the same priority is able to enter the critical section first, thus leading to "task B" stealing the message.

AxelLin, 2022-12-23:
> ## Implications on SMP in IDF
> This race condition isn't an issue in most IDF applications for the following reasons:

But @acf-bwl and @matt-mjm already prove this issue can happen in a real project, especially with "the same priority" case.

wuyuanyi135, 2023-01-21:
@Dazza0  @matt-mjm  @acf-bwl  Thanks for raising up this issue. I also notice the similar inconsistent behavior when using the C++20 semaphore to implement a producer-consumer-like library. See the code below:
# Problematic case on ESP32S3 (`v5.1-dev-3027-gdace03fccd`)
The following code will block very quickly at either producer or consumer's `acquire`. It worths noting that if I uncomment the delay line it could run fine. That's why I suspect this is a similar issue related to this one.
```cpp
#include <esp_log.h>
#include <mutex>
#include <semaphore>
#include <thread>
static const char* TAG = "MAIN";
std::counting_semaphore<10> filled_sem{0};
std::counting_semaphore<10> empty_sem{2};
extern "C" void app_main() {
  int count = 0;

  while (true) {
    const size_t repeat = 100;
    ESP_LOGI("MAIN", "Count = %d", count);
    std::thread producer_thread([&] {
      for (int i = 0; i < repeat; ++i) {
        empty_sem.acquire();
        ESP_LOGI("MAIN", "Produced an element");
        filled_sem.release();
      }
    });
    std::thread consumer_thread([&] {
      for (int i = 0; i < repeat; ++i) {
        filled_sem.acquire();
        ESP_LOGI("MAIN", "Consumed an element");
        empty_sem.release();
        //        vTaskDelay(pdMS_TO_TICKS(10)); // With this like it runs OK
      }
    });
    producer_thread.join();
    consumer_thread.join();
  }
}
```

Also see the stack backtrace:
```
#0  0x40375c69 in esp_ptr_executable (p=0xfffffffd)
    at C:/Espressif/frameworks/esp-idf-master/components/esp_hw_support/include/esp_memory_utils.h:207
#1  esp_backtrace_get_next_frame (frame=<optimized out>)
    at C:/Espressif/frameworks/esp-idf-master/components/esp_system/port/arch/xtensa/debug_helpers.c:33
#2  0x4037eaa4 in insert_free_block (sl=-2143819100, fl=1070552304, block=0x3fcf54b0, control=0x3fcf54f0)
    at C:/Espressif/frameworks/esp-idf-master/components/heap/tlsf/tlsf.c:371
#3  block_insert (block=0x3fcf54b0, control=0x3fcf54f0)
    at C:/Espressif/frameworks/esp-idf-master/components/heap/tlsf/tlsf.c:388
#4  block_trim_free (size=<optimized out>, block=0x3fc9d375, control=0x3fcf54f0)
    at C:/Espressif/frameworks/esp-idf-master/components/heap/tlsf/tlsf.c:502
#5  block_prepare_used (size=<optimized out>, block=0x3fc9d375, control=0x3fcf54f0)
    at C:/Espressif/frameworks/esp-idf-master/components/heap/tlsf/tlsf.c:579
#6  tlsf_malloc (tlsf=0x3fcf54f0, size=<optimized out>)
    at C:/Espressif/frameworks/esp-idf-master/components/heap/tlsf/tlsf.c:1005
#7  0x403856c8 in ?? ()
#8  0x403800fc in spi_flash_chip_mxic_opi_detect_size (chip=<optimized out>, size=0x0)
    at C:/Espressif/frameworks/esp-idf-master/components/spi_flash/spi_flash_chip_mxic_opi.c:44
#9  0x4037f24f in __atomic_fetch_add_8 (ptr=0x3fcecf20, value=4597840332355272704, memorder=0)
    at C:/Espressif/frameworks/esp-idf-master/components/newlib/stdatomic.c:447
#10 0x40375606 in call_start_cpu0 ()
    at C:/Espressif/frameworks/esp-idf-master/components/esp_system/port/cpu_start.c:513
```

# Working case on MINGW64
This is the similar code above but run on Windows MINGW64 platform. It runs flawlessly.
```cpp
#include <cstdio>
#include <mutex>
#include <semaphore>
#include <thread>
std::counting_semaphore<10> filled_sem{0};
std::counting_semaphore<10> empty_sem{2};
int main() {
  int count = 0;
  while (true) {
    const size_t repeat = 100;
    printf("Count = %d\n", count);
    std::thread producer_thread([&] {
      for (int i = 0; i < repeat; ++i) {
        empty_sem.acquire();
        printf("Produced an element\n");
        filled_sem.release();
      }
    });
    std::thread consumer_thread([&]{
      for (int i = 0; i < repeat; ++i) {
        filled_sem.acquire();
        printf("Consumed an element\n");
        empty_sem.release();
      }
    });
    producer_thread.join();
    consumer_thread.join();
  }
}
```

> The good news is that this race condition shouldn't be an issue in IDF applications, thus why the same queue/mutex/semaphore implementation has been used in IDF for many releases without this race condition causing a noticeable error (see explanation below).

@Dazza0 Sorry I may have missed some key information in your detailed messages. But I think this could be a very common issue since it can easily trigger deadlock in the multithread or buffered IO codes. Could you elaborate why is this race condition shouldn't be an issue in IDF applications and what would be the recommended approach to work around this issue? I found adding 1 tick delay the only useful workaround. Could you confirm this delay (or yield of time frame) is necessary when using semaphore?

nopnop2002, 2023-01-26:
>this issue can happen in a real project, especially with "the same priority" case.

That's exactly right.

Zmmfly, 2023-03-10:
This problem also exists in IDF version 5.0.1

AxelLin, 2023-07-14:
> @AxelLin
> 
> > How about the starvation issue happened with "the sam priority" case?
> 
> The race condition is the same, In this case the task B on the other core of the same priority is able to enter the critical section first, thus leading to "task B" stealing the message.

@Dazza0 @zikalino 

I just check FreeRTOS V10.6.0 changelog ( https://github.com/FreeRTOS/FreeRTOS-Kernel/releases/tag/V10.6.0 )

I'm wondernig if below fixes mentioned in changelog are realted to this issue?

+ Update eTaskGetState and uxTaskGetSystemState to return eReady for pending ready
  tasks. Contributed by @Dazza0.
+ Fix the scheduler logic in a couple of places to not preempt a task when an
  equal priority task becomes ready.

AxelLin, 2023-08-19:
> @acf-bwl @AxelLin I've taken a look at this issue and have been able to recreate this locally on multiple releases. The issue you're seeing is one scenario of the more general "spinlock acquire race [...]

@Dazza0 
Does upstream FreeRTOS-SMP has this bug? If yes, could you provide link for the issue on upstream?

AxelLin, 2023-09-22:
@Dazza0 

You said there is no easy way to fix this.
But you are aware the bug is real and this has been reported by many people.
So what make you think it's ok to live with this bug?

SravanMaddukuri, 2024-07-02:
Hi, I am using a recursive semaphore in one of my tasks to call the 'mqtt_enqueue' function. According to the logs, the MQTT_API_LOCK semaphore is acquired by the mqtt_task (from the framework - mqtt_client.c) and is never released until my task exits.

Is anyone else using a recursive semaphore experiencing a similar issue and has found a workaround?