espressif/esp-idf #8852: ESP HTTP Client slow TLS handshake and async not working

aeropagz, 2022-04-28:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version [v4.4]
- Build System:  [idf.py]
- Compiler version: [8.4.0]
- Operating System: [Windows]
- (Windows only) environment type: [ESP Command Prompt]
- Using an IDE?: [Yes, VS Code with Esp IDF Plugin]
- Power Supply: [USB]

## Problem Description

I use the esp-idf https client to talk with my API. My major problem is that each request is taking around 2-3 seconds. It is a simple PUT request with no body payload. Just an Url.
I am using client cert authentication and the server is running an self-signed certificate. 

I found out that the TLS handshake takes most of the time (up to 2.5 secs).

### Expected Behavior
A fast request. With postman it takes between 100-200ms. Some similar timings would be nice.

### Actual Behavior

Each request takes about 2 - 3 secs.

### Steps to reproduce

1. init an esp_http_client_config_t with put method, server and client cert
2. perform request 

### Code to reproduce this issue
Sync request:
```cpp
    char url[256] = {0};
    snprintf(url, sizeof(url), "https://***/board/e752bc25-9693-49db-9a59-c0bdabf46c21/goal/%s", side);
    ESP_LOGI(TAG, "buffer: %s", url);

    esp_http_client_config_t config = {
        .url = url,
        .method = HTTP_METHOD_PUT,
        .event_handler = _http_event_handler,
        .transport_type = HTTP_TRANSPORT_OVER_SSL,
        .cert_pem = cert_pem,
        .client_cert_pem = client_cert_pem,
        .client_key_pem = client_key_pem,
    };

    esp_http_client_handle_t client = esp_http_client_init(&config);
    ESP_LOGI(TAG, "PREPERFORM");

    esp_err_t err = esp_http_client_perform(client);

    if (err == ESP_OK)
    {
        ESP_LOGI(TAG, "HTTPS Status = %d, content_length = %d",
                 esp_http_client_get_status_code(client),
                 esp_http_client_get_content_length(client));
    }
    else
    {
        ESP_LOGE(TAG, "Error perform http request %s", esp_err_to_name(err));
    }
    esp_http_client_cleanup(client);
}
```

## Debug Logs

```
I (44151) GOAL SERVICE: buffer: https://example.de/board/e752bc25-9693-49db-9a59-c0bdabf46c21/goal/HOME
I (44151) GOAL SERVICE: PREPERFORM
D (44151) HTTP_CLIENT: Begin connect to: https://example.de
D (44161) esp-tls: host:aeropagz.de: strlen 11
D (44171) esp-tls: [sock=54] Resolved IPv4 address: 1.1.1.1
D (44181) esp-tls: [sock=54] Connecting to server. HOST: example.de, Port: 8080
D (44241) esp-tls: handshake in progress...
D (46771) HTTP_HANDLER: HTTP_EVENT_ON_CONNECTED
D (46771) HTTP_CLIENT: Write header[3]: PUT /board/e752bc25-9693-49db-9a59-c0bdabf46c21/goal/HOME HTTP/1.1
User-Agent: ESP32 HTTP Client/1.0
Host: example.de
Content-Length: 0

D (46781) HTTP_HANDLER: HTTP_EVENT_HEADER_SENT
D (47351) HTTP_CLIENT: on_message_begin
D (47351) HTTP_CLIENT: HEADER=Vary:Origin
....
I (47631) GOAL SERVICE: HTTPS Status = 200, content_length = -1
I (47641) gpio: Goal: Home
```

laukik-hase, 2022-05-10:
Hello, @aeropagz!

Thank you for sending the issue report. The TLS Session Resumption feature might help you.

You can refer the `https_request` example for implementation details - see [here](https://github.com/espressif/esp-idf/blob/a82e6e63d98bb051d4c59cb3d440c537ab9f74b0/examples/protocols/https_request/main/https_request_example_main.c#L214).

laukik-hase, 2022-06-06:
Closing the issue, @aeropagz - feel free to reopen it if you have any further questions.

closed 2022-06-06 by laukik-hase

ChrisEAlfred, 2024-10-04:
When using TLS session keys as suggested. It appears random whether connection with a reused session takes ~3s to connect, or ~200ms to connect. Once slow it stays slow; once fast it stays fast.
Why would this happen?

Emill, 2024-12-02:
Honestly I find it really strange that the esp http client with mbedtls takes several seconds for the tls handshake. The RSA hardware peripheral just takes a couple of milliseconds to do an encryption/verify operation. Even with a couple of certificates in the chain to verify it should not take more than a couple of tens of milliseconds to do this.

ChrisEAlfred, 2024-12-02:
Is it ok to use these:
CONFIG_MBEDTLS_DYNAMIC_BUFFER=y
CONFIG_MBEDTLS_DYNAMIC_FREE_PEER_CERT=y
CONFIG_MBEDTLS_DYNAMIC_FREE_CONFIG_DATA=y
CONFIG_MBEDTLS_DYNAMIC_FREE_CA_CERT=y

ChrisEAlfred, 2024-12-03:
As a work around for getting stuck with long TLS handshake, we check if the handshake takes more than 2s and discard the saved ticket and try again until we get a fast handshake.

Admittedly, I am starting to suspect there may be a load balancer issue.