espressif/esp-idf #8856: Micropython Tensorflow Guru Meditation Error: Core 1 panic'ed (InstrFetchProhibited)

MATTYGILO, 2022-04-28:
I am using a ESP32S3WROOM2

Ram size: 8mb
Flash size: 32mb

I am using Micropython and this module [https://github.com/mocleiri/tensorflow-micropython-examples](url).

The project executes correctly the first 8 times.

On the 9th I get this error

```
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x00000000  PS      : 0x00060c30  A0      : 0x8200cbf2  A1      : 0x3fce4500  
A2      : 0x3fcf47ec  A3      : 0x3fce3c44  A4      : 0x0000000b  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x3db9a7f0  A8      : 0x8200c119  A9      : 0x3fce44e0  
A10     : 0x3fcf478c  A11     : 0x3db9a7f0  A12     : 0x00000000  A13     : 0x42303448  
A14     : 0xffffff80  A15     : 0x3db3f428  SAR     : 0x0000001d  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  

Backtrace:0xfffffffd:0x3fce45000x4200cbef:0x3fce4530 0x42001b2e:0x3fce4550 0x4200195a:0x3fce4570 0x420655c9:0x3fce4590 0x4206d87d:0x3fce45b0 0x4037886d:0x3fce45d0 0x420656e0:0x3fce4670 0x4206d87d:0x3fce46e0 0x4037886d:0x3fce4700 0x420656e0:0x3fce47a0 0x4206d7bd:0x3fce47d0 0x403787e1:0x3fce47f0 0x420656e0:0x3fce4890 0x4206d7bd:0x3fce4900 0x403787e1:0x3fce4920 0x420656e0:0x3fce49c0 0x4206d7bd:0x3fce4a30 0x403787e1:0x3fce4a50 0x420656e0:0x3fce4af0 0x4206d7bd:0x3fce4b40 0x403787e1:0x3fce4b60 0x420656e0:0x3fce4c00 0x4206d7e5:0x3fce4c50 0x420ae45c:0x3fce4c70 0x420ae6ac:0x3fce4d00 0x42089564:0x3fce4d40 

ELF file SHA256: 74adcc789bc7d279
```

nikidimitrow, 2022-04-29:
[InstrFetchProhibited](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html#instrfetchprohibited) - It look like a stack overflow. Check this answer here: https://github.com/espressif/esp-idf/issues/8848

diplfranzhoepfinger, 2022-05-02:
> InstrFetchProhibited

no. your case is different: 

in my case PC was non-Zero. 

in your case PC is Zero. 

see: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html#instrfetchprohibited

nikidimitrow, 2022-05-02:
As I mention in my comment, the case scenario is different, that why the link I provide for the ticket IDFGH-7267 is InstrFetchProhibited, not like LoadProhibited in ticket IDFGH-7259.

But again look like stackoverflow problem.

closed 2022-05-02 by igrr