espressif/esp-idf #8863: External non-Espressif PSRAM support

samvangysegem, 2022-04-29:
## Environment

- Development Kit: none
- Module or chip used:  ESP32-S3
- IDF version: v4.4-144-gafe82b4e1b
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System: macOS
- Power Supply: external 3V3

## Problem Description

The menuconfig and Espressif documentation states that the ESP32-S3 supports a number of external PSRAM memories in addition to the ESP-PSRAM modules, one example of which is the IS25WP032 (Component config > ESP32S3-Specific > Support for external, SPI-connected RAM > SPI RAM config > Type of SPIRAM chip in use). In my current project, I have the latter connected through SPI1 to the Espressif module (ESP32-S3) but it has trouble finding it. Upon looking closer to the SPIRAM code in the ESP-IDF (components/esp_hw_support/port/esp32s3/spiram_psram.c) I noticed the code is specifically written to support the ESP-PSRAM modules and no other modules due to macros such as `PSRAM_IS_VALID()`. These checks are very specific to the Espressif PSRAM modules and will never pass for the "supported" PSRAM modules of other manufacturers. 

My question specifically is when the support for other PSRAM modules, especially the ones mentioned in the menuconfig and Espressif docs, will be added to the ESP-IDF, if ever?

## Debug Logs

```
E (168) psram: PSRAM ID read error: 0x0016709d
E (168) cpu_start: Failed to init external RAM!
```

huming2207, 2022-04-30:
We are also thinking to have a try on IS25WP128 for caching LCD waveforms and/or UI rendering. It seems like the PSRAM support for ESP32S3 is still somewhat a bit immature.

ginkgm, 2022-05-05:
Hi @samvangysegem ,

Sorry I couldn't quite catch your point. Which one of following do you want:

1. Contact a non-official PSRAM to the original place where Espressif modules put its PSRAM.
2. Add one extra PSRAM to the other place

For requirement 1, there's no many other alternatives when we design the product. If you have alternatives for the Espressif PSRAM, please provide us the information of these PSRAMs, and we will consider adding them to our product line. Or at least find some way for you to extend the support for it.

For requirement 2, to read/write PSRAM with cache, the only place we can contact PSRAM is the SPI0/1 CS1. This is a hardware limitaiton. If you don't want to read/write with cache, you may do it by writing a new driver, on SPI1/SPI2/SPI3 buses. But currently we don't provide such driver, even for official PSRAMs.

Michael

samvangysegem, 2022-05-05:
Hi @ginkgm,

I think our problem fits more in the first category. We are trying to connect the ESP32-S3 SoC to external PSRAM, more specifically the IS25WP032D-JKLE, through SPI0/1 in order to transparently integrate this external memory in the memory map of the ESP32-S3. Our choice for this specific external PSRAM was based on the information available in the menuconfig from the esp-idf, a screenshot of which is shown below.

[image]

This is also confirmed by the following code snippet of the Kconfig file of the ESP32-S3:
```
# Hint: to support SPIRAM across multiple chips, check CONFIG_SPIRAM instead
    config ESP32S3_SPIRAM_SUPPORT
        bool "Support for external, SPI-connected RAM"
        default "n"
        select SPIRAM
        help
            This enables support for an external SPI RAM chip, connected in parallel with the
            main SPI flash chip.

    menu "SPI RAM config"
        depends on ESP32S3_SPIRAM_SUPPORT

        choice SPIRAM_MODE
            prompt "Mode (QUAD/OCT) of SPI RAM chip in use"
            default SPIRAM_MODE_QUAD

            config SPIRAM_MODE_QUAD
                bool "Quad Mode PSRAM"

            config SPIRAM_MODE_OCT
                bool "Octal Mode PSRAM"
        endchoice

        choice SPIRAM_TYPE
            prompt "Type of SPIRAM chip in use"
            default SPIRAM_TYPE_AUTO

            config SPIRAM_TYPE_AUTO
                bool "Auto-detect"

            config SPIRAM_TYPE_ESPPSRAM16
                bool "ESP-PSRAM16 or APS1604"
                depends on SPIRAM_MODE_QUAD

            config SPIRAM_TYPE_ESPPSRAM32
                bool "ESP-PSRAM32 or IS25WP032"
                depends on SPIRAM_MODE_QUAD

            config SPIRAM_TYPE_ESPPSRAM64
                bool "ESP-PSRAM64 , LY68L6400 or APS6408"
        endchoice
```
Based on this information, we assumed that the esp-idf would provide support for the external PSRAM of our choice (IS25WP032D-JKLE). Based on a short analysis of the underlying code of the esp-idf, this support is only present for Espressif PSRAM (we tested our design with PSRAM from Espressif and these tests were successful). Hence, I was wondering whether there are any short-term plans of integrating support for PSRAM of other manufacturers into the esp-idf?

Kind regards,
Sam

huming2207, 2022-05-05:
I think `IS25WP032D-JKLE` is a NOR flash, not a PSRAM. Do you mean `IS66WVH16M8ALL` or something similar?

We are also thinking to use our own PSRAM, preferably 16MB RAM + 8 or 16MB Flash so that we can avoid getting in trouble for hardware memory mapping/addressing as per @ginkgm said. However there's no support or at least no documentation for those larger PSRAM...

herculanodavi, 2022-05-05:
@samvangysegem colleague here.

Yes, this is a NOR flash, but since it was in the `esp32s3/Kconfig` we imagined this could somehow be compatible. We are also open to try `APS6408`, `LY68L6400 `, `IS66WVH16M8ALL` or any other part number.

It would help us a lot it we could have more clarification as to how compatible are the modules listed in the `Kconfig` or why they show up there.

huming2207, 2022-05-05:
> Yes, this is a NOR flash, but since it was in the esp32s3/Kconfig we imagined this could somehow be compatible. We are also open to try APS6408, LY68L6400 , IS66WVH16M8ALL or any other part number.

Ah, indeed, they've got a typo there! 

Maybe Espressif can add some PSRAM and flash reconfiguration APIs in the currently existing secondary-stage bootloader, that would be great for these customization PSRAM/Flash combos.

ginkgm, 2022-05-06:
Hi @samvangysegem , @huming2207 , @herculanodavi ,

Yes you are right, that one was a typo, sorry for misleading and we will move it soon.

The problem here is that there is very few options for us to use. All the types you see (AP/LY/ESP, excluding the wrong IS25) are actually the same stuff with the same ID prefix, so for now we only support that and check for one ID in the code.

But if you have alternatives, we will be very interested!

Michael

huming2207, 2022-05-07:
Hi @ginkgm 

> But if you have alternatives, we will be very interested!

I had a quick glance and I guess these two might work:

- [66-67WVO16M8EDALL-BLL.pdf](https://github.com/espressif/esp-idf/files/8644217/66-67WVO16M8EDALL-BLL.pdf)
- [APM_PSRAM_OPI_Xccela_APS12808L_OBMx_v3_0a_PKG-1954889.pdf](https://github.com/espressif/esp-idf/files/8644219/APM_PSRAM_OPI_Xccela_APS12808L_OBMx_v3_0a_PKG-1954889.pdf)

The issue now is that some of our low-power HMI display products need high capacity (16MB, or 32MB with 16MB mapped in system heap + 16MB spare for users themselves to muck around on their own), relatively high throughput (80MHz Octal SPI), and relatively low sleep current (less than 100uA). To achieve these three goals, only ESP32-S3 and those two PSRAM chips can potentially do the trick. Therefore I think it will be fantastic if there are some sort of APIs or at least some documentation to allow us to use those PSRAMs.

Regards,
Jackson

huming2207, 2022-05-16:
Hmmm, I think this issue shouldn't be closed that soon, as the issue hasn't been really resolved yet...?

ginkgm, 2022-05-16:
Hi @huming2207 ,

Thanks for your information. These chips are possibly to be supported. We will try these chips first. The result could be one of the following.

1. Can work directly on the board once mounted and configured properly.
   The AP PSRAM you provided is promising to be this kind. But we are not sure until we have the condition to test it. If you already have one in hand, you may have a try: 1. replace the PSRAM on the official module, 2. enable the octal psram in the menuconfig. 3. build and flash to see what happened.

2. Needs special HW wiring or SW update.
   Since the PSRAM support is a system feature, which should be activeated before app_main(). IDF may either provide the support for it, or provide a way to override the default support logic to support the new PSRAM. 

3. Not compatible with Espressif SoC at all.
 
The testing of these chips will not happen so fast. Will give you feedback if we finished the testing.

huming2207, 2022-05-16:
Hi @ginkgm 

Thanks. Currently I do have some modules in hand but I haven't got any PSRAM chip listed above. I will speak to my company and see if they willing to buy me some chips later and I'll test it out. I also have a fast enough logic analyser that can help probing the signal and tune the timing if that's needed.

Jackson

huming2207, 2022-05-17:
Just realised that Winbond also has similar 128Mbit+ PSRAM products:

- https://www.winbond.com/hq/product/mobile-dram/hyperram/index.html?__locale=en&partNo=W957A8MFY
- https://www.winbond.com/hq/product/mobile-dram/pseudo-sram/index.html?__locale=en&partNo=W968D6DAG

But again, I've only quickly glanced the datasheet, no further investigation performed yet.

TGiles1998, 2022-05-17:
https://github.com/espressif/esp-idf/issues/8747#issue-1196525445

TGiles1998, 2022-05-17:
Hi @ginkgm,

Can the idf please allow for the mapping of 32MB of PSRAM?

https://github.com/espressif/esp-idf/issues/8747#issue-1196525445

Many thanks,

Thomas

huming2207, 2022-05-18:
> Can the idf please allow for the mapping of 32MB of PSRAM?

I guess we may be able to do something like the `himem` thing did for the classic ESP32 to access the upper 4MB of an 8MB PSRAM?

TGiles1998, 2022-05-18:
Can u suggest a patch so I could test this, really excited to have this wot as I've been stuck on the problem for a month, thank you very much in advance!

Thomas

TGiles1998, 2022-05-31:
@ginkgm Hi just following up on the 32MB compatibility?

ginkgm, 2022-12-19:
> Just realised that Winbond also has similar 128Mbit+ PSRAM products:
>
> * https://www.winbond.com/hq/product/mobile-dram/hyperram/index.html?__locale=en&partNo=W957A8MFY
> * [...]

W968: This is a 16-line chip. Currently no Espressif MCU support this kind of PSRAM.
W957: Sorry I can't open the datasheet of this device

ginkgm, 2022-12-19:
About the compatibility thing, I suggest you try APS12808 first. That one is 16MB. Maybe APS has PSRAM with bigger size.

About the map-able size, I see you are discussing that in #8747. I suggest continue the discussion there.

kosma, 2023-02-01:
Has anyone in this thread managed to get a 16MB PSRAM chip working on ESP32-S3? If so, can you share your experience, and the part numbers of things that have been found to work? I am currently going through the same pain, and hoping to use others' experience to hopefully cut down on the development time...

huming2207, 2023-05-22:
> Has anyone in this thread managed to get a 16MB PSRAM chip working on ESP32-S3? If so, can you share your experience, and the part numbers of things that have been found to work? I am currently [...]

@kosma  See this: https://github.com/huming2207/jumbo-s3 

For 16MB PSRAM, at least back in last year (2022), it worked out of the box and seems to be without any issue. But we haven't tested out other stuff like WiFi and BLE with our 16MB PSRAM.