espressif/esp-idf #8865: MQTT example project won't build...

davidlehrian, 2022-04-29:
SDK 4.4.1
Eclipse IDE

I'm trying to build the example project protocols/mqtt/ssl_mutual_auth and I get the following error:

> Building in: C:\Users\lehrian\Documents\ESP32\ssl_mutual_auth\build
> Configuring in: C:\Users\lehrian\Documents\ESP32\ssl_mutual_auth\build
> cmake -G Ninja -DCMAKE_TOOLCHAIN_FILE=C:\Users\lehrian\Documents\esp-idf-v4.4.1\tools\cmake\toolchain-esp32.cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCCACHE_ENABLE=1 -DIDF_TARGET=esp32 C:\Users\lehrian\Documents\ESP32\ssl_mutual_auth
> -- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.33.0.windows.1") 
> CMake Error at C:/Users/lehrian/Documents/esp-idf-v4.4.1/tools/cmake/utilities.cmake:108 (target_sources):
>   Cannot specify sources for target ".elf" which is not built by this
>   project.
> Call Stack (most recent call first):
>   CMakeLists.txt:11 (target_add_binary_data)
> 
> 
> CMake Error at C:/Users/lehrian/Documents/esp-idf-v4.4.1/tools/cmake/utilities.cmake:108 (target_sources):
>   Cannot specify sources for target ".elf" which is not built by this
>   project.
> Call Stack (most recent call first):
>   CMakeLists.txt:12 (target_add_binary_data)
> 
> 
> CMake Error at C:/Users/lehrian/Documents/esp-idf-v4.4.1/tools/cmake/utilities.cmake:108 (target_sources):
>   Cannot specify sources for target ".elf" which is not built by this
>   project.
> Call Stack (most recent call first):
>   CMakeLists.txt:13 (target_add_binary_data)

I get the same error when trying to build protocols/mqtt/ssl. It stems from the following lines in the CMakeLists.txt file.
```
target_add_binary_data(${CMAKE_PROJECT_NAME}.elf "main/client.crt" TEXT)
target_add_binary_data(${CMAKE_PROJECT_NAME}.elf "main/client.key" TEXT)
target_add_binary_data(${CMAKE_PROJECT_NAME}.elf "main/mosquitto.org.crt" TEXT)
```

davidlehrian, 2022-04-29:
I fixed the problem. It has to do with the order of the commands in the root CMakeLists.txt file. Here is the file in the examples:

```
# The following four lines of boilerplate have to be in your project's CMakeLists
# in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)

# (Not part of the boilerplate)
# This example uses an extra component for common functions such as Wi-Fi and Ethernet connection.
set(EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/examples/common_components/protocol_examples_common)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
get_filename_component(ProjectId ${CMAKE_CURRENT_LIST_DIR} NAME)
target_add_binary_data(${CMAKE_PROJECT_NAME}.elf "main/client.crt" TEXT)
target_add_binary_data(${CMAKE_PROJECT_NAME}.elf "main/client.key" TEXT)
target_add_binary_data(${CMAKE_PROJECT_NAME}.elf "main/mosquitto.org.crt" TEXT)
string(REPLACE " " "_" ProjectId ${ProjectId})
project(${ProjectId})

```
Here is a version that works:
```

# The following four lines of boilerplate have to be in your project's CMakeLists
# in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)

# (Not part of the boilerplate)
# This example uses an extra component for common functions such as Wi-Fi and Ethernet connection.
set(EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/examples/common_components/protocol_examples_common)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
get_filename_component(ProjectId ${CMAKE_CURRENT_LIST_DIR} NAME)

string(REPLACE " " "_" ProjectId ${ProjectId})
project(${ProjectId})

target_add_binary_data(${CMAKE_PROJECT_NAME}.elf "main/client.crt" TEXT)
target_add_binary_data(${CMAKE_PROJECT_NAME}.elf "main/client.key" TEXT)
target_add_binary_data(${CMAKE_PROJECT_NAME}.elf "main/mosquitto.org.crt" TEXT)

```

david-cermak, 2022-05-03:
@davidlehrian Yes, the `target_add_binary_data()` needs to be placed after project declaration, how it's done in the example Makefiles:

https://github.com/espressif/esp-idf/blob/8c8f700c1df05d8287316c78b98b940e119f900c/examples/protocols/mqtt/ssl/CMakeLists.txt#L9-L12

You can also find this described in the docs on IDF build system: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#embedding-binary-data

> Place this line after the project() line in your project CMakeLists.txt file. 

---

Could you please close this issue?

closed 2022-05-03 by davidlehrian

JarrettR, 2024-03-14:
This causes example projects generated from the ESP-IDF VSCode extension to fail. The extension sets the project name according to the user's project name.

I recommend cribbing a snippit of the above issue so the last line is changed as follows:

```
# The following four lines of boilerplate have to be in your project's CMakeLists
# in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.16)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(test-firmware)

target_add_binary_data(${CMAKE_PROJECT_NAME}.elf "main/mqtt_eclipseprojects_io.pem" TEXT)```