espressif/esp-idf #8868: settimeofday() not functioning

vogelfreiheit, 2022-04-30:
My version and hardware information:

```
Chip is ESP32-PICO-D4 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
```

ESP-IDF:

```
PLATFORM: Espressif 32 (4.1.0) > TTGO LoRa32-OLED V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf 3.40302.0 (4.3.2) 
 - tool-cmake 3.16.4 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - tool-mkfatfs 2.0.1 
 - tool-mklittlefs 1.203.210628 (2.3) 
 - tool-mkspiffs 2.230.0 (2.30) 
 - tool-ninja 1.7.1 
 - toolchain-esp32ulp 1.22851.191205 (2.28.51) 
 - toolchain-xtensa-esp32 8.4.0+2021r2-patch3
```

Problem: 

- settimeofday() returns 0, but when I try to get the time afterwards, it's still reporting 1970 epoch time 0.

Code:

```
static void update_rtc_time(gps_t *gps)
{
    struct tm tm = { 0 };
    time_t ntime, otime;

    tm.tm_hour = gps->tim.hour;
    tm.tm_min = gps->tim.minute;
    tm.tm_sec = gps->tim.second;
    tm.tm_year = gps->date.year + 100; /* since 1900 per struct tm */
    tm.tm_mon = gps->date.month;
    tm.tm_mday = gps->date.day;

    setenv("TZ", "UTC0", 1);
    tzset();

    otime = time(NULL);
    ntime = mktime(&tm);

    long int tdelta = ntime - otime;

    if (!rtc_fixed)
    {
        int err;
         struct timeval tv = { .tv_sec = ntime, .tv_usec = 0 };
        char strftime_buf[48];

        ESP_LOGI(TAG, "GPS->RTCL %lu (%s)", ntime, asctime(&tm));

        tv.tv_sec = ntime;        
        err = settimeofday(&tv, NULL);

        bzero(&tm, sizeof(tm));
        bzero(&ntime, sizeof(ntime));
        vTaskDelay(1200 / portTICK_PERIOD_MS);
        localtime_r(&ntime, &tm);

        strftime(strftime_buf, sizeof(strftime_buf), "%F %X %Z", &tm);
        ESP_LOGI(TAG, (delta=%ld, err %d): %s", tdelta, err, strftime_buf);

        rtc_fixed = true;
        return;
    }
}
```

vogelfreiheit, 2022-04-30:
The result:

` (delta=1653930410, err 0): 1970-01-01 00:00:00 UTC`

Note that delta is the actual UTC GPS time.

igrr, 2022-04-30:
It seems that ntime is zero because you aren't setting it between zeroing out and passing it to localtime_r?

```c++
        bzero(&ntime, sizeof(ntime));
        vTaskDelay(1200 / portTICK_PERIOD_MS);
        localtime_r(&ntime, &tm);
```

Perhaps `ntime = time(NULL);` is missing?

(Ref. https://en.cppreference.com/w/c/chrono/localtime)

vogelfreiheit, 2022-04-30:
I had been messing around the code and forgot to reset ntime, however this has nothing to do with my report. settimeofday is indeed failing, I added the second strftime() call merely to make it possible for you to test this without my full project. I can confirm that another task leveraging localtime, and gettimeofday(), get the wrong date-time consistently.

ntime in the sample is set to a proeprly adjusted tm structure containing the converted GPS timing, which, if you run strftime() against it, will yield a perfectly valid date-time (therefore it can be parsed).

I will update the sample code.

vogelfreiheit, 2022-04-30:
```
␛[0;32mI (2227) main.c: SET_TIME_1: GPS time: 2022-04-30 17:58:41 UTC (tv_sec 1651341520)␛[0m
␛[0;32mI (3427) main.c: EXPECTED TO MATCH SET_TIME_1: (delta=1651341520, ntime 1651341521, err 0): 2022-04-30 17:58:41 UTC␛[0m
␛[0;32mI (3557) mytask.c: Current RTC time: 1970-01-01 00:00:00 UTC␛[0m
```

The Current RTC print comes from another task:

```
        time_t now;
        struct timeval tm;
        char strftime_buf[48];
        localtime_r(&now, &tm);
        strftime(strftime_buf, sizeof(strftime_buf), "%F %X %Z", &tm);
        ESP_LOGI(TAG, "Current RTC time: %s", strftime_buf);
```

The task waits until rtc_fixed is set, so it is not actually running until update_rtc_time() has executed successfully and updated the status.

igrr, 2022-04-30:
Sorry that I misunderstood your code example!

Could you please try this simple snippet and see if it works on your side? https://gist.github.com/igrr/3f322b773eda81c713615a2337f37b94
If you can post a code snippet which can be compiled independently from the rest of your application and which reproduces the jssue, it would also help. The current one has a few undefined things (gps_t, rtc_fixed).

vogelfreiheit, 2022-05-01:
Absolutely no worries, that was my bad.

Admittedly there are a few things that seem unclear, and other people might find this page, so I will list what I have found out so far without a deep look into the implementation of settimeofday() in ESP-IDF:

- Which one of the tv_sec or tv_usec values is used seems implementation dependent, as usual with POSIX you cannot trust anyone to assume a fixed behavior. So, to be safe, both must be equal or one has to be zero. I am using seconds precision instead of microseconds because of the use case scenario.
- I detected a problem with my arithmetic that only manifested as soon as microsecond precision was lost due to setting the clock from a seconds precision source. If you are doing calculations of time deltas, and you assume microsecond precision, but get values that have the 000th values zeroed, well, things will go haywire pretty fast especially with signed data types. I have since fixed this.
- The delay seems to help, 100ms seem to be enough.
- I do not need a mutex so as long as the separate task never writes to the same structure.

The gps code is based off ESP-IDF's own NMEA parser, with the addition of a specific time sync status flag.

I will try to get back at this issue today or tomorrow.

igrr, 2022-05-05:
@vogelfreiheit Have you had a chance to look at this issue again? From my perspective `settimeofday` works correctly (based on the sample I posted), and our unit and example tests related to time are passing, I don't see any regression.

If you can create an example I can run locally to reproduce the issue, that would help!

vogelfreiheit, 2022-05-05:
Yes, I adjusted the tasks hold up time (the wait times inside the loops) and also added a mutex for handling the time synchronization event. It's gone. I left a 100ms pause between set and get and so far it is working.

If I encounter the issue again and it isn't operator error, I will reopen the issue if this is OK. Thank you for your time and the considerate help, much appreciated.

closed 2022-05-05 by vogelfreiheit