espressif/esp-idf #8873: bad behaviour esp_http_client when setting client->location

HomeACcessoryKid, 2022-05-01:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): v4.4.1
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [USB]
- Using Docker

## Problem Description
When using `    esp_http_client_set_redirection(client);` repeatedly, the URL gets appended to the previous one.

`D (13916) HTTP_CLIENT: Redirect to https://github.com/HomeACcessoryKid/life-cycle-manager/releases/tag/2.2.5https://objects.githubusercontent.com/github-production-release-asset-2e65be/127610504/f485d011-8f3c-49c9-ae06-8f341876d230`
```
static int http_on_header_value(http_parser *parser, const char *at, size_t length)
{
...
    if (strcasecmp(client->current_header_key, "Location") == 0) {
        http_utils_append_string(&client->location, at, length);
```

### Expected Behavior
http_utils_assign_string
### Actual Behavior
http_utils_append_string
### Steps to reproduce
see code
### Code to reproduce this issue

```cpp
based on idf/examples/system/ota/native-ota-example
    esp_http_client_config_t config = {
        .url = "https://github.com/HomeACcessoryKid/life-cycle-manager/releases/latest"
        .buffer_size    = 1024,
        .buffer_size_tx = 1024,
    esp_http_client_handle_t client = esp_http_client_init(&config);
    err = esp_http_client_open(client, 0);
    esp_http_client_fetch_headers(client);
    ota_write_data[esp_http_client_read(client, ota_write_data, BUFFSIZE)]=0;
    printf("flushed: %s\n",ota_write_data);
    esp_http_client_set_redirection(client);
    esp_http_client_set_url(client,"https://github.com/HomeACcessoryKid/life-cycle-manager/releases/download/0.1.0/native_ota.bin");
    err = esp_http_client_open(client, 0);
    esp_http_client_fetch_headers(client);
    if (code==302) {
        ota_write_data[esp_http_client_read(client, ota_write_data, BUFFSIZE)]=0;
        printf("flushed: %s\n",ota_write_data);        
        esp_http_client_set_redirection(client); // this fails because it still has the original location pre-pended to URL
        err = esp_http_client_open(client, 0);
        esp_http_client_fetch_headers(client);
    }
```
## Debug Logs

```
I (8406) native_ota_example: Starting OTA example
D (8416) esp_ota_ops: found ota app max = 2
D (8416) esp_ota_ops: finding factory app...
I (8426) native_ota_example: Running partition type 0 subtype 0 (offset 0x00010000)
D (8436) HTTP_CLIENT: Begin connect to: https://github.com:443
D (8436) esp-tls: host:github.com: strlen 10
W (8446) wifi:<ba-add>idx:1 (ifx:0, c6:86:42:23:12:43), tid:0, ssn:2, winSize:64
D (8466) esp-tls: [sock=54] Resolved IPv4 address: 140.82.121.4
D (8466) esp-tls: Enable TCP keep alive. idle: 5, interval: 5, count: 3
D (8476) esp-tls: [sock=54] Connecting to server. HOST: github.com, Port: 443
D (8516) esp-tls: handshake in progress...
D (12926) HTTP_CLIENT: Write header[3]: GET /HomeACcessoryKid/life-cycle-manager/releases/latest HTTP/1.1
[... 55 lines cut ...]
D (13666) HTTP_CLIENT: HEADER=Content-Security-Policy:default-src 'none'; base-uri 'self'; block-all-mixed-content; child-src github.com/assets-cdn/worker/ gist.github.com/assets-cdn/worker/; connect-src 'self' uploads.github.com objects-origin.githubusercontent.com www.githubstatus.com collector.github.com raw.githubusercontent.com api.github.com github-cloud.s3.amazonaws.com github-production-repository-file-5c1aeb.s3.amazonaws.com github-production-upload-manifest-file-7fdce7.s3.amazonaws.com github-production-user-asset-6210df.s3.amazonaws.com cdn.optimizely.com logx.optimizely.com/v1/events translator.github.com *.actions.githubusercontent.com wss://*.actions.githubusercontent.com online.visualstudio.com/api/v1/locations github-production-repository-image-32fea6.s3.amazonaws.com github-production-release-asset-2e65be.s3.amazonaws.com insights.github.com wss://alive.github.com; font-src github.githubassets.com; form-action 'self' github.com gist.github.com objects-origin.githubusercontent.com; frame-ancestors 'none'; frame-src render.githubusercontent.com viewscreen.githubusercontent.com notebooks.githubusercontent.com; img-src 'self' data: github.githubassets.com identicons.github.com github-cloud.s3.amazonaws.com secured-user-images.githubusercontent.com/ *.githubusercontent.com; manifest-src 'self'; media-src github.com user-images.githubusercontent.com/; script-src github.githubassets.com; style-src 'unsafe-inline' github.githubassets.com; worker-src github.com/assets-cdn/worker/ gist.github.com/assets-cdn/worker/
D (13796) HTTP_CLIENT: HEADER=Content-Length:653
D (13806) HTTP_CLIENT: HEADER=X-GitHub-Request-Id:B04B:0E10:B4108C:DF66D8:626E6053
D (13806) HTTP_CLIENT: http_on_headers_complete, status=302, offset=2701, nread=2701
D (13816) HTTP_CLIENT: http_on_body 42
I (13826) HTTP_CLIENT: Body received in fetch header state, 0x3ffb9ec4, 42
D (13826) HTTP_CLIENT: content_length = 653
D (13836) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=653
D (13846) HTTP_CLIENT: need_read=982, byte_to_read=982, rlen=611, ridx=42
D (13846) HTTP_CLIENT: http_on_body 611
D (13856) HTTP_CLIENT: http_on_message_complete, parser=3ffbe8f4
D (13856) HTTP_CLIENT: is_data_remain=0, is_chunked=0, content_length=653
flushed: <html><body>You are being <a href="https://objects.githubusercontent.com/github-production-release-asset-2e65be/127610504/f485d011-8f3c-49c9-ae06-8f341876d230?X-Amz-Algorithm=AWS4-HMAC-SHA256&amp;X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220501%2Fus-east-1%2Fs3%2Faws4_request&amp;X-Amz-Date=20220501T102627Z&amp;X-Amz-Expires=300&amp;X-Amz-Signature=6161a22a30d931e9859cd09fe37d23986da884db962843539c51e00143fa27b2&amp;X-Amz-SignedHeaders=host&amp;actor_id=0&amp;key_id=0&amp;repo_id=127610504&amp;response-content-disposition=attachment%3B%20filename%3Dnative_ota.bin&amp;response-content-type=application%2Foctet-stream">redirected</a>.</body></html>
D (13916) HTTP_CLIENT: Redirect to https://github.com/HomeACcessoryKid/life-cycle-manager/releases/tag/2.2.5https://objects.githubusercontent.com/github-production-release-asset-2e65be/127610504/f485d011-8f3c-49c9-ae06-8f341876d230?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220501%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220501T102627Z&X-Amz-Expires=300&X-Amz-Signature=6161a22a30d931e9859cd09fe37d23986da884db962843539c51e00143fa27b2&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=127610504&response-content-disposition=attachment%3B%20filename%3Dnative_ota.bin&response-content-type=application%2Foctet-stream
D (13986) HTTP_CLIENT: Write header[3]: GET /HomeACcessoryKid/life-cycle-manager/releases/tag/2.2.5https://objects.githubusercontent.com/github-production-release-asset-2e65be/127610504/f485d011-8f3c-49c9-ae06-8f341876d230?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220501%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220501T102627Z&X-Amz-Expires=300&X-Amz-Signature=6161a22a30d931e9859cd09fe37d23986da884db962843539c51e00143fa27b2&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=127610504&response-content-disposition=attachment%3B%20filename%3Dnative_ota.bin&response-content-type=application%2Foctet-stream HTTP/1.1
User-Agent: ESP32 HTTP Client/1.0
Host: github.com
Content-Length: 0

D (14276) HTTP_CLIENT: on_message_begin

```

## Other items if possible
n/a

shubhamkulkarni97, 2022-05-04:
Hi @HomeACcessoryKid, this issue is already fixed with https://github.com/espressif/esp-idf/commit/9a7d85c09baff80f94b099537954d7330eef586e on release/v4.4 branch.

Updating your branch to latest release/v4.4 will fix your issue.

HomeACcessoryKid, 2022-05-04:
Thanks @shubhamkulkarni97,
I am wondering about the underlying problem I want to solve, and the need to extract the LocationUrl.
The use of `esp_http_client_set_redirection(client);` is actually not the best way, but a workaround.
I now found that using this kills an existing TLS connection if the server changes.
What I have is that GitHub uses a main server which always redirects to a secondary server.
I hope to have two `client`, one for each server and both remain connected. This would speed up things a lot.
However, the struct where `client->location` is stored has been made private and cannot be accessed from user code.

So, asking for advice:
How can I extract the headerfield `Location:` without the side effect `esp_http_client_set_redirection` has?

HomeACcessoryKid, 2022-05-22:
Let me provide the answer to my last question myself and close the issue.
In order to capture the location without using the redirect you can make an event callback like this:
```C
char location[600];
esp_err_t ota_event_handler(esp_http_client_event_t *evt){
    if (evt->event_id==HTTP_EVENT_ON_HEADER && !strcasecmp(evt->header_key,"location")) {
        strcpy(location,evt->header_value);
    }
    return ESP_OK;
}

esp_http_client_config_t config1 = {
    ...,
    .event_handler = ota_event_handler,
};
```
This code can be optimised to use heap instead of a static char string but it works fine and is simple.
Thanks for your assistance

closed 2022-05-22 by HomeACcessoryKid