espressif/esp-idf #8875: ESP32S3: ESP-IDF light_sleep and deep_sleep examples bundled to IDF do not work

idea--list, 2022-05-01:
## Environment

- Development Kit:      Adafruit ESP32S3 Feather

- IDF version: V4.4.1
- Build System:         idf.py
- Compiler version: xtensa-esp32-elf-gcc V8.4.0
- Operating System:     Windows 10 x64
- (Windows only) environment type: ESP Command Prompt
- Power Supply:         USB

## Problem Description

On Adafruit ESP32S3 Feather board light_sleep and deep_sleep examples bundled to ESP-IDF compile, flash, but do not work as expected. Power consumption remains high and constant.

On an old ESP32 board both examples work as expected.

Interestingly using Arduino IDE V1.8.19, esp32 core V2.0.3-RC1, esp_sleep_enable_timer_wakeup() and esp_light_sleep_start() or esp_deep_sleep_start() i can reach the expected power consumptions.

### Expected Behavior
Low power consumptions with ESP-IDF on the new ESP32S3 chip

### Actual Behavior
My ESP32S3 board does not seem to enter ligt or deep sleep states

### Steps to reproduce

1. compile&flash the light_sleep or deep_sleep examples delivered with ESP-IDF V4.4.1 on an ESP32S3 target
2. monitor power consumption

idea--list, 2022-05-01:
Interestingly enough if i delete most conditional parts of the bundled examples until only these lines remain, then the ESP32S3 board behaves as expected and also the amp meter shows low consumption:

```/* Deep sleep wake up example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

#include <stdio.h>
#include <string.h>
#include <stdlib.h>
#include <time.h>
#include <sys/time.h>
#include "sdkconfig.h"
#include "soc/soc_caps.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_sleep.h"
#include "esp_log.h"
#include "driver/adc.h"
#include "driver/rtc_io.h"
#include "soc/rtc.h"

void app_main(void)
{

    const int wakeup_time_sec = 10;
    printf("Enabling timer wakeup, %ds\n", wakeup_time_sec);
    esp_sleep_enable_timer_wakeup(wakeup_time_sec * 1000000);

    printf("Entering deep sleep\n");

    esp_deep_sleep_start();
}

idea--list, 2022-05-01:
Hmmm... after flashing the above modified code once and reverting back to the original examples, also the original examplex work now on the ESP32S3 board.

This really puzzles me 🤔

igrr, 2022-05-05:
@idea--list did you manage to reproduce the issue again? Have tried the examples on an ESP32-S3 devkit, and both seem to be working...

Note that the ESP32-S3 DevKit-C board also has a CP2102 chip and an addressable LED. Both will continue to consume current when ESP32-S3 goes into deep sleep. It is better to use an ESP32-S3 module (instead of a devkit) to evaluate current consumption in deep sleep.

idea--list, 2022-05-24:
@igrr 
Today i tried the original light sleep and deep sleep examples that ship with esp-idf.
On my old ESP32 board both behave as expected. On my ESP32S3 board (Adafruit ESP32S3 Feather) only the deep sleep example behaves as expected.

When running the light sleep example on my ESP32S3 board i need to change BUTTON_GPIO_NUM_DEFAULT to let's say 12, but even after doing so, the power consumption drops to the expected level only while that pin is pulled up by connecting to the 3V3 pin. As soon as i disconnect pin 12 from 3V3, the power consumption increases to about 34mA without waiting until the sleep times out as set in line 50 of the example.

Do not know if this might be related somehow with #6612, but it seems to me the light sleep example might need some revision.

idea--list, 2022-05-24:
@igrr 
I can not really dubug this further as i am new to esp-idf and because of #6612 my board simply can not output any messages after going to light sleep (as i am using windows).

I am not sure if my finding would be related to a possible floating of pin 12. As if that would be the case, i guess my board would output "Entering light sleep" at random intervals. However my observation is that after seeing the terminal output "Waiting for GPIO 12 to go high..." i can wait however long i wish before pulling that pin up. I guess if this was a floating pin issue, my observation would be something else.

igrr, 2022-05-24:
> However my observation is that after seeing the terminal output "Waiting for GPIO 12 to go high..." i can wait however long i wish before pulling that pin up that leads to "Entering light sleep" output

Sorry that I wasn't clear. By "floating" I meant "high-impedance input not driven to any specific logic level". Maybe there is some very week pulldown on the pin which makes it default to low state. Because the pin reads back as "low", the example doesn't enter sleep [until it sees high logic level on the pin](https://github.com/espressif/esp-idf/blob/c5deb7a9713887263b63549a891282f51a65d167/examples/system/light_sleep/main/light_sleep_example_main.c#L54-L59). Connecting the pin to 3.3V makes the example proceed and enter the sleep mode.

> where i lose any further output because of https://github.com/espressif/arduino-esp32/issues/6612.

Yes, I think that's one of the limitations of the built-in USB_SERIAL_JTAG peripheral, that it is suspended during light sleep, and USB connectivity gets broken. The root cause of this issue is https://github.com/espressif/esp-idf/issues/8507 — it will need to be fixed in IDF first and then the fix should be pulled into Arduino. In the meantime, please try using UART as console output when developing sleep related features.

RS2021baseapp, 2022-08-26:
i am using esp32s3 in a custom board without any other component and trying the deep sleep example code.in this mode the current consumption will be 160uA. But in the datasheet only 7uA is mentioned.why esp32s3 is consuming this much amount of current in deep sleep mode??

[image]

xiaojie76, 2023-03-20:
@idea--list @RS2021baseapp did you reproduce this issue again?

closed 2023-04-24 by Alvin1Zhang